ct-Open-Source/tuya-convert #4: Stuck on flashing minimal build off tasmota on TECKIN SP22

fastriker, 2019-01-23:
I followed the [guide on heise](https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html). I already got the device info:

> ======================================================
> Getting Info from IoT-device
> VTRUST-FLASH 1.0
> (c) VTRUST GMBH https://www.vtrust.de/35c3/
> 
> READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
> ChipID: eab2f1
> MAC: DC:4F:22:EA:B2:F1
> SdkVersion: 1.5.3(aec24ac9)
> BootVersion: 4
> BootMode: 1
> FlashMode: 1M DOUT @ 40MHz
> FlashChipId: 144051
> FlashChipRealSize: 1048576
> FlashChipSize: 1048576
> FlashChipMode: 3
> system_upgrade_userbin_check: user2 0x81000
> 
> ======================================================

After this I try to get tasmota:

> curl http://10.42.42.42/flash3

And got the following response:

> Device should flash http://10.42.42.1/files/user2.bin and restart

Tasmota never started. Instead I see a SSID named `ESP_XXXXXX`. Connecting to it gives me the IP 192.168.4.2 and a default gateway on 192.168.4.1. I can ping the device on that address, but it didn't reply to http requests.

Because user2.bin is linked to `esp8266-ota-flash-convert.ino-0x81000.bin`: What is it? Why the command don't install `sonoff-minimal.bin`?

Do anybody have some hints for me, what I did wrong or some advises how I can continue without opening the device? Maybe this is also connected to issue #2?!

nanobyte84, 2019-01-23:
Can you telnet into 192.168.4.1?

fastriker, 2019-01-23:
Unfortunately not:

> Trying 192.168.4.1...
> telnet: Unable to connect to remote host: Connection refused

nanobyte84, 2019-01-23:
Did you cycle the power after doing the upgrade?

fastriker, 2019-01-23:
Yes, after a longer time. I guess more than 10 minutes. The `ESP_XXXXXX` network was already visible at that time.

krikk, 2019-01-23:
had exact the same experience on an Gosund SP1 (Version 2.4)

nanobyte84, 2019-01-23:
I just doublechecked and can ussure that after flash3 the message
> Device should flash http://10.42.42.1/files/user2.bin and restart

was just wrong and **thirdparty.bin** was flashed.
This message is fixed now.

Now I am right now trying to reproduce your error...

baxerus, 2019-01-24:
Device: **TECKIN SP22**

I got the new, right message after `curl http://10.42.42.42/flash3` (git tells me it is `commit 9a6970f8857018fe343ee53ad706c1534d5a9c89`), but now the led is just blinking. No network appears. Also pressing the button does nothing (it changed the relay state before flashing).

After power cycling it is blinking fast first in a around *half sec on, half sec off* rhythm. Than after about 15 secs it changes to a slower *one sec on, one sec off* blinking.

I tried some tasmota button usage (see https://github.com/arendst/Sonoff-Tasmota/wiki/Button-Usage) but nothing works.

I also tried if the device would reappear in a second attempt using `./start_flash.sh`, but also no luck.

chrisrock1984, 2019-01-25:
I had the same issue - no tasmota wifi after "Device should flash http://10.42.42.1/files/user2.bin and restart" message. Needed to open the LED bulb and flash it the old way ;-). I hope this will be fixed...

datayeah, 2019-01-25:
@GeorgeIoak you can open the bulb by pulling hard on the semi-transparent plastic-top of the bulb. It's just "clicked" in, so you will not harm the bulb. The rest inside is just hold together by a few screws.

austinlg96, 2019-01-25:
If it's just flashing, it sounds like it's in WPS mode. The outlets that I had never broadcasted an SSID, but they hooked right up when I pressed the WPS button on my router. The button did nothing because the default GPIO was wrong. The led on my outlet just flashed about once a second.

HeinrichLohse, 2019-01-25:
I have the same issue as @baxerus but with a Maxico W-DE004 (commit 9ff3d69b61780e43a98bd557c5a24d5c084eb3c0).

baxerus, 2019-01-26:
@HeinrichLohse I already tried using WPS with my Fritzbox, but it also didn't work.

But lucky me, after having the outlet connected to power for a longer time the network sonoff-XYZ appeared. Yeah! 👍 

Do not know exactly, something between 5 to 15mins. At least I discovered it after this time. But I was looking for it some days before for long enough for sure! So perhaps it is worth to try power cycling multiple times and waiting for some minutes inbetween (perhaps only a hickup)?

HeinrichLohse, 2019-01-26:
@baxerus, thanks for motivating me to just plug it in again. Surprisingly the sonoff-xxxx network showed up immediately. Now everythink works as expected.

sdwhite67, 2019-01-28:
I have just flashed 4 TECKIN SP23s (UK model) with Tasmota and it worked like a charm.  Set it up as a BlitzWolf and everything is working great.

willngton, 2019-02-01:
Pretty much same thing happened with SP20 US version.
The device just stops and no more led light just when trying to backup firmware stage.

sfossen, 2019-02-06:
> Pretty much same thing happened with SP20 US version.
> The device just stops and no more led light just when trying to backup firmware stage.

mine did the same, so I tried starting the script again, without restarting the plug, and then it backed up original and flashed the new.

modellbobby, 2019-02-13:
Hi Guys,

i have lot of TECKIN EU SP22 and i followed the procedure. I come till the SonoffXXXX AP turns on and then I only can setup my WIFI credentials on that but cannot set the model. Is this by design? All my TECKINS dissapear afterwards and seem to not connect to the net and are not reachable.

Please help!

kueblc, 2019-03-23:
Closing due to inactivity. If you're still having problems, please open a new issue with your device details and full logs.

closed 2019-03-23 by kueblc

nitinuae, 2020-04-12:
Hello
I dont know if replying to this closed issue is correct or not.
I installed the tuyaconvert on kali linux live CD and used it to flash a no name power strip (opened to see it was Tuya).

I am now stuck at ESP-xyz as wifi. It hands out 192.168.4.2 (also 4.3 if another device tries) but says refused to open page 192.168.4.1.

Am I stuck? Or this issue has opened again in latest build?