ct-Open-Source/tuya-convert #5: Globe branded bulbs

mssmison, 2019-01-23:
Device info:
```
READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: XXXXXX
MAC: xx:yy:xx:yy:xx:yy:xx:yy
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000
```

After trying to flash.
curl http://10.42.42.42/flash3
And got the following response:
Device should flash http://10.42.42.1/files/user2.bin and restart

Device went to 100% brightness but never seemed to restart. The ESP SSID didn't show up and the device no longer responds when I attempt to put it into pairing mode.

austinlg96, 2019-01-23:
I have a similar issue with Omoton branded smart outlets. Are you able to ping 10.42.42.42? 
   
ps: I can not thank the dev(s) enough, I have been *waiting* for this; literally months. I'm happy to help out however possible, I'm not very familiar with C and it's friends, but I'm no stranger to programming.

Jason2866, 2019-01-24:
@mssmison 
If you tried to flash Tasmota you have to use Flashmode DOUT (and only DOUT)
See https://github.com/arendst/Sonoff-Tasmota#important-user-compilation-information

drushbrook, 2019-01-25:
@mssmison if you flashed the version that was using sonoff-classic you may be able to recover it using WPS.