ct-Open-Source/tuya-convert #10: Successfully flashed Gosund SP111

cljk, 2019-01-24:
This is no issue at all - I just wanted to let you know that I just successfully flashed one of Gosund SP111 
 - these:
https://www.amazon.de/Steckdose-Gosund-Stromverbrauch-Fernsteurung-Kompatibel/dp/B07FVT15L5

First I tried it with a Dell XPS on Ubuntu 18.04 - but didn´t get this to work. The WLAN didn´t seem to come up.

Second try: raspberry pi B3+
The WLAN came up the device was recognized immediatly.... but the dump of the original firmware got stuck... I tried this ~~8 times - every time the process hang at another adress. I could attach a log later if interested.
Then I had the idea perhaps the WLAN connection might not be stable enough (plug was in another room). When I places the plug next to the raspberry it finally worked. Perhaps there is some space for error-handling on downloading the firmware..!?

It rebooted with a WLAN "sonoff-2152", I configured my AP-parameters, then switched the mode to "Blitzwolf SHP2" and then it worked like a charm (see https://www.bastelbunker.de/gosund-sp111-mit-tasmota/).

I didn´t calibrate measurement because currently not interested in it.

You can close the "issue" - just wanted to say thank you. No soldering needed - china-cloud eliminated.

cljk, 2019-01-24:
```
/home/pi/Desktop/tuya-convert/scripts /home/pi/Desktop/tuya-convert
======================================================
TUYA-CONVERT

https://github.com/vtrust-de/smarthome-smarthack
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
======================================================
  Starting AP in a screen
  Stopping any Webserver
  Starting Websever in a screen
  Starting Mosquitto in a screen

======================================================

IMPORTANT
1. Connect any another device (a smartphone or something) to the WIFI vtrust-flash
   The wpa-password is flashmeifyoucan
   This step is IMPORTANT otherwise the smartconfig will not work!
2. Put your IoT device in autoconfig/smartconfig/paring mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
3. Press ENTER to continue

======================================================
Starting pairing procedure in screen
RTNETLINK answers: File exists
Waiting for the upgraded device to appear
..................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
Create backup of entire FLASH from 10.42.42.42
Connected... Flashsize= 1048576
b'read 0x00000000 to 0x000003FF'
b'read 0x00000400 to 0x000007FF'
b'read 0x00000800 to 0x00000BFF'
b'read 0x00000C00 to 0x00000FFF'
b'read 0x00001000 to 0x000013FF'
b'read 0x00001400 to 0x000017FF'
b'read 0x00001800 to 0x00001BFF'
b'read 0x00001C00 to 0x00001FFF'
b'read 0x00002000 to 0x000023FF'
b'read 0x00002400 to 0x000027FF'
b'read 0x00002800 to 0x00002BFF'
b'read 0x00002C00 to 0x00002FFF'
b'read 0x00003000 to 0x000033FF'
b'read 0x00003400 to 0x000037FF'
b'read 0x00003800 to 0x00003BFF'
b'read 0x00003C00 to 0x00003FFF'
b'read 0x00004000 to 0x000043FF'
b'read 0x00004400 to 0x000047FF'
b'read 0x00004800 to 0x00004BFF'
b'read 0x00004C00 to 0x00004FFF'

...edited,,,.

b'read 0x0009C400 to 0x0009C7FF'
b'read 0x0009C800 to 0x0009CBFF'
b'read 0x0009CC00 to 0x0009CFFF'
b'read 0x0009D000 to 0x0009D3FF'
b'read 0x0009D400 to 0x0009D7FF'
b'read 0x0009D800 to 0x0009DBFF'
[... 69 lines cut ...]
system_upgrade_userbin_check: user2 0x81000

======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A minimal build of Sonoff-Tasmota v6.4.1 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!

```

datenzar, 2019-01-24:
@cljk What a nice coincidence
Just achieved the same (see my PR)! ;-)

cljk, 2019-01-24:
Nice... 

I just noticed my LED of the plug blinks slowly but constantly. Do you have this too?

datenzar, 2019-01-24:
yepp, my guess is that this can be altered with the module type

Jason2866, 2019-01-25:
See command `LedState`
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main

merlinschumacher, 2019-01-25:
Thanks for your addition!

closed 2019-01-25 by merlinschumacher