ct-Open-Source/tuya-convert #6: Choice of firmware

SynAckFin, 2019-01-24:
The sonoff-classic firmware goes into WPS config when it does not have any access point SSID and not into AP mode. This will cause problems for users whose router doesn't support this. 

The sonoff.bin v6.4.1  firmware is to large and will be rejected.

I had the same issues when developing my own OTA process. In the end I had to go back to previous versions of sonoff.bin and settled on sonoff.bin v6.2.1 which is small enough to be loaded.

I would recommend you did the same.

Jason2866, 2019-01-24:
Mhh, yes you are right there is no button to switch behaviour of wifi config.
But 6.2.1. isnt a good choice too. There is one bug in 6.2.1 which cause on some devices issues too.
So best choice is latest version sonoff basic. It has all needed functions included and is small enough.
https://github.com/arendst/Sonoff-Tasmota/blob/development/RELEASENOTES.md
http://thehackbox.org/tasmota/release/020300/sonoff-basic.bin

SynAckFin, 2019-01-24:
For some reason I never even looked at sonoff-basic. Good choice

SynAckFin, 2019-01-24:
Just tested it with the sonoff-basic.v6.4.1 and everything worked ok.

ctandi, 2019-01-24:
We decided to replace the current image with one (classic) that automatically connects to the VTRUST-Access-Point when installed. I will test the new image now and then update the guide.

SynAckFin, 2019-01-25:
Personally I think that is the wrong choice for a few reasons.

1. People may perform the firmware change on a headless Pi (I do) and as a result have issues configuring the device after the sonoff-classic-vtrust is installed. I don't even have a browser installed on it so had to do some port forwarding with ssh to configure it.

2. You now have an extra binary and code to maintain. When the next version of Tasmota appears you will have to apply the same changes and build your own binary.

3. I'm not sure that the sonoff-classic is that well used and it has a couple of unique features (e.g. WPS, SmartConfig) that don't appear in other firmware versions.

I think a better choice would be the standard sonoff-basic. It's smaller than the classic and supports more sensors/devices than the classic. See the [feature/sensor table](https://github.com/arendst/Sonoff-Tasmota/releases)

ctandi, 2019-01-25:
Thanks for the feedback. We thought that people could simply use their smartphone to move the device from the Flash-WLAN to their home network. One thing that we couldn't clarify yesterday was that if Tasmota creates an access-point when flashed for the first time. This was our goal from the beginning.

As far as I remember that didn't happen in the past, but would be crucial for almost all devices since GPIO 0 is rarely connected to a button. So can you confirm for us that sonoff-basic creates an access-point after flashing?

SynAckFin, 2019-01-25:
I can confirm that sonoff-basic does start an access point with the name sonoff-XXXX when it is first installed.

merlinschumacher, 2019-01-25:
I also tried. Sonoff-Basic is the obvious choice. I changed the files accordingly.

closed 2019-01-25 by merlinschumacher

modellbobby, 2019-02-11:
I have a real problem here. Flashed the device with Tuya convert and basic bin and the AP appears. But I cannot change the model there just the WIFI and the device restarts and Never appears in my WiFi nor does work the button because it is set to basic Sonoff. What shall I do ? One reason could be that the vtrust is still active?

commit 5e60cb5c (2019-01-25): Changed firmware to sonoff basic variant v6.4.1. Fixes #6 #9