ct-Open-Source/tuya-convert #7: Mirabella Genio Bulb Issue

drushbrook, 2019-01-24:
After the restart the bulb is flashing on and off.  No sonoff-xx AP  but the bulb is still connecting to the AP. Have attempted to reset it using the turn off and on process but it does do anything.

Device Flash Log

```
...
b'read 0x000FF800 to 0x000FFBFF'
b'read 0x000FFC00 to 0x000FFFFF'
~/src/tuya-convert
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 3a709f
MAC: 80:7D:3A:3A:70:9F
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000

======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A minimal build of Sonoff-Tasmota v6.4.1 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
pi@raspberrypi:~/src/tuya-convert $ curl http://10.42.42.42/flash2
Device is allready booting from userspace 2 (0x81000)pi@raspberrypi:~/src/tuya-convert $ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart
```

Web Log after restart
```
root@raspberrypi:~# screen -r
There are several suitable screens on:
        18855.smarthack-smartconfig     (24/01/19 09:16:18)     (Detached)
        18828.smarthack-mqtt    (24/01/19 09:15:30)     (Attached)
        18722.smarthack-web     (24/01/19 09:15:29)     (Detached)
        18599.smarthack-wifi    (24/01/19 09:15:28)     (Detached)
Type "screen [-d] -r [pid.]tty.host" to resume one of them.
root@raspberrypi:~# screen -r 18722
d3a3ac5d9&t=1451&sign=fa9b8c5bfa1ade0bb18ddab4544168fb (10.42.42.39) 7.60ms

URI:/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1466&sign=746562e93cd5d1d6fe7755146caf19b1
WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1466&sign=746562e93cd5d1d6fe7755146caf19b1)
[I 190124 09:35:49 web:2162] 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1466&sign=746562e93cd5d1d6fe7755146caf19b1 (10.42.42.39) 5.29ms

URI:/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1481&sign=2a2a4877765edb38ec500e8a450b7264
WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1481&sign=2a2a4877765edb38ec500e8a450b7264)
[I 190124 09:36:04 web:2162] 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1481&sign=2a2a4877765edb38ec500e8a450b7264 (10.42.42.39) 5.34ms

URI:/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1496&sign=a8a30b0b117b79db84431592801688d3
WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1496&sign=a8a30b0b117b79db84431592801688d3)
[I 190124 09:36:19 web:2162] 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1496&sign=a8a30b0b117b79db84431592801688d3 (10.42.42.39) 5.10ms

URI:/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1511&sign=fa6315e64c5fbe594b4f0ea1ce0059c2
WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1511&sign=fa6315e64c5fbe594b4f0ea1ce0059c2)
[I 190124 09:36:34 web:2162] 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=14682127807d3a3ac5d9&t=1511&sign=fa6315e64c5fbe594b4f0ea1ce0059c2 (10.42.42.39) 5.33ms
```
mqtt log after restart
```

1548322179: Sending PINGRESP to 14682127807d3a3ac5d9
1548322239: Received PINGREQ from 14682127807d3a3ac5d9
1548322239: Sending PINGRESP to 14682127807d3a3ac5d9
1548322299: Received PINGREQ from 14682127807d3a3ac5d9
1548322299: Sending PINGRESP to 14682127807d3a3ac5d9
1548322359: Received PINGREQ from 14682127807d3a3ac5d9
1548322359: Sending PINGRESP to 14682127807d3a3ac5d9
1548322419: Received PINGREQ from 14682127807d3a3ac5d9
1548322419: Sending PINGRESP to 14682127807d3a3ac5d9
1548322479: Received PINGREQ from 14682127807d3a3ac5d9
1548322479: Sending PINGRESP to 14682127807d3a3ac5d9
1548322539: Received PINGREQ from 14682127807d3a3ac5d9
1548322539: Sending PINGRESP to 14682127807d3a3ac5d9
1548322599: Received PINGREQ from 14682127807d3a3ac5d9
1548322599: Sending PINGRESP to 14682127807d3a3ac5d9
1548322659: Received PINGREQ from 14682127807d3a3ac5d9
1548322659: Sending PINGRESP to 14682127807d3a3ac5d9
1548322719: Received PINGREQ from 14682127807d3a3ac5d9
```

wifi log
```
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:c4:1f:ad and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA 80:7d:3a:3a:c5:d9 IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 80:7d:3a:3a:c5:d9
wlan0: STA 80:7d:3a:3a:c5:d9 RADIUS: starting accounting session 5C498231-00000000
wlan0: STA 80:7d:3a:3a:c5:d9 WPA: pairwise key handshake completed (RSN)
wlan0: STA <my mobile mac address> IEEE 802.11: associated
wlan0: AP-STA-CONNECTED <my mobile mac address>
wlan0: STA <my mobile mac address> RADIUS: starting accounting session 5C498231-00000001
wlan0: STA <my mobile mac address> WPA: pairwise key handshake completed (RSN)
wlan0: STA <my mobile mac address> WPA: group key handshake completed (RSN)
wlan0: STA 80:7d:3a:3a:c5:d9 WPA: group key handshake completed (RSN)
wlan0: STA 80:7d:3a:3a:c5:d9 WPA: group key handshake completed (RSN)
```

Jason2866, 2019-01-24:
See https://github.com/ct-Open-Source/tuya-convert/issues/5#issuecomment-457134737

drushbrook, 2019-01-24:
> See [#5 (comment)](https://github.com/ct-Open-Source/tuya-convert/issues/5#issuecomment-457134737)
I flashed the included sonoff-classic as per commit b2760dd6ac5aa45a561868acecd82ea6b1cbb4fd .
I thought you could flash DOUT over the top of a QIO flash.

I can't understand how the device is still sending tuya commands to the web.

Jason2866, 2019-01-24:
Strange. You have to flash Tasmota with option DOUT. 
All ESP devices do support DOUT. DOUT needs the fewest hardware (wiring of flash) connections.

drushbrook, 2019-01-24:
I wonder what firmware file they flashed for this [article](https://www.heise.de/ct/artikel/Tuya-Convert-Escaping-the-IoT-Cloud-no-solder-need-4284830.html)

ctandi, 2019-01-24:
I have to apologize. This might be due to changes in the sonoff-minimal. When we built these scripts, I had in mind that the minimal-image was used with SonOTA to overwrite the Sonoff-firmware and update to a full tasmota from there. This is still stated in the documentation, but has obviously changed and now the minimal image needs previous data of Tasmota present and might not work without it. We will work on resolving this today. @drushbrook I think you might need to open up the bulb and solder a serial connection to it :( Sorry.

drushbrook, 2019-01-24:
I have two bulbs.  One of them in the sonoff-minimal situation (which I believe I will need to use serial to recover) and the other is using sonoff-classic. The logs above are from the sonoff-classic firmware.

ctandi, 2019-01-24:
This seems rather strange. I suppose then flashing didn't work. We will get too this in a few hours.

drushbrook, 2019-01-24:
Here are the three times I did firmware backups using tuya-convert for the same device.  i.e. firmware-backup1.bin is the first backup, firmware-backup2.bin is the backup after doing an undo, firmware-backup3.bin is the third backup after doing an undo the second time after I did a git pull to get the sonoff-classic change.

[firmware-backups.zip](https://github.com/ct-Open-Source/tuya-convert/files/2791616/firmware-backups.zip)

ctandi, 2019-01-24:
We decided to replace the current image with one that automatically connects to the VTRUST-Access-Point when installed. I will test the new image now and then update the guide.

drushbrook, 2019-01-24:
Have you tried pushing the WPS button on your router which apparently is one of the ways Tasmota Classic can receive your AP information?  I tried with my router not realising that WPS is disabled.completely on Tomato firmware.  I'm going to setup another router with WPS support tonight with mine.

datayeah, 2019-01-25:
let me check that out. meanwhile, i managed to brick a second bulb. i managed to install the temporary firmware again, then tried to flash sonoff classic, afterwards: bright white light -> bulb doesn't connect to anything. i think it has to do with an unstable wifi AP. my raspberry seems to constantly drop all AP clients and so bricks the device while it tries to load the firmware from the temporary raspberry webserver. fun fact: when i reset the second bulb (turn off and on 3 times), it flashes constantly in bright white and warm white. never seen this before. i'll try the WPS thing now...

drushbrook, 2019-01-25:
My bulb was very similar in flashing action but instead of bring white and warm white. It would go on and off.  Using WPS made the device connect to my network and I upgraded it straight away to sonoff-basic.bin using the web interface. I then configured it to use Arilux LC11 (38) as a module as per https://github.com/arendst/Sonoff-Tasmota/wiki/Mirabella-Genio-Bulb

closed 2019-01-25 by drushbrook