ct-Open-Source/tuya-convert #9: ct article isnt correct at all

Jason2866, 2019-01-24:
There is no need to change this firmware -> if you have used the basic build and just hard coded the wifi credentials. Setup wifi ssid2 and wifi pwd2 with needed values in web console.
After restart Tasmota will try to connect to both APs. Even changing ssid1 and pwd1 is possible
but not recommended if you do a typo error you are unlucky...
If the user wants to reflash with a other build there is still NO need to flash minimal version.
Just do a OTA upgrade from http://thehackbox.org/tasmota/release/sonoff.bin
Tasmota loads itself the needed minimal version and finally the target firmware.
Just let the device alone and wait until webfrontend is back again (OTA magic).
```
curl http://10.42.42.42/flash3

in der Konsole aus. Die vorläufige Firmware lädt die Datei herunter und schreibt sie in den Flash-Speicher. Aufgrund der limitierten Größe der möglichen Firmwares, stellen wir zur Installation eine speziell angepasste Variante von Sonoff-Tasmota bereit. Diese Variante verbindet sich nach dem erfolgreichen Flashen zunächst mit dem Access-Point des Raspi.

Um die IP-Adresse des Tuya-Gerätes nach dem Flash-Vorgang zu identifizieren, bietet sich ein Netzwerkscanner fürs Smartphone wie etwa die HE.NET Network Tools oder Fing. Haben Sie die IP herausgefunden, greifen Sie auf das Webinterface des Geräts zu und können von dort ein Sonoff-Tasmota Minimal-Image flashen um schlussendlich ein normales Tasmota-Image aufzuspielen. Wir arbeiten noch einer Vereinfachung dieser Prozedur.
```

ctandi, 2019-01-24:
We chose the classic version because the normal version with its 522 kilobytes could be slightly too large when the temporary vtrust firmware is also running. I think the classic version doesn't automatically load the minimal version, because I tried installing sonoff-DE from there and it failed due to insufficient memory. Please educate us, if we're wrong at any point. We're still fairly new to digging this deep into the ESP.

sylvandb, 2019-01-25:
I prefer sonoff.bin because with no WiFi configured it defaults to setting up an access point instead of the various other configuration schemes.

The size may be an issue, and I do not believe any tasmota has code to automatically do a two-step OTA upgrade (minimal then the desired version). I'm willing to be proven wrong as I am also fairly new at this. Maybe there is a different path that I have not yet investigated.

Jason2866, 2019-01-25:
@ctandi 
The way you choose with "classic" is okay. With connecting to vtrust AP there is no need to have a WPS capable router. This firmware can be used and there is no need to replace. Just add wifi credentials via webfrontend and device is ready to go.
If you want to replace firmware there is no need to load manually the minimal version.
Do the upgrade via OTA URL `Upgrade by web server`. This way Tasmota does load needed minimal itself from OTA server.
You have to load minmal version if you do `Upgrade by file upload` if firmware sizes are too big.

I hope to clarify a bit. Two ways to upgrade the firmware:
-  OTA way (load minimal not needed, Tasmota does it itself if needed)
- local file upgrade (minimal has to be flashed if needed)

merlinschumacher, 2019-01-25:
. Sonoff-Basic is the obvious choice. I changed the files accordingly.

closed 2019-01-25 by merlinschumacher