ct-Open-Source/tuya-convert #15: #5 Globe branded smart bulbs

mssmison, 2019-01-25:
I'm afraid this isn't resolved with the latest changes to the version/bin of tasmota.  
I tried again today with a new bulb and it seems that it's not actually being flashed with anything. You can see below I initiated flash3, the bulb didn't seem to flash anything/restart and I was able to even undo and return this one to the original firmware.
I think the first bulb might be stuck with the flash loader on it still but it's not reconnecing to the AP unfortunately.

```
HAVE FUN!
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash2
Device is allready booting from userspace 2 (0x81000)pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restartpi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash2
Device is allready booting from userspace 2 (0x81000)pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 3a717d
MAC: 80:7D:3A:3A:71:7D
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000
pi@raspberrypi:~/tuya-convert $ ls
2019-01-23_15-13-50_readout.bin  config.txt       files              LICENSE    scripts         stop_flash.sh
2019-01-25_07-18-30_readout.bin  device-info.txt  install_prereq.sh  README.md  start_flash.sh
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/backup
File Not Found

pi@raspberrypi:~/tuya-convert $ cd files
pi@raspberrypi:~/tuya-convert/files $ ls
1.bin  esp8266-ota-flash-convert.ino-0x81000.bin  esp8266-ota-flash-convert_upg.bin  thirdparty.bin  user2.bin
pi@raspberrypi:~/tuya-convert/files $ cd ..
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash2
Device is allready booting from userspace 2 (0x81000)pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restartpi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/undo
Removing esp8266-ota-flash-convert and reboot
```''

datayeah, 2019-01-25:
Same with my bulbs ("Jinvoo"). Already bricked 2 of them. Will be able to reflash by cable and further investigate by tomorrow. I'll let you know if i find something.

======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: c6767f
MAC: CC:50:E3:C6:76:7F
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000

======================================================

mssmison, 2019-01-25:
This issue can be closed.  I used my own copy of sonoff basic and everything worked without issue.

Thanks for putting this together!

closed 2019-01-28 by merlinschumacher

folkg, 2019-06-14:
How were you able to configure your bulb?