ct-Open-Source/tuya-convert #19: cant connect to wifi 'vtrust-flash'

klccjeremy, 2019-01-26:
Hi, 

I was trying to use my phone to connect to the wifi AP using the credential given

WIFI: vtrust-flash
PASS: flashmeifyoucan
IP: 10.42.42.42

but it shows "unable to join the network"

Any idea what i need to do?

Thanks!

moneybag, 2019-01-26:
hm, did you use a static ip 10.42.42.42 ?
i thought thats the ip from the plug?

klccjeremy, 2019-01-26:
I thought we need to use phone to connect to the vtrust ap first in order to set the Tuya device to connect to it as well?

moneybag, 2019-01-26:
i didn t used a static ip, the device made a backup. but at the moment i cant flash it.
it has the ip 10.42.42.42 and i can ping it. if i disconnect it from the socket ping doesnt work.

dhami220, 2019-01-26:
If you have a different WiFi adapter, give that a try. Not all WiFi hardware can serve as access points, and that could be a cause.

moneybag, 2019-01-26:
ok,  but why it does successfully a backup of the device?

klccjeremy, 2019-01-26:
I'm using raspberry pi 3 B+ which should be ok I guess? 

> On 26 Jan 2019, at 10:30 PM, Gurjinder Dhami <notifications@github.com> wrote:
> 
> If

kurim, 2019-01-26:
With which device did you  try to connect to pi3b+?
With an Apple device? - if yes it might be the root cause, I use a notebook to connect to tht pi3b+

klccjeremy, 2019-01-26:
Yeah I'm using iPhone to connect. But if using laptop to connect to the AP, how can I set the Tuya device to connect to the AP without doing in mobile app?

kurim, 2019-01-26:
You dont neet a phone, notebook is fine.

just start_flash.sh > yes > connect with your np to vtrust-flash > set your smarthome device to searchmode, by holding the power button > press enter on pi3 > wait, now it took some time to find and backup.

Than follow the screen instruction.

klccjeremy, 2019-01-26:
Oh.. thanks for the info! I'll try later when I get home

klccjeremy, 2019-01-26:
> 
> 
> You dont neet a phone, notebook is fine.
> 
> just start_flash.sh > yes > connect with your np to vtrust-flash > set your smarthome device to searchmode, by holding the power button > press enter on pi3 > wait, now it took some time to find and backup.
> 
> Than follow the screen instruction.

I was using headless setup with Pi 3 B+, so the pi is connected to my home wifi. It might be the reason that my phone cant connect to vtrust AP earlier. After I connected the Pi to monitor and doing it on screen, I disconnected the pi from wifi and my phone is able to connect to vtrust now.

However the problem is that, the Tuya device itself supposed to be connected to vtrust with ip address of 10.42.42.42

anyone can elaborate how can I do it? @moneybag 

Thanks!

klccjeremy, 2019-01-26:
ok found the problem.. stupid me tested with a non tuya device. it works fine after i use a tuya device.

so probably can put a note on Readme to warn people it wont work on Rpi headless setup?

drushbrook, 2019-01-27:
I did all my tuya devices from a RPI3 headless (monitoring it via ssh over it's ethernet adapter). 

I believe you were meaning RPI connected wirelessly to your network?

In this situation I would have started start_flash.sh in a screen session.  In this way I would have the start_flash.sh to reattach to when the rpi3 disconnected from my router to start AP mode.

DeferoKohll, 2019-01-29:
I also have an issue with connecting to the vtrust-flash. In the smarthack-wifi.log it says:

> Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:d5:58:f3 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED

The network disappears after i try connecting, not sure what crashes. Can't find anything in the /var/logs. I'm using Raspbian Stretch Lite on a raspberry 3b+

jamietang68, 2019-01-30:
@DeferoKohll 
Is your raspberry connecting to your network using wifi? if yes please forget the network, I had same issue, and solve it after I stop it to connect to wifi.

nukleuz80, 2019-01-30:
For me too, stop the running WIFI an try again. It's working.

jimkernsjr, 2019-01-31:
I had that problem too.  

The monitoring interface fails.  I blame it on bad drivers, I could be wrong.  See my post, 3rd one:
https://github.com/ct-Open-Source/tuya-convert/issues/51

jimkernsjr, 2019-01-31:
**UPDATE:

If you are having trouble getting the accesspoint vtrust-flash to start, run this command BEFORE you run ./start_flash.sh:

killall wpa_supplicant

worked for me!

rb2k, 2019-02-10:
I had to bring up the monitoring interface first: 

iw wlan0 interface add mon0 type monitor; ifconfig mon0 up

kueblc, 2019-03-23:
Sounds like this was resolved, closing. If you need help, pull the latest, try again, and open a new issue with your device details and logs.

closed 2019-03-23 by kueblc

minax007, 2022-04-17:
Dear all, 

@Raumzeit shard following information, which has resolved my connection issue with vtrust-flash:

https://github.com/ct-Open-Source/tuya-convert/issues/892#issuecomment-1100248239