ct-Open-Source/tuya-convert #22: Incompatible? utorch LE7

chrisrock1984, 2019-01-26:
I have two "utorch LE7" ->https://www.gearbest.com/smart-lighting/pp_009633100556.html
They can be configured via the tuya app. When I start the flash ("start_flash.sh"), the bulb leaves the configuration state (red blinking turns into constant white) but the script does not progress any further. It keeps printing dots. 
Has anyone the same problem, with other brands?

chrisrock1984, 2019-01-27:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/2799823/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/2799824/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/2799825/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/2799826/smarthack-wifi.log)

nanobyte84, 2019-01-27:
@chrisrock1984 Can you provide the serial number and software version you can see in the tuya app?
I have a bulb looking exactly the same (serial starting with 21273133...) and sw ver 1.0.2 behaving the same.
The reason is a wrong implementation of the update mechanism. This is what i get in the serial console (Baudrate 74800):

> [notice]smart_wf_frame.c:3796 fw_url:http://10.42.42.1/files/1.bin
[notice]smart_wf_frame.c:3797 fw_md5:88eaa4b09dc5a493a77adf0a41c87ba0
[notice]smart_wf_frame.c:3798 serv_sw_ver:9.0.0
[notice]smart_wf_frame.c:3815 serv_sw_ver:90000 sw_ver:10002
[notice]smart_wf_frame.c:978 smt_frm_cntl.fw_ug.tp:1

> [err]mem_pool.c:589 tuya_httpc.c:475
[err]mem_pool.c:590 malloc failed size:0x400
[err]mem_pool.c:591 remain size:616
[err]tuya_httpc.c:2224 DownLoad Bin Fails.CurrIdx:0 3 1

chrisrock1984, 2019-01-27:
Hi ->
MCU-Modul: 1.0.2
WiFi-Modul: 1.0.2
There is no info about serial number just following:
Virtuelle ID: 51658673 + mac address

Where can I find the correct serial number in the tuya app?

chrisrock1984, 2019-03-23:
still no progress in implementing utorch le7? they are really cheap! please find the latest log files with the latest version of tuya-convert

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/2999727/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/2999728/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/2999729/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/2999730/smarthack-wifi.log)

kueblc, 2019-03-23:
Unfortunately, if the update mechanism is implemented incorrectly, there isn't much we can do. You'll have to flash these the old fashion way, with a physical connection over serial.

kueblc, 2019-10-04:
Closing, feel free to try the latest `tuya-convert` and open an issue if you have any other problems

closed 2019-10-04 by kueblc