ct-Open-Source/tuya-convert #24: WiFi network never appears

jptrsn, 2019-01-27:
I'm trying to use this project, but I get stuck at the step where I need to connect to the vtrust-flash network. Using a couple different phones and a couple different computers, I'm never able to see the network listed.

I've tried running tuya-convert on an rPi 3 as well as an old netbook running Ubuntu. Never see the wifi network appear on any device when I'm scanning.

qingz2004, 2019-01-27:
I'm using Raspberry Pi 3 B+. I can't see the access pointer either.
Got the following messages.
[image]
[image]

drushbrook, 2019-01-27:
When did you last do a git pull?

qingz2004, 2019-01-27:
I did it today.
I will try the previous version.

nanobyte84, 2019-01-27:
Make git pull again and it should work...

qingz2004, 2019-01-27:
The new script works now.
Thanks!

drushbrook, 2019-01-27:
Have you tried WPA PSK?

nanobyte84, 2019-01-27:
When you cannot enter the WIFI you probably need to use a different (usb) wifi adapter.
Which Hardware are you using?

botak13, 2019-01-27:
I use an Lenovo Laptop with Linux Mint and an onboard Wifi Adapter.
After I have entered the Wifi Passwort my Phone tries to get an IP Adress from vtrust-flash but it doesnt work. When I use Static IP the Phone connects but I still cant flash my Device.

qingz2004, 2019-01-27:
I was using Raspberry Pi 3 B+ build in WiFi adapter. It showed the access point, but could not connect to it.
After I added a Netgear USB WiFi adapter and change the config to wlan1, it worked.
Thanks!

botak13, 2019-01-27:
hmm I dont have an usb WiFi Adapter to test.
What is the difference to an Onboard WiFi?

jptrsn, 2019-01-27:
The latest version did not work, but checking out the commit before -Logfile option worked for me.

kurim, 2019-01-28:
It did work with PI 3B+ (I only get it work with raspbian not with Kali!)
I've just remove everything from my SD Card to have a clear setup:

Raspberry Pi 3B+ & Raspbian Stretch (2018-11-13-raspbian-stretch-lite.img)
Do pre-setup because ssh is disabled by default you need to enable it first

> sudo apt update && apt update -y
> sudo apt install git
> git clone https://github.com/ct-Open-Source/tuya-convert
> cd tuya-convert
> ./install_prereq.sh
> ./start_flash.sh

Wifi was come up and is working fine.
[image] [image]

Since I currently dont have any device to flash it stand still here:
[image]

Zefau, 2019-01-28:
Unfortunately, WiFi does not show up for me. I'm on the same system as kurim (Raspberry Pi 3B+ & Raspbian Stretch using 2018-11-13-raspbian-stretch-lite.img).

kurim, 2019-01-28:
Thats strange.
It might not necessary but did you setup your wifi region? (raspi-config)

Zefau, 2019-01-28:
Apparently I had the WiFi setup to connect to my home WiFi, which is why the scripts couldn't use it. I deleted the WiFi settings and the script works now.
Thanks

geertvercamer, 2019-01-28:
started from scratch, didn't setup wifi, the network doesn't show. Reboot, nothing, set WiFi country, reboot, nothing...
Pi2B + usb wifi adapter

Vendo232, 2019-01-29:
same on my end, I`m using HP laptop with built in WLAN and USB WLAN ( wlan1).
edited config.txt , tried fresh Kali, ubuntu, Mint install

can`t make the AP appear.

plazomat, 2019-02-04:
How many places do you change the WLAN# and which files? Same issue here, Mint fresh install on IBM laptop.

Vendo232, 2019-02-04:
> How many places do you change the WLAN# and which files? Same issue here, Mint fresh install on IBM laptop.

what worked for me was renaming WLAN0 from the random name Linux OS will assigned to WIFI module to wlan0 ( use command ifconfig)

I used this command

$ sudo nano /etc/default/grub
Look for GRUB_CMDLINE_LINUX and add the following net.ifnames=0 biosdevname=0.
From:
GRUB_CMDLINE_LINUX=""
To:
GRUB_CMDLINE_LINUX="net.ifnames=0 biosdevname=0"
Generate a new grub file using the following command.
$ sudo grub-mkconfig -o /boot/grub/grub.cfg
Restart PC

kueblc, 2019-02-04:
You can modify the interface in `config.txt`. I definitely do NOT recommend modifying your GRUB configuration for this.

Vendo232, 2019-02-04:
> You can modify the interface in `config.txt`. I definitely do NOT recommend modifying your GRUB configuration for this.

I think if you setup a machine to one and only purpose ( flash TUYA devices ) the modification of GRUB should not be an issue. Besides you can revert the modification once you are done with flashing.

plazomat, 2019-02-04:
Thanks - I will try with the just the config.txt modify to whatever ifconfig tells me the adapter is.

plazomat, 2019-02-04:
If that don't work go after the GRUB file, right now the machine is only to flash tuya, it was a paperweight until tuya-convert..

jptrsn, 2019-02-05:
Closing issue as my install worked. Please open a new issue for problems you encounter.

closed 2019-02-05 by jptrsn

ashfaaaa, 2019-03-22:
anyone found a fix for this issue? tried with both internal and external wifi adapter but vtrust-flash does not  show

kueblc, 2019-03-22:
@ashfaaaa Please open a new issue with your logs and details about your problem and hardware