ct-Open-Source/tuya-convert #25: brick devices

nnhhuu, 2019-01-27:
hello, thank you for this effort, 
I started the process,  bit i think i bricked my devices (i tried with tow)

first o had to modify the start_flash.sh, because there was an error in the 'screen' log, i changed  
sudo screen **-L -L**ogfile smarthack-wifi.log -S smarthack-wifi -m -d ./setup_ap.sh  FOR
 screen **-L smarthack-wifi.log** -S smarthack-wifi -m -d ./setup_ap.sh 

then 
**./start_flash.sh**    with this result

OKVTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: c8fa34
MAC: 68:C6:3A:C8:FA:34
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440a1
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000
curl http://10.42.42.42   with this result 

 
then 
**curl http://10.42.42.42/flash2**

and then the device die and no answer to any bottom :) 

Any suggestion ... thank you

nanobyte84, 2019-01-27:
Can you still ping 10.42.42.42 ?
Is there a new wifi called vtrust-recover?
Did you restart the device?

uav2608, 2019-01-27:
I do have the same issue, I flashed Gosund Socket but it looks it becomes brick. I cannot ping to 10.42.42.42 and no vtrust-recover wifi exists.

nnhhuu, 2019-01-27:
no ping to 10.42.42.42
the device does not do anything, there is not light in the leds. 
honestly i did not check for a wifi called vtrust-recover

after that happened the device just died.  is there any way to relive it?

nnhhuu, 2019-01-27:
i don't know what i did, press both buttons at the same time for a long period of time or press one and then the second,  all the possible combinations, or  may be i scream too loud,  but now they are alive and with tasmota installed... really thank you for this project....  i still have one device to flash, i'll do very carefully...

closed 2019-01-27 by nnhhuu