ct-Open-Source/tuya-convert #26: After flashing, device reboot loop

Diamond2, 2019-01-27:
here is log after flashing:

VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 43d1cc
MAC: 2C:3A:E8:43:D1:CC
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440e0
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user1 0x01000

URI:/gw.json?a=s.gw.token.get&gwId=012005132c3ae843d1cc&other={"token":"00000000","region":"EU","tlinkStat":{"configure":"smartconfig","time":8,"source":"ap","path":"broadcast"}}&t=8&v=3.0&sign=13f055737f99ca2b82add6fbdd34a8a1
Answer s.gw.token.get
[32m[I 190127 08:39:20 web:2162][m 200 POST /gw.json?a=s.gw.token.get&gwId=012005132c3ae843d1cc&other={"token":"00000000","region":"EU","tlinkStat":{"configure":"smartconfig","time":8,"source":"ap","path":"broadcast"}}&t=8&v=3.0&sign=13f055737f99ca2b82add6fbdd34a8a1 (10.42.42.12) 0.59ms

URI:/gw.json?a=s.gw.dev.pk.active&gwId=012005132c3ae843d1cc&other={"token":"00000000"}&t=7&v=3.0&sign=b865a96f66a474027130865de25de4fd
Answer s.gw.dev.pk.active
READ GW ID 012005132c3ae843d1cc
TRIGGER UPGRADE IN 10 SECONDS
[32m[I 190127 08:39:21 web:2162][m 200 POST /gw.json?a=s.gw.dev.pk.active&gwId=012005132c3ae843d1cc&other={"token":"00000000"}&t=7&v=3.0&sign=b865a96f66a474027130865de25de4fd (10.42.42.12) 8.04ms
Trigger upgrade in 10 seconds

URI:/gw.json?a=s.gw.update&gwId=012005132c3ae843d1cc&t=10&v=2.0&sign=17d74a0c863c7e36dbccea4088ceeb28
Answer s.gw.update
[32m[I 190127 08:39:24 web:2162][m 200 POST /gw.json?a=s.gw.update&gwId=012005132c3ae843d1cc&t=10&v=2.0&sign=17d74a0c863c7e36dbccea4088ceeb28 (10.42.42.12) 2.57ms

URI:/gw.json?a=s.gw.dev.update&gwId=012005132c3ae843d1cc&t=8&v=2.0&sign=2eda8f6c8763b98a9a1fbdd08ac017d4
Answer s.gw.update
[32m[I 190127 08:39:26 web:2162][m 200 POST /gw.json?a=s.gw.dev.update&gwId=012005132c3ae843d1cc&t=8&v=2.0&sign=2eda8f6c8763b98a9a1fbdd08ac017d4 (10.42.42.12) 1.97ms

URI:/gw.json?a=atop.online.debug.log&gwId=012005132c3ae843d1cc&t=8&sign=7c4076ba2302dfc40e1e68713dae491c
Answer atop.online.debug.log
[32m[I 190127 08:39:27 web:2162][m 200 POST /gw.json?a=atop.online.debug.log&gwId=012005132c3ae843d1cc&t=8&sign=7c4076ba2302dfc40e1e68713dae491c (10.42.42.12) 2.89ms

URI:/gw.json?a=tuya.device.dynamic.config.get&gwId=012005132c3ae843d1cc&t=7&v=1.0&sign=3e4877a1b0b5b193976ce93ed1d0d769
WARN: unknown request: tuya.device.dynamic.config.get (/gw.json?a=tuya.device.dynamic.config.get&gwId=012005132c3ae843d1cc&t=7&v=1.0&sign=3e4877a1b0b5b193976ce93ed1d0d769)
[32m[I 190127 08:39:27 web:2162][m 200 POST /gw.json?a=tuya.device.dynamic.config.get&gwId=012005132c3ae843d1cc&t=7&v=1.0&sign=3e4877a1b0b5b193976ce93ed1d0d769 (10.42.42.12) 0.71ms
437213e8010938d96e0fccf3b1184caa
b'2.1010938d96e0fccf3t5inCMDHpcJKQpxfOe2RbRJQzq/8g9mFpomEPpw6F4r+CnlIYZNKR5d4Pa/nSetXJsm5uMafEvnzLLz/1MQ4v8xa0Zm6fetNRMAjbEOLCBHdg4SXZL9kLEC9aqKXAOwI'

URI:/gw.json?a=s.gw.upgrade&gwId=012005132c3ae843d1cc&t=10&sign=840cf218e207096f6efc7eb76a39a750
Answer s.gw.upgrade
b'{"result":{"auto":3,"fileSize":"508475","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"88eaa4b09dc5a493a77adf0a41c87ba0"},"t":100,"e":false,"success":true}'
[32m[I 190127 08:39:31 web:2162][m 200 POST /gw.json?a=s.gw.upgrade&gwId=012005132c3ae843d1cc&t=10&sign=840cf218e207096f6efc7eb76a39a750 (10.42.42.12) 14.21ms

URI:/gw.json?a=s.gw.upgrade.updatestatus&gwId=012005132c3ae843d1cc&t=100&sign=0626f8711c5498fbdd454fc20aba3e90
Answer s.gw.upgrade
b'{"result":{"auto":3,"fileSize":"508475","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"88eaa4b09dc5a493a77adf0a41c87ba0"},"t":100,"e":false,"success":true}'
[32m[I 190127 08:39:31 web:2162][m 200 POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=012005132c3ae843d1cc&t=100&sign=0626f8711c5498fbdd454fc20aba3e90 (10.42.42.12) 6.13ms
[32m[I 190127 08:39:37 web:2162][m 206 GET /files/1.bin (10.42.42.12) 11.43ms
[32m[I 190127 08:39:38 web:2162][m 206 GET /files/1.bin (10.42.42.12) 1499.07ms
[32m[I 190127 08:43:53 web:2162][m 200 GET /files/user2.bin (10.42.42.42) 7718.20ms
[32m[I 190127 08:44:09 web:2162][m 200 GET /files/thirdparty.bin (10.42.42.42) 8874.73ms

I guess firmware sonoff-basic was transferred successfully to device but it can't boot. Device is now reboot loop. I can hear the switch on and off repeat.

nanobyte84, 2019-01-27:
When did you do the last git pull?
Can you press the WPS Button on your router?
Please wait up to 30 min in the reboot loop and check periodically for appearing WIFIs

Diamond2, 2019-01-27:
> 
> 
> When did you do the last git pull?
> Can you press the WPS Button on your router?
> Please wait up to 30 min in the reboot loop and check periodically for appearing WIFIs

Thanks for reply nanobyte84.
It is a7eb9335fc4e8a2513814e409414992beac79d58		branch 'master' of https://github.com/ct-Open-Source/tuya-convert
I'm afraid I can't wait 30 mins... The wifi switch will have smell... due to like 60 times/mins on then off...
So the firmware is finding SSID or something?

Diamond2, 2019-01-28:
Update: I can see Sonoff-xxxx SSID from the device. I can connect to it with no password. My phone have IP address 192.168.4.2.
Please advice...

Diamond2, 2019-01-28:
Ah, I can connect to 192.168.4.1 and see the tasmota interface there so the flash was successful. It seems that I need to reconfigure the button or something in the firmware. Touch buttons don't work. Led isn't correct.

closed 2019-01-28 by Diamond2