ct-Open-Source/tuya-convert #33: One way flash Gosund SP111

uav2608, 2019-01-28:
Flushed successfully Gosund SP111 with Raspberry Pi 3B with Raspbian Stretch but it worked only one way, I could not undo. Also button does not work any more? Any ideas?

digiblur, 2019-01-28:
Did you get it into Tasmota?

uav2608, 2019-01-28:
yes, it flashed with Tasmota and web interface works but I cannot use button to on/off and reset wifi settings in Tasmota.

digiblur, 2019-01-28:
Did you assign it as a generic template? Then figure out which pin is the button?

erwill2, 2019-01-28:
Configuration--> Configure Module-->▼ BlitzWolf SHP (45) and Click Save.

uav2608, 2019-01-28:
Honestly, Tasmota is new for me and will work with configuration. Thank you for advice.

erwill2, 2019-01-28:
I'm assuming this will work. The Gosund SP111 is listed as compatible with BlitzWolf SHP (45) configuration  in the wiki [Compatible Devices](https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices).

digiblur, 2019-01-28:
@uav2608 go to about 8:30 in this video.
https://youtu.be/5Oa27pCHtYo

merlinschumacher, 2019-01-28:
After you flashed Tasmota it's not possible to undo the flash via the scripts provided.

Is this still an issue?

uav2608, 2019-01-28:
> @uav2608 go to about 8:30 in this video.
> https://youtu.be/5Oa27pCHtYo

Thank you for the link. I will try to apply it.

uav2608, 2019-01-28:
> After you flashed Tasmota it's not possible to undo the flash via the scripts provided.
> 
> Is this still an issue?

I think it is possible as soon as I can put the device into paring mode. I will configure the modules as proposed above.