ct-Open-Source/tuya-convert #37: Wireless interface isn't named "wlan0"

SteveDinn, 2019-01-28:
I'm attempting to run this script on a Linux Mint 18.3 laptop. It doesn't have a "wlan0" wireless interface. Instead, it's called "wlp2s0". The scripts seem to be inferring or assuming that my interface is "wlan0". I've tried hacking a few scripts to set the $WLAN variable, but I still get this when attempting to actually flash.
```
======================================================
Starting pairing procedure in screen
Cannot find device "wlan0"
```

With my hacked-in variable, the log file looks much better and references the correct interface, but it still won't flash. Where does the script come up with "wlan0"? I can't find out where it gets this name from.

merlinschumacher, 2019-01-28:
Edit config.txt and change it accordingly.

SteveDinn, 2019-01-28:
Hah. Thanks. I was grepping through *.sh and *.py.

closed 2019-01-28 by SteveDinn

merlinschumacher, 2019-01-28:
Maybe we should make that clearer in the description.