ct-Open-Source/tuya-convert #39: MQTT port 1883 in use

ragenhe, 2019-01-28:
My intent was to use one of my existing pi's.  Of the three, I choose the one running node-red.  Not a huge deal, I monitored my node-red env to ensure no conflicts happened.  Well, I do have a conflict, it's my existing mqtt broker that runs within node-red.

Please advise which resolution / option would be better:
1) Uninstall tuya-convert and redo on a different pi / image.  Details of how please?
2) Shutdown node-red briefly while performing flash.
3) Change mqtt default port on flash installed mosquitto broker.  Where are the defaults loaded from?

kueblc, 2019-01-28:
Your first two solutions may work, the third may not, given that the Tuya device expects to connect to a broker on a specific port.

I think that ideally, we should detect that a broker is already running and use that, rather than stopping and starting a new instance of mosquitto. I suppose this may bring about some new issues if the broker is password protected.

In the meantime, you could comment out these two lines in `start_flash.sh`:

    sudo service mosquitto stop >/dev/null 2>&1
    $screen_with_log smarthack-mqtt.log -S smarthack-mqtt -m -d mosquitto -v

ragenhe, 2019-01-28:
Thank you @kueblc -  will comment the lines out and let it use my existing broker, which isn't password protected.

GeorgeIoak, 2019-01-28:
Along these lines what happens if you have a different computer running mosquito broker, is that going to interfere with this process?

merlinschumacher, 2019-01-29:
Can this be closed?

ragenhe, 2019-01-29:
Yes sir! I got it!

closed 2019-01-29 by ragenhe

kueblc, 2019-01-29:
@merlinschumacher Perhaps we can open a new issue tagged as a feature request regarding automatic MQTT broker conflict resolution. I have some ideas, if you'd like help, or to assign the issue to me.