ct-Open-Source/tuya-convert #40: flaky wifi vtrust-flash

behrisch, 2019-01-28:
The vtrust-flash network only appears irregularly on my Raspberry Pi 3 (with the built-in WLAN adapter and a clean Raspbian Stretch). The best chances are after a fresh reboot. But even then my phone can connect to the wifi but immediately disconnects afterwards. Any help is welcome. Here is my wifi log:
```
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:e8:55:5a and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Failed to create interface mon.wlan0: -95 (Operation not supported)
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:e8:55:5a and ssid "vtrust-flash"
random: Cannot read from /dev/random: Resource temporarily unavailable
random: Only 0/20 bytes of strong random data available from /dev/random
random: Not enough entropy pool available for secure operations
WPA: Not enough entropy in random pool for secure operations - update keys later when the first station connects
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
random: Cannot read from /dev/random: Resource temporarily unavailable
```

kueblc, 2019-01-28:
Looks like you're out of entropy, try installing `haveged` via `sudo apt-get install haveged`.

mobidarabbit, 2019-01-28:
I had an similar problem on my pi zero w with build in adapter. I had to kill my running wpa_supplicant process to make the ap work.

closed 2019-01-29 by merlinschumacher

behrisch, 2019-01-29:
I just wanted to mention that haveged did not fix it. I will try the wpa_supplicant later.

kueblc, 2019-01-29:
Let us know what you find

kueblc, 2019-01-29:
Are you still getting `random: Cannot read from /dev/random: Resource temporarily unavailable`?

behrisch, 2019-02-10:
The errors about randomness disappeared but it was still not working reliably. It seems to be the wpa_supplicant. I cannot always kill it because I lose my connection to the RPi then, unless I connect to my RPi using LAN. I tried that and the AP seems to be up. Unfortunately my smartphone still disconnects after about five seconds from the vtrust-flash network but this does not seem to be a problem for the flashing process which continued nevertheless.

mobidarabbit, 2019-02-11:
Hi, so it seems the flash works for you. It is better to connect over lan or directly over keyboard/HDMI, because the AP settings make wlan0 unreachable for your home network, so your are not able to connect.