ct-Open-Source/tuya-convert #42: Video walk through is available for use.

digiblur, 2019-01-28:
I did a video walk through of the entire process of a blank SD card to a flashed Tasmota device. This might help cut down on some of the issues people are having as more and more people try the process out.

https://youtu.be/O5GYh470m5k

ragenhe, 2019-01-28:
I for one was led here by your video. Thanks for putting it together.

scfigg, 2019-01-28:
Agreed - I came here from your video.

In the event of a SSID change, is there any way to get back into Tasmota to configure settings? Haven't run into the issue now. Just wondering for the future.

digiblur, 2019-01-28:
If you configure the your button as a button correctly then yes you can do a 4 button press to pull up the AP again which it might already be in if it can't connect to the SSID. You can also do a 40 second hold as well. Or worse comes to worse but in a backup SSID and password in the AP2 position.

scfigg, 2019-01-28:
Awesome, good to know. Just trying to figure out what switches are compatible and what I should buy before they patch this :)

digiblur, 2019-01-28:
Seems most I have heard from are having success.

digiblur, 2019-01-29:
Livestream where I showed how to setup two plugs and figure out the in's and out's of things of what pins go to which switches/buttons/relays etc..

https://www.youtube.com/watch?v=m_O24tTzv8g

kueblc, 2019-09-28:
Created a [wiki page](https://github.com/ct-Open-Source/tuya-convert/wiki/Additional-Resources) to host these sorts of things, feel free to edit and add

closed 2019-09-28 by kueblc