ct-Open-Source/tuya-convert #43: Help configuring module

ragenhe, 2019-01-28:
I successfully flashed three of these...

https://www.amazon.com/gp/product/B07GRLQV47/ref=ppx_yo_dt_b_asin_title_o02__o00_s00?ie=UTF8&psc=1

However, I bricked the third one when I selected "Generic" module type and saved.  It never recovered from the reboot.

Lesson learned.  I will wait right here until I understand better how to pick which module type.

Thanks in advance!

digiblur, 2019-01-28:
Generic shouldn't have bricked anything as that's a profile with no inputs or outputs configured.

digiblur, 2019-01-29:
Do some trial and error to pick the pins... had a few people asking me today about this process so I'm going to do a video on it tonight.

digiblur, 2019-01-29:
Livestream is archived here.  I showed how to setup two plugs and figure out the in's and out's of things.  

https://www.youtube.com/watch?v=m_O24tTzv8g

ragenhe, 2019-01-29:
Thanks @digiblur - HUGE help to me!!!!

closed 2019-01-29 by ragenhe

jnalley20, 2019-01-30:
> Livestream is archived here. I showed how to setup two plugs and figure out the in's and out's of things of what pins go to which switches/buttons/relays etc..
> 
> https://www.youtube.com/watch?v=m_O24tTzv8g

Thank you for your help!  This came out just at the right time for me and the rest of the Tuya-Converters!

digiblur, 2019-01-30:
@jnalley20 that's the reason I did it.  People were asking and getting stuck all during the day so a live stream was the quickest way to get the content out the door plus have some fun with some folks doing it.