ct-Open-Source/tuya-convert #45: Successful flash for Lyasi Light Bulb

Eichhoernchen, 2019-01-29:
Hi,

I successfully flashed this Lyasi bulb: [Amazon Link](https://www.amazon.de/gp/product/B07KYGDS3P)

Tosmota requires small adaptations (GPIO pins have a different mapping to the LED controller) that I'm trying to get in, see https://github.com/arendst/Sonoff-Tasmota/issues/1822 

Some details for the Wiki:

|Vendor|Socket-Type|Device Name|Vendors device ID|GPIOs|Notes|2nd MCU|Flash size/mode
|---------|:--------------:|:-----------:|:---------:|:------------:|:-------------------:|:-----------:|:------------:
|Lyasi|E27|??|??|DI=GPIO4,DCLI=GPIO5|https://www.amazon.de/gp/product/B07KYGDS3P|?|1048576/1M QIO

merlinschumacher, 2019-01-29:
Thanks for your feedback!

closed 2019-01-29 by merlinschumacher

uav2608, 2019-01-30:
> DCKI

How did you change them? Did you update any module? I do have the same bulbs and flashed them but having difficulties with configuration. Thank you in advance.

Eichhoernchen, 2019-01-30:
Please take a look at the issue in the tasmota git: https://github.com/arendst/Sonoff-Tasmota/issues/1822
at the end of the discussion I describe what I did.
I currently ended up writing my own module but I don't think this is the right way to go (from the point of having a maintainable and future proof codebase), but in the meantime these are my changes if you want to apply them locally: https://pastebin.com/0aWxp9ZQ

uav2608, 2019-01-30:
Thank you a lot. I will use your way until the module is added to next version of firmware.

uav2608, 2019-01-30:
> Please take a look at the issue in the tasmota git: [arendst/Sonoff-Tasmota#1822](https://github.com/arendst/Sonoff-Tasmota/issues/1822)
> at the end of the discussion I describe what I did.
> I currently ended up writing my own module but I don't think this is the right way to go (from the point of having a maintainable and future proof codebase), but in the meantime these are my changes if you want to apply them locally: https://pastebin.com/0aWxp9ZQ

is it the only places you made the changes? I do have error in compilation like: 
"sonoff/sonoff_template.h:1814:1: error: too many initializers for 'const mytmplt         {aka const MYTMPLT}'
};
^
"
is it declared somewhere else? Thank you for support.

Eichhoernchen, 2019-01-30:
how did you apply the diff? This is a diff to tasmota 6.4.1 not the current head of the git.
I just tried applying the diff and it works and compiled

uav2608, 2019-01-31:
All works. Thank you, Unfortunately the bulbs become only white. I hope it would be fixed later.

Eichhoernchen, 2019-01-31:
you should be able to change the color through rgb commands

uav2608, 2019-01-31:
Thank you. Will try. Much appreciated your help.

Vendo232, 2019-01-31:
how did you put the bulb in pairing mode?

I turn on/off the bulb 3 times, it starts blinking, I run ./start_flash.sh after 4th~ dot it will turn off and the dots continue, no flashing

Eichhoernchen, 2019-01-31:
much like that, but I had problems initially as well.

Since I installed tuya-convert on my already running raspberry, I had to make sure that my wifi actually handout ips (dhcp). So maybe ensure that the AP that is opened will give you actually an IP (for example on your smartphone). This did the trick for me

Vendo232, 2019-01-31:
all works on my end, just finished flashing all my plugs and it worked.

as a cherry on a cake I though will flash my one and only Tuya bulb ( cheapo 10$ one ) bud since I don`t have manual was not sure how to enter the paiting mode.

I`m glad you confirm that it is to be turned off and on 3 times.

well will keep fighiting

Vendo232, 2019-01-31:
OK setting this as "AiLight" helped

GeorgeIoak, 2019-01-31:
You can try to set it as a generic module and then set all the GPIOs to relays. Then toggle each relay and see if it activates the bulb

DerGuteWolf, 2019-09-07:
@Vendo232 could you elaborate a bit on the 'right' timing to flash this bulb? Thanks!