ct-Open-Source/tuya-convert #49: Gosund WP3 TASMOTA GPIO Configuration

meingraham, 2019-01-29:
For the wiki: Gosund WP3
Generic Module
GPIO4: Switch1
GPIO12: LED2i
GPIO13: LED1i
GPIO14: Relay1

ragenhe, 2019-01-29:
This should work also for WP5, as it is the same plug, with a different number.  Will confirm after work.  Thanks for sharing @meingraham!

meingraham, 2019-01-29:
WP5 is a newer design... and may have different GPIO configuration. Let us know.

If the WP3 configuration doesn't work, @digiblur has a procedure in his latest [YouTube livestream](https://www.youtube.com/watch?v=m_O24tTzv8g) to determine the proper GPIO configuration. Here's how I would summarize that:

- Set every GPIO (excluding, at first,  "dedicated" like Tx/Rx) to relays. Save configuration. Use web GUI toggles to find which GPIO control relays and LEDs. If unable to find relays and LEDs, set the "dedicated" GPIO and retry. Set the "active" GPIO to associate them with the `RelayX` or `LEDX`. Operation may dictate regular or inverted (`RelayXi`/`LEDXi`).
- Set every remaining GPIO (excluding, at first,  "dedicated" like Tx/Rx) to Switches. Save configuration. Display `Status 8` (sensors) in the Console to see the current state of the GPIO. Then, while pressing the button(s), display `Status 8` again. Whichever GPIO changes state is the GPIO tied to the button. If none change state, then configure the "dedicated" GPIO and retry. Change the GPIO to a `Button` (may need to determine whether the internal pull-up is used or not - `ButtonXn` for no pull-up). If you want to use short-press/long-press functionality (described in a couple other of Travis' videos), configure the button as a `Switch` instead.
- Set remaining GPIO to `None`

Mike

ragenhe, 2019-01-29:
I watched that video this morning.  It closes the loop one what I need to do.  While I was "researching" I came across a pdf letter from Gosund to the FCC.  They advised that all of the models were the same and had different model numbers for marketing purposes.  On Amazon the WP5 is "upgraded".  I will report back my findings tonight.
Thanks!
Ragen

merlinschumacher, 2019-01-29:
Thanks for your contribution!

meingraham, 2019-01-29:
@merlinschumacher,

THank you for this project!

I'm not sure if one is able to set a particular page as editable or if the setting has to apply to the entire repository. If the former, you may want to consider making that page editable by the user community to save yourself some work... and possibly to make the page more dynamic/accurate. I saw a couple of comments on other forums about new devices and configurations but they hadn't provided them since they were not able to edit the page and had not thought about how to provide you with the information.

Regards.

Mike

digiblur, 2019-01-29:
@merlinschumacher yes. I fully agree! Or even just a few of us be able to edit things in as I would like to put several configs in there as well.

ragenhe, 2019-01-29:
> For the wiki: Gosund WP3
> Generic Module
> GPIO4: Switch1
> GPIO12: LED2i
> GPIO13: LED1i
> GPIO14: Relay1

Works like a charm on WP5 as well.

dvanderb, 2019-03-12:
> > For the wiki: Gosund WP3
> > Generic Module
> > GPIO4: Switch1
> > GPIO12: LED2i
> > GPIO13: LED1i
> > GPIO14: Relay1
> 
> Works like a charm on WP5 as well.

Did you have any issues/what did you flash to the WP5?  I flashed the default tasmota binary that was included.  I can access the HTTP page for the device and set the GPIO config.. but I never hear the relay click over when turning on and off.

ragenhe, 2019-03-12:
No issues.  I flashed the default as well.  I did change Switch1 to button1
so I could use the button on the side for manual control.

Module Type: Generic (18) right?

dvanderb, 2019-03-13:
Yeah just tried this on a 2nd one with the same issue.  Even set to Button1 but the button isn't causing the Relay to click over either.  Attaching config screenshot just to so someone can say if something looks amiss.
[image]

meingraham, 2019-03-14:
See if this procedure helps identify your configuration issues - https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices

dvanderb, 2019-03-14:
Thanks I figured it out. Looks like not all Gosund WP5 are created equal.  All of mine are from the same batch and I've tried two with the same result.  In my case I had to use the same config as the  Teckin SP10 as referenced in: https://github.com/SynAckFin/TuyOTA/issues/6#issuecomment-455742716

> See if this procedure helps identify your configuration issues - https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices

meingraham, 2019-03-14:
I updated the Tuya-Convert supported devices wiki to add the SP10 and to reference that configuration for the WP5.

pcgeek86, 2020-08-28:
I have 4x of the Gosund WP5 outlets, and I used the following configuration.

* When the socket is ON, the RED LED is on
* When the socket is OFF, no LEDs are on

**Question**: How can I configure the blue LED to turn on? 

[image]

pcgeek86, 2020-08-28:
I ended up using this configuration, which keeps the blue "link" LED active 

[image]