ct-Open-Source/tuya-convert #50: successfully flashed Koogeek KLOE4 Smart Power Strip

callahanharry, 2019-01-29:
Thanks for great tool! I just successfully flashed Koogeek KLOE4 Smart Power Strip. 

This is my GPIO configuration in Tasmota:

gpio4	socket 2
gpio5	socket 1
gpio12	socket 3
gpio13	socket 4
gpio14	usb
gpio3	button
gpio1	power LED

[image]

[image]

merlinschumacher, 2019-01-30:
Thanks for your contribution!

closed 2019-01-30 by merlinschumacher

sgrimee, 2019-03-18:
Worked great for me, but I had to use Led1i (56) instead to have the Led lit when one relay is on, and keep it off when all relays are off.