ct-Open-Source/tuya-convert #51: Are there any success storied using PC with Ubuntu/Mint/Kali?

Vendo232, 2019-01-29:
I have been trying all day long and no success

drushbrook, 2019-01-30:
What is in the smarthack-wifi.log file?

Vendo232, 2019-01-30:
I can not find such file Sir. it is not in Script folder.

jimkernsjr, 2019-01-30:
Hello Vendo232

I had good luck with Ubuntu 16 on my Dell Inspiron 15.  I had to change the config.txt to point to the proper wifi adapter.  (name wlp6s0)  My success was with a TanTan WP2 Dual outlet that had SmartLife app for it's chinesium crap app of choice.

I will not tell you it was without pain.  Of about 6 outlets, this was the only one that actually worked.  Sadly I tried to save the log file but accidentally overwrote it when I tried to flash the next one.  Was able to save the bin however.

To answer your question about the access point not appearing:
I have tried to switch to the PI with great failure.  I tried pi3 with raspbian and internal adapter,no luck.  Same for Kali and external adapter.  What appears to be the failure point is attaching the monitor interface to the wifi interface.  It will sometime start then stop the access point.  The log yields this, or a variant of:
Failed to create interface mon.wlan0: -95 (Operation not supported)
For the Pi with Kali and USB dongle,I blame the problem on the USB dongle driver.
For the PI with Raspbian, I guess I need to try again with a fresh image.  This surprised me.

For the reason your log didnt appear, try checking for it under the name screenlog.0  This will especially be true if you had to edit and remove the -logfile from the start script. 

One more thing - if when you are done, you cannot manage your wifi interface anymore, you may need to undo the mess in /etc/NetworkManager/NetworkManager.conf.  Take the unmanaged devices line out.  See line 14 of setup_ap in scripts folder for the line that it adds - you need to undo that.  This may happen if you control-c out of it and restart it without running the stop script. 

If anyone has any ideas about the monitor interface problem?  
I'll put a full log later...

jimkernsjr, 2019-01-31:
**UPDATE:

If you are having trouble getting the accesspoint vtrust-flash to start, run this command BEFORE you run ./start_flash.sh:

killall wpa_supplicant

worked for me!

Vendo232, 2019-01-31:
I have SUCCESS!

I was able to flash all my plugs and LED bulb using UBUNTU 16.04, old USB wifi adapter and old PC.

- BESTEK MRJ1011
- ISELECTOR SM-PW702
- LED bulb NO NAME, cheapo 10$ from ebay

the timing of the flash is important!
following all steps exactly is important!

what worked for me was renaming WLAN0 from the random name to wlan0

I used this command

$ sudo nano /etc/default/grub
Look for GRUB_CMDLINE_LINUX and add the following net.ifnames=0 biosdevname=0.
From:
 GRUB_CMDLINE_LINUX=""
To:
 GRUB_CMDLINE_LINUX="net.ifnames=0 biosdevname=0"
Generate a new grub file using the following command.
$ sudo grub-mkconfig -o /boot/grub/grub.cfg
Restart PC

Vendo232, 2019-01-31:
btw, thank you Jim for your support ! :-)

kueblc, 2019-01-31:
If this is resolved, you can close the issue.

closed 2019-01-31 by Vendo232