ct-Open-Source/tuya-convert #52: Successfully flashed Zhilde ZLD-44EU-W

fillwe, 2019-01-29:
Successfully flashed my Zhilde ZLD-44EU-W with the espurna bin file for the device. No issues! https://github.com/xoseperez/espurna/wiki/Hardware-Zhilde-ZLD-44EU-W

marcusramberg, 2019-01-29:
Curious, which sonoff module type did you use for it? I just flashed a similar thing (https://www.aliexpress.com/item/WiFi-Smart-Power-Strip-Socket-Voice-Control-Timer-Switch-Power-Strip-Outlet-with-4-AC-Outlets/32915120795.html?spm=a2g0s.9042311.0.0.27424c4dd1njrX), and can't find an appropriate module to allow me to control all 5 (4 power + usb) ports. The closest so far was Manzuko strip. Other than that I flashing worked great :)

marcusramberg, 2019-01-29:
Never mind, found out how to configure it as a generic device from this post https://www.aliexpress.com/item/WiFi-Smart-Power-Strip-Socket-Voice-Control-Timer-Switch-Power-Strip-Outlet-with-4-AC-Outlets/32915120795.html?spm=a2g0s.9042311.0.0.27424c4dobF8IM

digiblur, 2019-01-29:
@marcusramberg

Try this thread for help on figuring out the pins. https://github.com/ct-Open-Source/tuya-convert/issues/42

callahanharry, 2019-01-29:
Zhilde ZLD-44EU-W actually looks exactly like mine KLOE4. See #50 .

merlinschumacher, 2019-01-30:
Thanks for your contribution!

closed 2019-01-30 by merlinschumacher