ct-Open-Source/tuya-convert #53: Successfully flashed "MoesHouse" WiFi Plug

micgus, 2019-01-29:
Heya, 
Million thanks for great development. You guys rock.

Successfully flashed:
https://www.aliexpress.com/item/UK-US-EU-WiFi-Smart-Socket-Power-Plug-Outlet-Remote-Control-Works-with-Amazon-Alexa-Google/32954272490.html 

GPIO12 = Relay
GPIO5= Led, Inverted

tkotz8105, 2019-01-29:
That socket looks similar to the Geekbes I'm working with in the US (https://slickdeals.net/f/10540212-3-pack-geekbes-smart-wifi-mini-plug-26-free-s-h).  Were you able to determine what GPIO the on-device button goes to?  Many thanks for any info you can provide.

micgus, 2019-01-29:
I did indeed, got everything working including the button but I couldn't remember the button configuration.
I'll post a screenshot for ya asap.

Cheers

merlinschumacher, 2019-01-30:
Thanks for your contribution

closed 2019-01-30 by merlinschumacher