ct-Open-Source/tuya-convert #54: Successfully flashed BSD23

pilehave, 2019-01-29:
Vendor: Unknown
Area: EU
Device Name: WiFi Smart Plug
Vendors device ID: BSD23
GPIOs:
GPIO12 = Relay1 (+ blue LED, lights when relay is on)
GPIO4 = Led1 (Red LED, lights when relay is off)
GPIO13 = Button1
Notes: Tasmota Module type: Generic
Flash size/mode: 1M QIO

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 5f3896
MAC: XX:XX:XX:XX:XX:XX
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440a1
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000

merlinschumacher, 2019-01-30:
Thanks for your contribution

closed 2019-01-30 by merlinschumacher

AS137430, 2019-02-28:
The UK version is BSD29.

GPIO12 =  Relay 1
GPIO13 = Button 1
GPIO14 = Led1i

remarkua, 2019-11-02:
Name: Wireless Smart Plug
Model: BSD23 EU

GPIO12 = Relay  (when relay is on, lights Blue Led, Blue Led not connected to other GPIO)
GPIO4 = Led    (when relay is off, lights Red Led, Blue off)
GPIO13 = Button

ChipID: 3ff108
MAC: xxxxxxxxxxxxxx
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000