ct-Open-Source/tuya-convert #55: Possibly bricked an outlet

shreram, 2019-01-29:
I followed all the steps described in this [video](https://www.youtube.com/watch?v=O5GYh470m5k) to flash this firmware to one of my outlets (looks like [this](https://www.amazon.com/dp/B07DPJ7NJ2))

```
sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install network-manager
sudo apt install git
wget https://nodejs.org/dist/v10.15.0/node-v10.15.0-linux-armv6l.tar.xz (to install npm as install_prereq script requires this)
tar -xJf node-v10.15.0-linux-armv6l.tar.xz
pi@raspberrypi:~/node-v10.15.0-linux-armv6l $ sudo cp -R * /usr/local/
git clone https://github.com/ct-Open-Source/tuya-convert
cd tuya-convert
./install_prereq.sh
./start_flash.sh
Connected to 'vtrust-flash' access point
Paired the outlet 
curl http://10.42.42.42/flash3
connected to sonoff-xxxx access point from another device
browsed to 192.168.4.1
scan for wifi networks
selected my router's Wifi access point
At the last step I accidentally forgot to enter password (I thought password was populated in the API Password field) and clicked 'Save' 
```
Now I can't ping to 10.42.42.42. 

I tried all following methods but none worked. 

- I restarted my router (Google Wifi) 
- I tried multiple times to reset the outlet (it's not blinking anymore)
- I executed undo command to force stock firmware.
- I tried to run start_flash command again but that didn't work as I couldn't pair the outlet. 

Addl info:

```
Starting pairing procedure in screen
RTNETLINK answers: File exists
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
.............................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
Create backup of entire FLASH from 10.42.42.42
Connected... Flashsize= 1048576

======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 4ce0c0
MAC: xx:xx:xx:xx:xx:xx
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 1440a1
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000
```

```
pi@raspberrypi:~/tuya-convert/scripts $ uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
pi@raspberrypi:~/tuya-convert/scripts $ npm -v
6.4.1
pi@raspberrypi:~/tuya-convert/scripts $ node -v
v10.15.0
```

I noticed this message when I ran install_prereq.sh script. This message was still shown even after I installed node-v10.15.0-linux-armv6l.tar.xz binary.

> ## You appear to be running on ARMv6 hardware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later.

Is this outlet dead in the water?

kueblc, 2019-01-29:
Sounds like you successfully installed Tasmota but misconfigured it. Try temporarily disabling the password on your router and see if it connects.

digiblur, 2019-01-29:
I did a test with a Wemos D1 mini and the sonoff-classic bin that this repo uses and put in a wrong password and it looks like it just keeps trying, I had it running for about 5 minutes.  I tried removing the SSID and it just keeps trying and trying.  It might not be a bad idea to change the bin this repo pulls down to one that has wifi_manager configured as default as it will try a couple of times then kick wifi manager back in.

shreram, 2019-01-30:
@kueblc I will try your suggestion. 

Is 10.42.42.42 IP always assigned to that outlet?

shreram, 2019-01-30:
> Oh my mistake, it actually does appear to be prepopulated with a value. Looks like its "********"

Thanks for pointing out the source code.  👍 

Yes, it's defined as below. I will try this one as the new password. 😄 

`#define D_ASTERIX "********"`

shreram, 2019-01-30:
@kueblc  This didn't work. I tried following.

- Changed the password on Google Wifi router to ********
- Tried to reset the outlet (pressed side button multiple times) but light didn't blink at all
- Couldn't browse to 192.168.4.1
- All following commands failed
pi@raspberrypi:~ $ curl http://10.42.42.42
curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/undo
curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host

```
MacBook-Air:~ shream$ ping 10.42.42.42
PING 10.42.42.42 (10.42.42.42): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
^C
--- 10.42.42.42 ping statistics ---
4 packets transmitted, 0 packets received, 100.0% packet loss

MacBook-Air:~ shreram$ ping 192.168.4.1
PING 192.168.4.1 (192.168.4.1): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
^C
--- 192.168.4.1 ping statistics ---
5 packets transmitted, 0 packets received, 100.0% packet loss

```
He's dead, Jim.

digiblur, 2019-01-30:
No other access points showing? I updated a couple posts for folks that don't want to use their own compiled bins on ways to try to mitigate this with a secondary password/AP and also on any other devices you have make sure you set wifi manager as the default config option going forward.

shreram, 2019-01-30:
> also on any other devices you have make sure you set wifi manager as the default config option going forward.

> (from Reddit) I would highly recommend changing the WiFi Config tool to WiFi Manager by typing "wificonfig 2" (without the quotes on the console), it should respond back with that WiFiManager is selected

When do I have to enter that command? Later I am going to try flashing another outlet.

- Connect to 'vtrust-flash' access point
- Pair the outlet 
- curl http://10.42.42.42/flash3
- connect to sonoff-xxxx access point from another device
- browse to 192.168.4.1
- scan for wifi networks
- select my router's Wifi access point and enter password and click 'Save'
- wificonfig 2 ??

digiblur, 2019-01-30:
@shreram Pretty good!  If you want the sure fire way you could of course compile your own with Wifi Manager selected or even your own with you default SSID and Password in the binary for full proof.  It's surprisingly, I did a live stream on it a full ground up install of PlatformIO to build Tasmota as I wanted to show people how easy it was without any video editing tricks.  

Now with the prebuilt binaries, yes do the standard part of browsing to 192.168.4.1 and I would put in your stuff for AP1 and Password1.  I would also put in something stupid simple to type in AP2 (even though it isn't broadcasting!).  That way if something ever happens that switch/plug will look for that secondary AP and you could in a pinch setup a wifi network to fix the switch.  

Once you get into the Tasmota GUI and it is on your WiFi network, click console then type wificonfig 2 and it should say something about wifi manager selected.  This will stay like this until you factory default Tasmota.  Since now you have wifi manager selected it won't keep trying a bad SSID/password combo and get stuck, it will time out after a couple tries then setup the AP broadcast.   In the event you are rebooting your wifi network, your devices will go into this AP broadcast mode but they will time out after 180 seconds, reboot and try to connect to your WiFi again.

Jason2866, 2019-01-31:
> I did a test with a Wemos D1 mini and the sonoff-classic bin that this repo uses and put in a wrong password and it looks like it just keeps trying, I had it running for about 5 minutes. I tried [...]

Since a while the included file is sonoff-basic. Wifimanager is default there

digiblur, 2019-01-31:
@Jason2866 something else must be changing it back then.  Steps to reproduce:

1. Connect d1 mini to PC.  
2. Erase flash with esptool
3. Flash sonoff-basic.bin from releases tab on Tasmota github.  
4. Find sonoff-xxxx AP and connect with PC/phone.
5. Put in wrong password for the AP.
6. Watch on a serial monitor and it just kept trying and trying to connect to the wrong WiFi.

I did the same exact test with another bin file with wifi manager as default and had no issue with the sonoff-xxxx AP coming back up after the device tried a couple times to connect and failed.

Jason2866, 2019-01-31:
@digiblur Strange shouldnt be...
Please open a issue on Tasmota github.
Thx!

shreram, 2019-02-25:
Possibly bricked another outlet.

I followed all the steps and this time I entered correct wifi password and also entered backup Wifi credentials in the second access point (AP2).

After that those entries were saved and noticed a message that the device will restart soon.

After that I didn't find that 'sonoff-xxxx' in my Wifi access list or in my connected devices list in Google Wifi router.

Now I couldn't browse to 10.42.42.42 or 192.168.4.1

Any ideas?

```
b'read 0x000FFC00 to 0x000FFFFF'
~/tuya-convert
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 67e5f0
MAC: YY:YY:YY:YY:YY:YY (masked)
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144068
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000

======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.4.1 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flashURL?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!

pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restartpi@raspberrypi:~/tuya-convert $ cd

pi@raspberrypi:~ $ ping 10.42.42.42
PING 10.42.42.42 (10.42.42.42) 56(84) bytes of data.
^C
--- 10.42.42.42 ping statistics ---
8 packets transmitted, 0 received, 100% packet loss, time 7277ms

pi@raspberrypi:~ $ curl http://10.42.42.42/flash3
curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host

pi@raspberrypi:~ $ ping 10.42.42.42
PING 10.42.42.42 (10.42.42.42) 56(84) bytes of data.
From 10.42.42.1 icmp_seq=1 Destination Host Unreachable
From 10.42.42.1 icmp_seq=2 Destination Host Unreachable
From 10.42.42.1 icmp_seq=3 Destination Host Unreachable
From 10.42.42.1 icmp_seq=4 Destination Host Unreachable
From 10.42.42.1 icmp_seq=5 Destination Host Unreachable
From 10.42.42.1 icmp_seq=6 Destination Host Unreachable
^C
--- 10.42.42.42 ping statistics ---
8 packets transmitted, 0 received, +6 errors, 100% packet loss, time 7266ms
pipe 4

pi@raspberrypi:~ $ ping 10.42.42.1
PING 10.42.42.1 (10.42.42.1) 56(84) bytes of data.
64 bytes from 10.42.42.1: icmp_seq=1 ttl=64 time=0.113 ms
64 bytes from 10.42.42.1: icmp_seq=2 ttl=64 time=0.098 ms
64 bytes from 10.42.42.1: icmp_seq=3 ttl=64 time=0.095 ms
64 bytes from 10.42.42.1: icmp_seq=4 ttl=64 time=0.101 ms
64 bytes from 10.42.42.1: icmp_seq=5 ttl=64 time=0.098 ms
^C
--- 10.42.42.1 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4153ms
rtt min/avg/max/mdev = 0.095/0.101/0.113/0.006 ms

pi@raspberrypi:~ $ curl http://10.42.42.42/flash3
curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host

pi@raspberrypi:~ $ curl http://10.42.42.42/undo
curl: (7) Failed to connect to 10.42.42.42 port 80: No route to host

pi@raspberrypi:~ $ ping 10.42.42.1
PING 10.42.42.1 (10.42.42.1) 56(84) bytes of data.
64 bytes from 10.42.42.1: icmp_seq=1 ttl=64 time=0.116 ms
64 bytes from 10.42.42.1: icmp_seq=2 ttl=64 time=0.100 ms
64 bytes from 10.42.42.1: icmp_seq=3 ttl=64 time=0.110 ms
64 bytes from 10.42.42.1: icmp_seq=4 ttl=64 time=0.110 ms
64 bytes from 10.42.42.1: icmp_seq=5 ttl=64 time=0.110 ms
64 bytes from 10.42.42.1: icmp_seq=6 ttl=64 time=0.104 ms
^C
--- 10.42.42.1 ping statistics ---
6 packets transmitted, 6 received, 0% packet loss, time 5214ms
rtt min/avg/max/mdev = 0.100/0.108/0.116/0.009 ms
pi@raspberrypi:~ $ 

```

kueblc, 2019-02-25:
@shreram Please direct Tasmota questions to the Tasmota project https://github.com/arendst/Sonoff-Tasmota. Once you have flashed Tasmota on your device, tuya-convert can not interact with your device as it only handles Tuya OTA flashing. You may close this issue.

Jason2866, 2019-02-25:
@kueblc  There will be probably no help be possible from Tasmota side. 
If we dont know the device we cant help if it doesnt work. 
Just the fact flashing Tasmota to a device means not it will work with.
Flashing Tasmota to a unknow devices inherents the risk to get a brick.

kueblc, 2019-02-25:
@Jason2866 whether or not the Tasmota community wishes to support any given device is irrelevant. The tuya-convert repository is not a support forum for Tasmota.

shreram, 2019-02-25:
@kueblc  Thank you. How do I troubleshoot if WiFi settings were indeed stored on the device?

kueblc, 2019-02-25:
See the [Sonoff-Tasmota wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting) for support troubleshooting Sonoff-Tasmota

closed 2019-02-25 by shreram