ct-Open-Source/tuya-convert #56: Flashing Geekbes YM-WS-1 Plug

tkotz8105, 2019-01-29:
I successfully flashed Tasmota onto a Geekbes plug.  Model is:  YM-WS-1.  I bought it in the US.

In the Tasmota Configuration I used the Generic module and set:

GPIO 14:  Relay1 (21)

I'm having trouble getting the device button to work to turn outlet on/off.

Flash size is 1024kB

merlinschumacher, 2019-01-30:
Thanks for your contribution

closed 2019-01-30 by merlinschumacher

tkotz8105, 2019-01-30:
After some trial and error, I figured out the GPIOs for the LED and Button on the Geekbes YM-WS-1.  The configuration is:
Generic Module
GPIO1: Button1 (17)
GPIO13: Led1i (56)
GPIO14: Relay1 (21)

Please also change area to 'US' and you can remove the note.

Thanks much for developing this capability to flash these Tuya devices.  I'm now able to use these Geekbes sockets with Home Assistant and not have to use the Tuya cloud.  You guys roick!!!!

goluftwaffe, 2020-04-03:
Hi, thank you for this, but on mine the settings made the led flash when off and the led off when the switch was on.

So I changed the D7 GPIO13 led1i to led2(53) and now the led is off when off and on when on.

Hope it can help someone else.
[image]

thelefty, 2020-05-14:
The above templates did not work for me. But I debugged it and found the right the template:

[image]

drewdah, 2023-05-31:
Sorry to necro such an old thread, but I had three of these Geekbes YM-WS-1 laying around that I picked up from Newegg.com in 2017 and was trying to free them from Smart Things for Home Assistant, but I wasn't able to get this to work. I was using a Raspberrry Pi Zero W with the latest Raspbian. my mobile phone connected to the vtrust AP, but it said "An ESP82XX based device connected according to your wifi log" but that timed out waiting for the device to reconnect. I have the logs saved if anyone is around and still interested, maybe Tuya pushed a firmware update that broke this