ct-Open-Source/tuya-convert #57: Prevent "bricks" with wrong passwords or SSIDs

digiblur, 2019-01-29:
I've seen a couple people stating they put in the wrong password or SSID in their plug which caused the plug get stuck looking for this wrong WiFi network.  To fix this issue I would suggest that the sonoff-basic.bin not be used from Tasmota's page and a similar bin be used for this.

Compile the bin file with the sonoff-basic options is fine except in the my_user_config.h the line 

#define WIFI_CONFIG_TOOL       WIFI_RETRY  

Needs to be changed to

#define WIFI_CONFIG_TOOL       WIFI_MANAGER

This will effectively stop people from "bricking" their devices (I use the term loosely as technically it is alive, just looking for the wrong wifi)

digiblur, 2019-01-30:
For the time being....

Be very careful that you put in the correct SSID and Password.  It might not be a bad idea to put in a simple backup SSID and Password in position 2.  If for some reason you mistype the slot 1 password you could always change your WiFi router to send out the simple SSID and password in slot 2 to fix the issue with slot 1.  Also once you get into the switch/plug, I would highly recommend changing the WiFi Config tool to WiFi Manager by type "wificonfig 2" (without the quotes on the console), it should respond back with that WiFiManager is selected.  This will make the device go into access point mode anytime there is an issue with connecting to the wifi.

drushbrook, 2019-01-30:
If it has a button you could always try pressing that in for 40 seconds to reset the config as described on the Tasmota Troubleshooting page

digiblur, 2019-01-30:
Right. That's if the device has a button that just so happens to be on GPIO 0. Lots of devices don't have this luxury unfortunately.

Interloping, 2019-02-01:
Okay ... I seem to be in this situation. I don't think I entered anything wrong. But I have a lot of unusual symbols in my password and I read somewhere that can pose a problem. The device is a smart bulb and has no reset button. Switching on and off x5 is the out-of-box method to enter pairing mode; doesn't seem to do anything. Read somewhere that creating an open network with same SSID could work. Problem is ... how? My router won't let me, and no matter what I do in hostapd it seems to create a password-protected AP. Any ideas?

digiblur, 2019-02-02:
@bartekd123 unfortunately not that I know of, in my testing the the stock firmware will just look for that AP/password combo(s) over and over.

bartekd123, 2019-02-02:
That’s too bad, but thanks for your reply. :)

Interloping, 2019-02-02:
Have you tried creating an open (no password) SSID of the same name for both your primary and your backup? As I discovered, certain password characters just won't work. If it finds an open SSID same name it will connect. You can then find the IP and get into the configuration and try a new password with just alphanumeric characters, and set that password for your access point as well.

seaverd, 2019-02-03:
Interloping,

Thank you so much for your comment.  I performed tuya-convert on a Monoprice Stitch Bulb and mistyped my wifipassword.  Temporarily making my wifi ssid open allowed me to connect and update the password.  I was pulling my hair out!   I agree with digiblur, would love default to be wifi manager.

onebyside, 2019-02-03:
Creating a hotspot on my phone with the suspect wifi ssid and no pass got me back in. TY

bartekd123, 2019-02-03:
Thanks for the suggestion. I tried both the main and the backup SSID without a password and different combinations of passwords but nothing works. I don’t know if it is trying to connect to a WiFi hotspot or if something else went wrong just after I put in my wireless info. I did type the right password when I set it up (I’m 99% certain).

ffidan61, 2019-03-05:
Any ways to put the gosund sp1 into AP Mode? 
I think I mistyped mine :( It's blinking red all the time.

digiblur, 2019-03-09:
Firmware to prevent a bad password "brick" update!   https://github.com/digiblur/Sonoff-Tasmota/raw/development/generic/tasmota-6.4.1.19-generic-wifiman-2.3.0.bin
Download the above bin file to your computer and copy it to your Pi as "thirdparty.bin" in the /tuya-convert/files folder.   You can use an app such as WinSCP to transfer it over.  Overwrite any other thirdparty.bin files that might be in that folder.  Once you flash the Tuya device, you will see the command to copy and paste to send over the thirdparty.bin file instead of the regular sonoff 6.4.1 bin.  If you make a mistake on the SSID or password the device will come back in access point mode after it fails to connect.

dony71, 2019-04-24:
@digiblur, can you make default wifimanager firmware based on sonoff firmware (not sonoff basic) ?
sonoff basic does not have mqtt discovery which has issue connecting to home assistant mqtt server

digiblur, 2019-04-24:
@dony71 that one had discovery but it had that version of tasmota had the mqtt discovery bug.  

I did make one based off 6.5.0.  I'll link the folder in this thread instead of the bin in case it gets updated in the future.

https://github.com/digiblur/Sonoff-Tasmota/tree/development/generic

kueblc, 2019-09-28:
Resolved with #277 :+1:

closed 2019-09-28 by kueblc