ct-Open-Source/tuya-convert #59: Successfully flashed Maxcio RGB LED bulb but Tasmota isn't running

nukleuz80, 2019-01-30:
Hi there,

flash process was successful.
I'm running Sonoff-Tasmota (basic) on my RGB LED bulb.
(MAXCIO RGB LED BULB)

I can't switch/toggle to ON or OFF.

I tried with "Generic" and "Sonoff B1" Module Type.
Played with GPIOs but nothing worked

Any suggestions for this issue?

Another question: how to flash the original firmware back to the bulb?

Device-Info:
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: e325fb
MAC: 68:C6:3A:E3:25:FB
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440a1
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000

Thanks for any help...

Vendo232, 2019-01-31:
same on my end
flashed generic RGB LED from China, I`m running full Tasmota.

Can not figure out how to set GPIOs to turn it ON even

Vendo232, 2019-01-31:
try setting it as "AiLight" it worked on my end

kueblc, 2019-01-31:
> I tried with "Generic" and "Sonoff B1" and other Module Types.
> Played with GPIOs but nothing worked for me.

#42 may help you, @digiblur did a walk through for figuring out pin assignments through trial and error

Vendo232, 2019-02-01:
here is the script for configuration.yaml / HA

- platform: mqtt
  name: 'Wintergartenlicht'
  state_topic: "stat/ledlampe1/RESULT"
  value_template: "{{ value_json.POWER }}"
  command_topic: "cmnd/ledlampe1/POWER"
  payload_on: "ON"
  payload_off: "OFF"
  brightness_state_topic: "tele/ledlampe1/STATE"
  brightness_command_topic: "cmnd/ledlampe1/Dimmer"
  brightness_scale: 100
  brightness_value_template: "{{ value_json.Dimmer }}"
  rgb_command_topic: "cmnd/ledlampe1/Color"
  rgb_command_template: "{{ '%02x%02x%02x00' | format(red, green, blue) }}"
  rgb_state_topic: "tele/ledlampe1/STATE"
  rgb_value_template: "{{(value_json.Channel[1]*2.55)|int}},{{(value_json.Channel[0]*2.55)|int}},{{(value_json.Channel[2]*2.55)|int}}"
  qos: 1
  retain: false
  optimistic: false

nukleuz80, 2019-02-01:
> 
> 
> try setting it as "AiLight" it worked on my end, also color works using commands in Tasmota

AiLight doesn't work for me...

closed 2019-03-05 by kueblc