ct-Open-Source/tuya-convert #60: Bricked light bulb

Spajker7, 2019-01-30:
Hi, I was messing with my tuya light bulb and this project and after a few failed attempts I managed to stop the script while it was running...

Log from smarthack-web.log
```
Listening on port 80
Listening on port 80

URI:/gw.json?a=s.gw.token.get&gwId=02200075dc4f220c2fcb&other={"token":"00000000","region":"EU","tlinkStat":{"configure":"smartconfig","time":6,"source":"ap","path":"broadcast"}}&t=372&v=3.0&sign=c99de4d3461f3b466efd1ba0f1a55a88
Answer s.gw.token.get
[I 190130 10:50:10 web:2162] 200 POST /gw.json?a=s.gw.token.get&gwId=02200075dc4f220c2fcb&other={"token":"00000000","region":"EU","tlinkStat":{"configure":"smartconfig","time":6,"source":"ap","path":"broadcast"}}&t=372&v=3.0&sign=c99de4d3461f3b466efd1ba0f1a55a88 (10.42.42.28) 3.29ms

URI:/gw.json?a=s.gw.dev.pk.active&gwId=02200075dc4f220c2fcb&other={"token":"00000000"}&t=7&v=3.0&sign=c118cf20102a5f8332b4d3ace55b47e7
Answer s.gw.dev.pk.active
READ GW ID 02200075dc4f220c2fcb
TRIGGER UPGRADE IN 10 SECONDS
[I 190130 10:50:10 web:2162] 200 POST /gw.json?a=s.gw.dev.pk.active&gwId=02200075dc4f220c2fcb&other={"token":"00000000"}&t=7&v=3.0&sign=c118cf20102a5f8332b4d3ace55b47e7 (10.42.42.28) 8.43ms
Trigger upgrade in 10 seconds

URI:/gw.json?a=s.gw.update&gwId=02200075dc4f220c2fcb&t=9&v=2.0&sign=a60943c2c321ea7665207279b6a01925
Answer s.gw.update
[I 190130 10:50:13 web:2162] 200 POST /gw.json?a=s.gw.update&gwId=02200075dc4f220c2fcb&t=9&v=2.0&sign=a60943c2c321ea7665207279b6a01925 (10.42.42.28) 6.36ms

URI:/gw.json?a=atop.online.debug.log&gwId=02200075dc4f220c2fcb&t=7&sign=96197508b3e60f9247b28388107ca6b1
Answer atop.online.debug.log
[I 190130 10:50:14 web:2162] 200 POST /gw.json?a=atop.online.debug.log&gwId=02200075dc4f220c2fcb&t=7&sign=96197508b3e60f9247b28388107ca6b1 (10.42.42.28) 6.89ms
80db9143c2545851906b78fc76881a36
b'2.1c2545851906b78fct5inCMDHpcJKQpxfOe2RbXd/UiyzGDknxQnzqPqY7WDxwxjl9xf/QJTcIYRl2uIkJsm5uMafEvnzLLz/1MQ4vzedtTfWe7urVb6lQlYZk0Xdg4SXZL9kLEC9aqKXAOwI'

URI:/gw.json?a=s.gw.upgrade&gwId=02200075dc4f220c2fcb&t=14&sign=3b2c9f635db708460e020f7c94c59da7
Answer s.gw.upgrade
b'{"result":{"auto":3,"fileSize":"508475","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"88eaa4b09dc5a493a77adf0a41c87ba0"},"t":100,"e":false,"success":true}'
[I 190130 10:50:21 web:2162] 200 POST /gw.json?a=s.gw.upgrade&gwId=02200075dc4f220c2fcb&t=14&sign=3b2c9f635db708460e020f7c94c59da7 (10.42.42.28) 32.80ms

URI:/gw.json?a=s.gw.upgrade.updatestatus&gwId=02200075dc4f220c2fcb&t=100&sign=c0c7f5c0fb3e10b9406a861be62eace1
Answer s.gw.upgrade
b'{"result":{"auto":3,"fileSize":"508475","etag":"0000000000","version":"9.0.0","url":"http://10.42.42.1/files/1.bin","md5":"88eaa4b09dc5a493a77adf0a41c87ba0"},"t":100,"e":false,"success":true}'
[I 190130 10:50:21 web:2162] 200 POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=02200075dc4f220c2fcb&t=100&sign=c0c7f5c0fb3e10b9406a861be62eace1 (10.42.42.28) 7.36ms
[I 190130 10:50:28 web:2162] 200 GET /files/1.bin (10.42.42.28) 4871.93ms
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 132, in <module>
    main()
  File "./fake-registration-server.py", line 128, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 132, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt
```

The bulb isn't giving any life signs. Is there any way I can unbrick it?

uav2608, 2019-01-30:
Did you check your phone wifi settings? You may have new SSID sonoff-XXXX if it flashed. What kind of light bulb is it?

digiblur, 2019-01-30:
Yes! The base firmware is waiting for you.

Spajker7, 2019-01-31:
Whenever I try to flash using the provided web server, by doing `sudo ./scripts/fake_registration_server.py`, I keep getting 3 404 errors in the output, like it can't find the files/thirdparty.bin. 

Any ideas?

Spajker7, 2019-01-31:
I switched back to vtrust-flash ap hosted on a raspberry pi instead of the provided vtrust-recovery and successfully flashed the image.

closed 2019-01-31 by Spajker7