ct-Open-Source/tuya-convert #61: Script over-wrote my existing hostapd config

Interloping, 2019-01-31:
Just tried tuya-convert yesterday. It overwrote my existing hostapd config and appears not to have backed it up. Was able to restore. Still, might be good to add a backup process like for NetworkManager.conf.

jimkernsjr, 2019-01-31:
Even the backup process for NetworkManager.conf is not flawless - I lost mine.

I really think that's why they suggest to run that on an empty image.  Alternatively, backup your config!  You can use webmin to do it.

ctandi, 2019-02-12:
Hi @Interloping ,

[jimkernsjr](/jimkernsjr) has got it right. It would be a lot of work for us to make sure that our scripts leave everything as before in every use case. Since swapping SD cards is easy with the Raspberry Pi, we decided against doing that. Of course, we are always open for other solutions. Please feel free to contribute :)

Cheers,
Andi

closed 2019-02-12 by ctandi

schildbach, 2019-09-28:
Just wanted to add that I lost my original NetworkManager.conf too because the script overwrites the backup with another backup (which is then the modified version).

kueblc, 2019-09-28:
This should be resolved with #225 if you'd like to help test we can get this merged quickly