ct-Open-Source/tuya-convert #63: Source code for esp8266-ota-flash-convert

jalmeroth, 2019-01-31:
I am looking for the source code of the esp8266-ota-flash-convert used in files `esp8266-ota-flash-convert_upg.bin` and `esp8266-ota-flash-convert.ino-0x81000.bin`.

As this project aims to liberate IoT devices from proprietary firmware, I would rather not flash some other binary blobs to my devices.

Thank you,
Jan

digiblur, 2019-01-31:
Aren't you overwriting them afterwards?

jalmeroth, 2019-01-31:
I don't think so. AFAIK it stays in user2, that's why you can do a roll-back. But even if, it handles the image transfer, so it could alter it or do whatever.

digiblur, 2019-01-31:
Nuke it. 😏

jalmeroth, 2019-01-31:
> Nuke it. 😏

That's not the point of this issue. ;)

kueblc, 2019-01-31:
> I don't think so. AFAIK it stays in user2, that's why you can do a roll-back. But even if, it handles the image transfer, so it could alter it or do whatever.

You can only roll back 1.bin, the first step. Once you've flashed a new firmware that's it, you're only running the new firmware, and you can't (easily) go back to the original firmware.

kueblc, 2019-01-31:
That aside, I do agree for the sake of open source, it would be great to have all binary sources available.

cscheib, 2019-02-03:
I also would appreciate the source.  The way this project is laid out and documented can be a little tough to understand, and I'd like to either improve the documentation via PR or rewrite this whole thing so it feels a little more intuitive.  Being able to see what's happening on the device side in code would be quite helpful to either scenario.

cscheib, 2019-02-03:
I think I found it: https://github.com/vtrust-de/esp8266-ota-flash-convert

@jalmeroth

jalmeroth, 2019-02-03:
Thanks for pointing out, @cscheib! I did a search but couldn't find it. ¯\_(ツ)_/¯

closed 2019-02-03 by jalmeroth