ct-Open-Source/tuya-convert #64: vtrust-flash: Failed to obtain IP address

loglux, 2019-01-31:


loglux, 2019-01-31:
Cannot login into vtrust-flash AP, getting an error 'Failed to obtain IP address'

kueblc, 2019-01-31:
Please post the contents of `scripts/smarthack-wifi.log`

loglux, 2019-01-31:
`Backing up NetworkManager.cfg...
Restarting NetworkManager...
wlan0: ERROR while getting interface flags: No such device
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
wlan0: ERROR while getting interface flags: No such device
SIOCSIFADDR: No such device
wlan0: ERROR while getting interface flags: No such device
SIOCSIFNETMASK: No such device
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
Cannot find device "wlan0"
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
Backing up NetworkManager.cfg...
Restarting NetworkManager...
wlan0: ERROR while getting interface flags: No such device
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
wlan0: ERROR while getting interface flags: No such device
SIOCSIFADDR: No such device
wlan0: ERROR while getting interface flags: No such device
SIOCSIFNETMASK: No such device
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
Cannot find device "wlan0"
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started
Backing up NetworkManager.cfg...
Restarting NetworkManager...
wlan0: ERROR while getting interface flags: No such device
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
wlan0: ERROR while getting interface flags: No such device
SIOCSIFADDR: No such device
wlan0: ERROR while getting interface flags: No such device
SIOCSIFNETMASK: No such device
Applying iptables rules...
Starting DNSMASQ server...

dnsmasq: failed to create listening socket for port 53: Address already in use
[... 766 lines cut ...]
wlan0: STA 60:01:94:c0:42:32 RADIUS: starting accounting session B69FE939D64DEF64
wlan0: STA 60:01:94:c0:42:32 WPA: pairwise key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED 60:01:94:c0:42:32
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: disassociated
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: authenticated
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: associated (aid 1)
wlan0: AP-STA-CONNECTED 60:01:94:c0:42:32
wlan0: STA 60:01:94:c0:42:32 RADIUS: starting accounting session 6BA9070E353061B7
wlan0: STA 60:01:94:c0:42:32 WPA: pairwise key handshake completed (RSN)
wlan0: STA 80:86:f2:26:d4:19 WPA: group key handshake completed (RSN)
wlan0: STA 60:01:94:c0:42:32 WPA: group key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED 60:01:94:c0:42:32
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: disassociated
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: authenticated
wlan0: STA 60:01:94:c0:42:32 IEEE 802.11: associated (aid 1)
wlan0: AP-STA-CONNECTED 60:01:94:c0:42:32
wlan0: STA 60:01:94:c0:42:32 RADIUS: starting accounting session F0AB5F9B51DA31FA
wlan0: STA 60:01:94:c0:42:32 WPA: pairwise key handshake completed (RSN)
wlan0: AP-STA-DISCONNECTED 60:01:94:c0:42:32
wlan0: AP-STA-DISCONNECTED 80:86:f2:26:d4:19
Failed to set beacon parameters
wlan0: INTERFACE-DISABLED 
rfkill: WLAN soft blocked`

kueblc, 2019-01-31:
From what I can quickly glean, `wlan0: ERROR while getting interface flags: No such device` looks like you don't have an interface called `wlan0`. Run `iwconfig` to tell you what your available interfaces are.

kueblc, 2019-02-01:
OK next up would be `dnsmasq: failed to create listening socket for port 53: Address already in use`. Do you have anything running on UDP port 53? `sudo netstat -ulp --numeric-ports`

Interloping, 2019-02-01:
Using ubuntu? I had the same dnsmasq error for socket 53 and had to do:
`sudo systemctl start systemd-resolved.service`
before running the start_flash.sh
Then my phone could get an IP.

loglux, 2019-02-01:
Thanks a lot for sudo systemctl stop systemd-resolved.service.
It's helped. 
But I don't see a configuration option for Domoticz in the Menu - Configuration.
Where it's gone?

loglux, 2019-02-02:
Thanks a lot! It worked!

kueblc, 2019-02-02:
Great! If this issue is resolved you may close it. Thanks for your feedback!

closed 2019-03-05 by kueblc