ct-Open-Source/tuya-convert #66: Successfully flashed Brilliant Smart Wifi Plug (Bunnings AU)

ideasman69, 2019-02-01:
Thanks! Just flashed two of these Brilliant model 20676/05 smart plugs:

https://www.bunnings.com.au/brilliant-smart-wifi-plug-and-usb-charger_p0091644

And used these settings:

[image]

Interloping, 2019-02-01:
That looks almost identical to the kmart mirabella genio smart plug (https://www.kmart.com.au/product/mirabella-genio-wi-fi-1-usb-adaptor/2326810) - but I just tried that with your settings with no luck. Can you possibly explain how you worked out to use those particular settings?

ideasman69, 2019-02-01:
Trial and error. I assigned relay1 to each of them one by one until it responded to the on/off command. 

Then same thing for the LED - just assigned led1 to each of them one by one until it lit up when on.

The button was tricky - i tried button1 on each of the left overs and it seemed to work on GPIO14 but was very intermittent. So started to try each of the button options on that GPIO but the second one down the list (button1n) worked fine.

oe3osb, 2019-02-01:
> Good to know. I was going to try that for the mirabella switch but then I
> found this:
> https://github.com/arendst/Sonoff-Tasmota/wiki/Mirabella-Genio-Smart-Plug

Flashed a plug successful, was in the sonoff basic settings and defined the
wlan settings.

Restart

Now i can ping the device, nothing more is possible. No access anymore. The
switch on the plug does not change the color, even i hold it for 40 secs.
It starts to flash but no differents. The switch also does not switch.

Tks. Erich

closed 2019-03-23 by kueblc

kezakez, 2019-03-27:
For anyone else trying this device, be sure to flash the device straight away, if you use the 'brilliant' app it will load new firmware which when in pairing mode will not connect to tuya convert linux proxy.
I am not sure, but I think the new firmware wants to connect over https rather than http.

kueblc, 2019-03-27:
Thanks @kezakez, we have been aware of an update rolling out that patches this. If you have managed to capture your network traffic and/or a firmware backup, your data would assist us tremendously in analyzing the changes and hopefully developing a workaround.

kezakez, 2019-04-01:
I can't get the new firmware backup as it doesn't connect to do the backup.
Here is a section of a tshark capture. Let me know if there is a better way to capture what you are after. 
```
 1667 73.604404228   10.42.42.1 → 255.255.255.255 UDP 413 59092 → 30011 Len=371
 1668 73.609552816   10.42.42.1 → 255.255.255.255 UDP 402 59092 → 30011 Len=360
 1669 73.614700571   10.42.42.1 → 255.255.255.255 UDP 298 59092 → 30011 Len=256
 1670 73.619845879   10.42.42.1 → 255.255.255.255 UDP 298 59092 → 30011 Len=256
 1671 73.624992957   10.42.42.1 → 255.255.255.255 UDP 298 59092 → 30011 Len=256
 1672 73.723083880 Espressi_b7:85:92 → Broadcast    XID 20 Basic Format; Type 1 LLC (Class I LLC); Window Size 0
 1673 73.726441642 Raspberr_13:24:1c → Espressi_b7:85:92 EAPOL 113 Key (Message 1 of 4)
 1674 73.733038265 Espressi_b7:85:92 → Raspberr_13:24:1c EAPOL 135 Key (Message 2 of 4)
 1675 73.734139031 Raspberr_13:24:1c → Espressi_b7:85:92 EAPOL 169 Key (Message 3 of 4)
 1676 73.746934282 Espressi_b7:85:92 → Raspberr_13:24:1c EAPOL 113 Key (Message 4 of 4)
 1677 73.753136971      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Discover - Transaction ID 0xefa86ec0
 1678 73.753644857 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.14? Tell 10.42.42.1
 1679 73.849781364 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1680 74.779967163 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.14? Tell 10.42.42.1
 1681 74.859940393 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1682 75.164751380 fe80::1003:cf99:21c8:f76f → ff02::1:ff66:8ff4 ICMPv6 86 Neighbor Solicitation for fe80::10c6:68a2:c066:8ff4 from d8:8f:76:a7:ab:fa
 1683 75.330221977      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Discover - Transaction ID 0xefa86ec0
 1684 75.819966047 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.14? Tell 10.42.42.1
 1685 75.899938230 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1686 76.212857391 fe80::1003:cf99:21c8:f76f → ff02::1:ff66:8ff4 ICMPv6 86 Neighbor Solicitation for fe80::10c6:68a2:c066:8ff4 from d8:8f:76:a7:ab:fa
 1687 76.757875768   10.42.42.1 → 10.42.42.14  ICMP 62 Echo (ping) request  id=0x38f6, seq=0/0, ttl=64
 1688 76.758077110   10.42.42.1 → 10.42.42.14  DHCP 342 DHCP Offer    - Transaction ID 0xefa86ec0
 1689 76.758573121   10.42.42.1 → 10.42.42.14  DHCP 342 DHCP Offer    - Transaction ID 0xefa86ec0
 1690 76.889750560 fe80::1003:cf99:21c8:f76f → ff02::1:ff66:8ff4 ICMPv6 86 Neighbor Solicitation for fe80::10c6:68a2:c066:8ff4 from d8:8f:76:a7:ab:fa
 1691 76.919307947      0.0.0.0 → 255.255.255.255 DHCP 350 DHCP Request  - Transaction ID 0xd6663183
 1692 76.928027093   10.42.42.1 → 10.42.42.14  DHCP 342 DHCP ACK      - Transaction ID 0xd6663183
 1693 76.931225020 Espressi_b7:85:92 → Broadcast    ARP 42 Who has 10.42.42.14? Tell 0.0.0.0
 1694 77.333255249 Espressi_b7:85:92 → Broadcast    ARP 42 Who has 10.42.42.14? Tell 0.0.0.0
 1695 77.488864091 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1696 77.833424124 Espressi_b7:85:92 → Broadcast    ARP 42 Gratuitous ARP for 10.42.42.14 (Request)
 1697 78.539956362 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1698 78.807721713 Espressi_b7:85:92 → Broadcast    ARP 42 Who has 10.42.42.1? Tell 10.42.42.14
 1699 78.807792177 Raspberr_13:24:1c → Espressi_b7:85:92 ARP 42 10.42.42.1 is at b8:27:eb:13:24:1c
 1700 78.810012664  10.42.42.14 → 10.42.42.1   DNS 73 Standard query 0x7ab6 A a3.tuyaus.com
 1701 78.810356653   10.42.42.1 → 10.42.42.14  DNS 89 Standard query response 0x7ab6 A a3.tuyaus.com A 10.42.42.1
 1702 78.815954014  10.42.42.14 → 10.42.42.1   TCP 58 31704 → 443 [SYN] Seq=0 Win=4380 Len=0 MSS=1460
 1703 78.816041873   10.42.42.1 → 10.42.42.14  TCP 54 443 → 31704 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 1704 79.579953986 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1705 80.832845418 Espressi_b7:85:92 → Broadcast    ARP 42 Gratuitous ARP for 10.42.42.14 (Request)
-- iphone traffic was here --
 1712 81.128077500 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1713 81.944478047  10.42.42.14 → 10.42.42.1   DNS 73 Standard query 0x9fbb A a3.tuyaus.com
 1714 81.944805631   10.42.42.1 → 10.42.42.14  DNS 89 Standard query response 0x9fbb A a3.tuyaus.com A 10.42.42.1
 1715 81.951227157  10.42.42.14 → 10.42.42.1   TCP 58 21374 → 443 [SYN] Seq=0 Win=4380 Len=0 MSS=1460
 1716 81.951327568   10.42.42.1 → 10.42.42.14  TCP 54 443 → 21374 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 1717 81.979937607 Raspberr_13:24:1c → Espressi_b7:85:92 ARP 42 Who has 10.42.42.14? Tell 10.42.42.1
 1718 81.986737912 Espressi_b7:85:92 → Raspberr_13:24:1c ARP 42 10.42.42.14 is at 60:01:94:b7:85:92
 1719 82.139940757 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1720 83.179954322 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1721 84.766723911 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1722 85.074816621  10.42.42.14 → 10.42.42.1   DNS 73 Standard query 0x3b37 A a3.tuyaus.com
 1723 85.075149412   10.42.42.1 → 10.42.42.14  DNS 89 Standard query response 0x3b37 A a3.tuyaus.com A 10.42.42.1
 1724 85.081671036  10.42.42.14 → 10.42.42.1   TCP 58 6337 → 443 [SYN] Seq=0 Win=4380 Len=0 MSS=1460
 1725 85.081766499   10.42.42.1 → 10.42.42.14  TCP 54 443 → 6337 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
[... cut ...]
 1801 116.373330882   10.42.42.1 → 10.42.42.14  DNS 89 Standard query response 0x3c75 A a3.tuyaus.com A 10.42.42.1
 1802 116.379660266  10.42.42.14 → 10.42.42.1   TCP 58 8360 → 443 [SYN] Seq=0 Win=4380 Len=0 MSS=1460
 1803 116.379754427   10.42.42.1 → 10.42.42.14  TCP 54 443 → 8360 [RST, ACK] Seq=1 Ack=1 Win=0 Len=0
 1804 117.516596596 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
 1805 118.539957629 Raspberr_13:24:1c → Broadcast    ARP 42 Who has 10.42.42.42? Tell 10.42.42.1
```

kueblc, 2019-04-01:
You're absolutely right, this capture reveals that this firmware version is demanding an HTTPS connection, and thus will not work with `tuya-convert` in its current state.

If you have a moment and would like to help out, please trying recording some `pcap`s while pairing the device to the `vtrust-flash` AP using the app, and then using a few commands.

If you manage to open the device to flash it directly the old fashion way, be sure to grab a backup before overwriting.

You can send these to my gmail by the same name if you prefer not to share publicly.

Thank you very much for the data and anything else you can collect!

markdueck1, 2019-04-10:
Hey, I've got one of these things and the firmware has been updated so it won't connect. How can I help? What kind of capture do you need? I'd love to get this working.
I've captured it like this:

 sudo tcpdump -i wlan0 > capture.txt

[capture.txt](https://github.com/ct-Open-Source/tuya-convert/files/3063320/capture.txt)

vk1ajm, 2019-04-10:
I managed to brick mine (Brilliant Smart WiFI Plug from Bunnings) so had to connect wires to the GPIO TX RX and Reset lines to load firmware using a FTDI USB to ttl converter. Finally loaded the Sonoff_Basic firmware  using esptool.py .  Using slightly different template settings.  LED1(WiFi status) is on GPIO4, LED2(Relay status) is on GPIO5, Relay is on GPIO12 and switch is on GPIO14
[image]

[image]

kueblc, 2019-04-10:
Hi @markdueck1 thanks for sharing. We're working on this. At this point we have enough network capture data, I would still happily accept any firmware backups though. For this you would need to connect manually as @vk1ajm did.

@vk1ajm did you manage to retrieve a backup of the stock firmware while connected?

vk1ajm, 2019-04-11:
I have the firmware backup that was made as part of the tuya-convert upgrade process.  I can post it if you want it.  I also have another device that has yet to be powered up.  I am looking at flashing it at the weekend.

Cheers
JMCI

markdueck1, 2019-04-12:
@vk1ajm Would you mind posting a picture of how you connected your wires up? Sorry if this is off topic, I'd love to be able to flash this thing.

vk1ajm, 2019-04-12:
Picture showing setup for flashing. Yellow and white wires are connected to the RX and TX for the TYWE2S board where it mates to the main board.  (TX from the converter goes to RX on the TYWE2S etc) The small white wire is connected to the pad marked IO0 on the blue PCB.  Ground this while powering up  forces the board into flash mode.  The small red board is a USB to TTL adapter that converts the USB serial data to serial for 5V or 3.3V devices. Switch is set to 3.3V for the TYWE2S board (DO NOT connect 5V TTL outputs to the TYWE2S as it is a 3.3V device) .  The grey cable connected to the USB port on the device provides 5Volts to power the device.  DO NOT POWER the device from mains when you have it in pieces.  The grey cable is 2 USB cables with the small ends cut off and the power wires spliced together so that the USB port on the computer (in my case a Raspberry Pi ) can power the device.  

[image]

scompton93, 2019-04-13:
@kezakez Would just to clarify that the update is optional not forced when setting up. I set up mine just a couple of days ago and elected to not upgrade the firmware on the several nag screens it showed me and was able to successfully use tuya-convert today.

dtmeldrum, 2019-05-02:
@ideasman69 Brilliant! Confirmed working with these settings. I had to muck around with creating a new template first as only the parameters below first appeared.

[image]

hipsi, 2019-07-20:
I couldn't get the AP to work properly on my rpi B so gave up on that and opened one up. I made the same connections as pictured by vk1ajm, but on action. For my sanity I tried on a second switch, same result. The esptool can't connect. I've tried 2 different ftdi cables, I don't get it. It seems so easy.

yahms, 2019-10-17:
Also successfully flashed, but a custom version of this:

https://github.com/Gruppio/Sonoff-Homekit

So far created a mostly working homekit/webhook capable smart switch with Au certification yay

Relay gpio 5
Led(red I think) gpio 12
Haven’t worked out the switch. Tried 14 but it’s not working, need to hunt

tonysprenk, 2019-10-17:
@yahms mind sharing your custom version? Been looking to get this working on plugs and switches...

yahms, 2019-10-18:
@tonysprenk of course; once I get a moment I’ll properly fork into its own project and share what I have (compiled binaries + code)

yahms, 2019-10-23:
@tonysprenk https://github.com/yahms/Homekit-BrilliantPlug/blob/master/README.md

its rough, but works. mostly due to the GPIO14 weirdness. to compile manually you need to follow the below instructions to have a docker image with the sdk, then put the sonoff_advanced dir into the /examples dir in esp-homekit-demo, and compile in there (its how gruppio did it i guess, i kept that)

https://www.studiopieters.nl/esp-homekit-sdk-full-installation/

ideasman69, 2021-02-23:
> Picture showing setup for flashing. Yellow and white wires are connected to the RX and TX for the TYWE2S board where it mates to the main board. (TX from the converter goes to RX on the TYWE2S etc) [...]

Hi mate - massive thank you for your details and picture. I bricked two of my devices last night whilst trying to update without following the migration path.

Soldered in the 3 wires, flashed and had them back up and running in no time. Thanks again!

vk1ajm, 2021-02-24:
Good to hear that you found my post useful.

The other modification I have done on the smart switches is to rewire the USB connection .  I have removed the pull up and pull down resistors where the USB connector attaches to the  main PCB.  I have then connected the pin next to the ground pin to GPIO13 and the pin next to the +5V pin to 3.3v.  I then use modified USB cables to connect sensors (in my temperature sensors or motion sensors).

Regards
John

plambrechtsen, 2021-04-03:
I had the same problem while trying to flash with tuya convert and seemed to brick it, so then I managed to flash it using Tasmotizer after soldering the pins directly on to board and using the 5V rail to connect to my TTL connector. And then shorting the IO0 to ground when plugging the TTL serial adapter into my computer. Attached is the backup of the bricked firmware if that is helpful at all.
I also have a spare which I have connected up to the Tuya cloud which I can take a backup of if that would help too.
[backup_20210404_000234.zip](https://github.com/ct-Open-Source/tuya-convert/files/6252466/backup_20210404_000234.zip)

Barmooka, 2022-04-11:
Wondering if anyone can point me to a copy of the Brilliant factory firmware for the smart plugs?

I upgraded all of mine to Tasmota a few years ago and am about to sell them off and need to restore the original firmware. I have looked high and low and cant find my backup.

Thanks, Anthony

RyanCarrier, 2022-06-19:
How on earth are you guys even getting the device open?!

hipsi, 2022-06-19:
Lol yeh it's a bit of a struggle. I used a tiny jeweller's screwdriver to poke and pry the screw caps off.

RyanCarrier, 2022-06-20:
Ah yep, got it off this morning, mine is the 20925, no screw caps. Used a knife to break the plastic cement and managed to pry from there :)

roldengarm, 2023-08-05:
In case someone runs into this topic by Googling, it's possible to flash ESPHome/LibreTiny using Cloudcutter. No soldering or opening required.
Just pick "Flash by firmware version" and select this version: oem-bk7231n-plug-1.1.8

Then, flash ESPHome Kickstarter, and connect it to your WiFi.

After that I extracted the ESPHome YAML via ltchiptool, the important part to get the switch to work is:

```
button:
- platform: restart
  name: Restart

# debug:
#   update_interval: 30s

# text_sensor:
# - platform: debug
#   reset_reason:
#     name: Reset Reason
# - platform: libretiny
#   version:
#     name: LibreTiny Version

# sensor:
# - platform: uptime
#   name: Uptime

binary_sensor:
- platform: gpio
  id: binary_switch_1
  pin:
    number: P26
    inverted: true
    mode: INPUT_PULLUP
  on_press:
    then:
    - switch.toggle: switch_1

switch:
- platform: gpio
  id: switch_1
  name: Relay 1
  pin: P6

status_led:
  pin: P24
```

snzsnz, 2024-08-16:
> In case someone runs into this topic by Googling, it's possible to flash ESPHome/LibreTiny using Cloudcutter. No soldering or opening required. Just pick "Flash by firmware version" and select this version: oem-bk7231n-plug-1.1.8
> 
> Then, flash ESPHome Kickstarter, and connect it to your WiFi.
> 
> After that I extracted the ESPHome YAML via ltchiptool, the important part to get the switch to work is:
> 
> ```
> esphome:
>   name: brilliant-smart-plug-1
>   friendly_name: Brilliant Smart Plug 1
> 
> libretiny:
>   board: generic-bk7231n-qfn32-tuya
>   framework:
>     version: dev
> 
> #... add e.g. HA config, WiFi config, etc.
> 
> button:
> - platform: restart
>   name: Restart
> 
> binary_sensor:
> - platform: gpio
>   id: binary_switch_1
>   pin:
>     number: P26
>     inverted: true
>     mode: INPUT_PULLUP
>   on_press:
>     then:
>     - switch.toggle: switch_1
> 
> switch:
> - platform: gpio
>   id: switch_1
>   name: Relay 1
>   pin: P6
> 
> status_led:
>   pin: P24
> ```

Yeah this didn't work for me. 
Tuya Convert says exploit was not successful. 
Tried a handful of other profiles also.

https://www.pbtech.co.nz/product/ADPBRS0002/Brilliant-Smart-Smart-WiFi-Wall-Plug-with-1-USB-Ch

Model: 20676/05

Using the above.

[image]
[image]
[image]

Will give USB TTY a crack tomorrow