ct-Open-Source/tuya-convert #68: almost there :)

huenemeca, 2019-02-02:
Hi, I am in the middle of the upgrade.  My device now puts out a vtrust-recovery wifi.  When I connect to it and go to 10.42.42.42 I get this.
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 62eaba
MAC: 5C:CF:7F:62:EA:BA
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440e0
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user1 0x01000

When I go to 10.42.42.42/flash2 I get 
Device should flash http://10.42.42.1/files/user2.binto userspace 0x81000 and restart

From the device hosting the vtrust-flash AP.  When I run **curl http://10.42.42.42** I get Failed to connect ... no route to host.

Anything that I can do?

huenemeca, 2019-02-02:
No advice?  Is there anything I can do to get this device working?  

**system_upgrade_userbin_check: user1 0x01000**
Does this mean I am out of luck?

huenemeca, 2019-02-03:
No luck, when I run curl http://10.42.42.42/flash2.  The unit downloads the file from the webserver, I can see that in the logs.  But then the unit just stops responding.  I left it for a while but still nothing.

euro910, 2019-02-11:
Hi
seems that i have the same issue

i could successfully flash 2 Blitzwolf with following settings
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: f89214
MAC: BC:DD:C2:F8:92:14
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440a1
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user2 0x81000

but these Teckin SP22 will not work.
Flash started, but the never come up with wifi, nor will pressing button do anything (1x;4;press and hold)

Teckin SP22
Getting Info from IoT-device
VTRUST-FLASH 1.0
(c) VTRUST GMBH https://www.vtrust.de/35c3/

READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: c1b355
MAC: 60:01:94:C1:B3:55
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000
-----------

i see difference in flash mode, but after reading the c´t article 
"Das enthaltene Tasmota-Image nutzt den DOUT-Flashmodus der auf allen Geräten funktionieren sollte"

even the Blitzwolf (QIO) should have problems, not the Teckin (DOUT)  

the device is evrytime in leraning mode (without LED blinking)
whenever i started the start_flash Skript, it connects to the SP22

br stefan

kueblc, 2019-03-23:
Please update and try again, if you're still having issues, open a new issue with your device details and logs.

closed 2019-03-23 by kueblc