ct-Open-Source/tuya-convert #72: Incompatible Devices (Bricked)

demtat, 2019-02-02:
I love this hack. thank you so much for developing it.  However it would be handy if you could add a signature detector to run after the backup, that could warn us for know incompatible devices.  

I understand the warnings and all and I take full responsibility for the 3 devices that I just bricked.  They were all glued case devices that I hadn't been willing to pry open. Two of them were different styles but both were based on the TYWE2S module.  One of them (https://www.amazon.com/gp/product/B07CXL9ZRG) now constantly toggles one of the 4 relays once per second, while the other (https://www.amazon.com/gp/product/B07B4CNJKG) appears totally dead (I am going to see if I can resuscitate both of these with soldered connections).  The third is a dual plug (https://www.amazon.com/gp/product/B076HKHSSX) with a module labeled NX-SP201 and it appears totally dead as well (I can't find a pinout for this one, so it is probably really dead).

I have had great success with TYWE3S modules.

kueblc, 2019-02-02:
I had a similar experience when I flashed my power strip, where one of the relays toggled really fast for a minute or so. It might not be bricked, don't give up on it just yet!

After that I was able to continue on to configuration by connecting to the sonoff-* AP. Configuration should be similar to #50 and #71

Interloping, 2019-02-03:
I had devices that have seemed dead until the correct module was set in the tasmota config. Are they connecting to wifi, and can you access their web page?

tube0013, 2019-02-05:
I have a dual plug that also has the NX-SP201 marking - https://www.amazon.com/gp/product/B07F6X4KX3/ref=ppx_yo_dt_b_asin_title_o00__o00_s00 looks the same as the one you posted.  I got both flashed with tuya convert, but one bricked while adjusting module settings in tasmota.  working cautiously with the second one from the dual pack.

mrjvb, 2019-02-06:
Two smartplugs https://www.amazon.de/JinvooSmart-packungen-mit-fernbedienung-versandkostenfrei/dp/B07BNJTYJW/ref=sr_1_4?ie=UTF8&qid=1549454568&sr=8-4&keywords=jinvoo+smart+steckdose converted that both worked well for a couple of hours on tasmota. One died after a configuration reset and one by itself. The only thing I can see is a faint red light when plugging in that immediately goes off. Did not try to flash them yet using a serial cable. I am puzzled what could be the problem here, devices were working properly as S20 device.

uspass, 2019-02-07:
I have one of those tuya water leak sensor.
I have drawn the schematics and made some comments on.
All info so far is here:
https://github.com/uspass/Tuya-Water-Leak-Sensor

Any info is welcome.

Thanks,
us

balvant813, 2019-02-09:
I have converted four [SP201](https://www.amazon.com/gp/product/B07L63S731) , details at (https://github.com/SynAckFin/TuyOTA/wiki)

onebyside, 2019-02-11:
Two Epicka https://www.amazon.com/EPICKA-Mini-Wireless-Compatible-Assistant-Required/dp/B07BW9QB1G dead after sucessfully flashing a Gosund SW2. No lights, no connection and no ping.

demtat, 2019-02-11:
OK, I tried it on the second NX-SP201 that I had and everything worked correctly.  So, I decided to try working on the first one again.  It had been unplugged for about 5 days now.  Plugged it back in and tried to find it on the network but could not see it on my router.  Then I happened to run TasmoAdmin to try to add a couple of other plugs and guess what it found it.  Somehow, the NX-SP201 was showing up in my router with the host name "--".  I have a couple of -- devices on my network already (I think they are my DirecTV receivers), so I just wasn't seeing it.  My router, unfortunately doesn't help here, as when it assigns the DHCP addresses it does so randomly, so I can't just look at the largest IP in the list.  Anyhow, all of my devices are now working.

DavinKD, 2019-02-13:
> I have converted four [SP201](https://www.amazon.com/gp/product/B07L63S731) , details at (https://github.com/SynAckFin/TuyOTA/wiki)

I checked out your link and noticed you had an issue with XS-SSA06 RGB.  I've been able to flash 8 of them with no issues.

qwertangel, 2019-02-21:
@onebyside , https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices indicates Epicka is compatible. did you have any issues during flash ?

onebyside, 2019-02-21:
I flashed a Gosund successfully and then failed on 2 Epicka 1000. They both need some serial  flash attention. I never got through the backup portion of the procedure to begin with.

onebyside, 2019-02-26:
Did you end up flashing some Epicka's? I still have 2 of the 4 unflashed and Im itching to flash them.

demtat, 2019-02-26:
Yes, I finally got them loaded using the OTA method.

Thanks,

Tim

qwertangel, 2019-02-26:
> Yes, I finally got them loaded using the OTA method. Thanks, Tim From: onebyside <notifications@github.com> Sent: Monday, February 25, 2019 6:16 PM To: ct-Open-Source/tuya-convert [...]

what OTA method ?

qwertangel, 2019-03-02:
@demtat can you please share how did you flash your epickas ? I have 3 epicka's waiting to hear from you.. :-)

kueblc, 2019-03-05:
@qwertangel This project, tuya-convert, is the OTA method.

@demtat Glad you recovered all your devices. I'll close this issue now, if anyone has further questions please open a new issue.

closed 2019-03-05 by kueblc

qwertangel, 2019-03-06:
I am confused. Is OTA still possible even after epicka's stop responding ?

onebyside, 2019-03-06:
I think your in the same boat as I. 2 dead and need serial recovery.

demtat, 2019-03-06:
The OTA worked the first time, but I had an issue where my router wasn’t listing the IP that it clearly gave out via DHCP.  It showed up in TasmoAdmin and I was able to get the IP, log in, and complete the setup.

The Configuration is:
  Generic
  GPIO4   – Led1i
  GPIO12 – Relay1
  GPIO14 – Button1

Thanks,

Tim

onebyside, 2019-03-06:
I finally flashed the last 2 epicka ota successfully. The first 2 will have to be serial flashed. I have to say the instructions aren't completly clear for newbs on parts of the process.

qwertangel, 2019-03-06:
@onebyside good to hear that ! any specific part you want to share which I should look out for when I flash mine ?

onebyside, 2019-03-07:
@qwertangel   Both of the Epicka's I did had "system upgrade userbin check: user2 0x81000"

Flash2 (moves the conversion software to user2)
and then Flash3

there wont be any lights on the plug after either flash  (dont worry its not dead)
Check you wifi after and you should see sonoffXXXX

vinny75, 2019-11-16:
> I flashed a Gosund successfully and then failed on 2 Epicka 1000. They both need some serial flash attention. I never got through the backup portion of the procedure to begin with.

What did you use as the tasmota configuration for your Gosund SW2? Can't figure it out.

onebyside, 2019-11-16:
I used the epicka config

vinny75, 2019-11-17:
> I used the epicka config

Couldn't get the dimmer to work. You are talking about this switch? https://www.amazon.com/dp/B07X1QRC23

onebyside, 2019-11-17:
Did you not ask me about the gosund sw2 and what config I used?