ct-Open-Source/tuya-convert #73: Flashing successfully and after wifi setting no connection

botak13, 2019-02-03:
I have managed to flash my universal remote control under Linux Mint.
After I connected to the sonoff Wlan and set up my home Wi-Fi in the web interface I had to restart the device.

Since then nothing works. I suspect that I have typed something wrong at the wifi. A renewed call of the sonoff hotspot over 4x press did not bring anything.
The LED on the device is always off now.

Can I reset Tasmota completely?

botak13, 2019-02-03:
I have specified the Wifi Setup my normal wifi but the password was already filled with stars. My device is no longer showing any signs of life.

But I'll test that with the hotspot.

onebyside, 2019-02-04:
> Does anyone have an idea why my device is no longer displayed in the router? I only changed the password but not the SSID.

So when you created a hotspot with no password, did the plug connect? Did you power cycle....and or held button on the plug for 40 secs?

botak13, 2019-02-04:
I deactivated the encryption on my router and then my device connected. I have already restarted the device several times and already experimented with the button. But unfortunately I see no connection attempts in the router.

Interloping, 2019-02-05:
Some wifi passwords just plain don't work. I had to change the password for my wifi to not have symbols like * or punctuation. Before that, the sonoff devices would never connect to my wifi and would seem to be bricked: no LEDs working, no response to pressing the button, etc. So the steps were like this:

1. After the sonoff device seemed dead, create a wifi network with the same SSID I set up the sonoff to connect to, but NO password/security
2. connect to the tasmota settings page (by identifying the IP address my router gave the sonoff and putting that in a web browser on a device on the same network)
3. leave the SSID unchanged in the tasmota settings, but change the password to something long but with plain characters (like correcthorsebatterystaple - shout out to xkcd)
4. change the password on my router wifi to be the same as the password I just put in the tasmota settings
5. once everything connected, get into the tasmota settings and choose the correct module settings so all of the relays, buttons, etc. work as they should.

Lucky for me I had already put all my smart devices on a separate wifi AP, so step 3 was not the huge nuisance it could be. For routers that support adding a virtual AP with different SSID and wifi password, that would save you having to re-authenticate every wifi device in your home.

botak13, 2019-02-05:
I did that but after a reboot, the device does not even appear with the MAC address in the router.
No chance for the SSID.

My Wifi has _ in the name maybe its a problem for Tasmota?

DavinKD, 2019-02-05:
My ssids have _ in them and seemed to work fine.

botak13, 2019-02-05:
lly confusing. I was able to log in to my device when I deactivated WPA2. Then I just changed the password in Tasmota and now my hotspot will not find the device again. Even without a password.

botak13, 2019-02-06:
I've just flashed a power outlet just like the remote and set my wifi. The outlet has also connected directly to it. What's up with the remote control?

Is there a way to flash the image again so that my device opens a hotspot?

kueblc, 2019-02-06:
Glad you have figured out your connection issues with Tasmota. Please feel free to close this issue.

closed 2019-02-06 by botak13

bensil1, 2019-02-06:
1. After the sonoff device seemed dead, create a wifi network with the same SSID I set up the sonoff to connect to, but NO password/security

2\. connect to the tasmota settings page (by identifying the IP address my router gave the sonoff and putting that in a web browser on a device on the same network)

the hotspot ip ?

Interloping, 2019-02-06:
> 1. After the sonoff device seemed dead, create a wifi network with the same SSID I set up the sonoff to connect to, but NO password/security
>
> 2. connect to the tasmota settings page (by identifying [...]

Not the hotspot IP, the IP of the sonoff device (the smart switch, plug or bulb or whatever you are flashing). Once tasmota is flashed and your device is connected to a wifi network, it runs an http server that allows you to configure the device using a web browser.