ct-Open-Source/tuya-convert #76: More than 1 discovery method on Tuya?

danps1, 2019-02-06:
I'm wondering if there's possibly more than 1 way that Tuya devices are discovered?
I have successfully flashed 3 different types of device with Tuya Convert, so I'm fairly confident that my setup is working, however, there is a Tuya based wall power socket that I have which is resisting all attempts, yet I can discover and add it into the Tuya app. When I put it into Smartconfig mode, the LEDs flash rapidly, but then I watch the Wifi log on Tuya Convert and don't see any attempts at all to connect. Everything is physically close enough to work.
Is there some extra steps I could take to troubleshoot?
Or maybe is there a variation of this technique that could work if I put the device into AP mode, and connected to its Wifi?
Thanks

Interloping, 2019-02-06:
Have you had any success flashing any devices since trying the one that isn't working? I have the same problem with a bulb never connecting - but today I flashed a switch on a different system and had the same problem. Discovered I had to switch to using a different fork of tuya, as something to do with npm seemed to be broken. I'm wondering if a recent ubuntu update caused it to stop working.

The tuya fork and branch I used that worked again was obtained this way:
`git clone -b smart-link-py https://github.com/kueblc/tuya-convert.git`

danps1, 2019-02-06:
Interesting. Yes I did manage to flash a couple of other devices after this, but will give the fork a go anyway

Interloping, 2019-02-07:
Any success? I've now successfully flashed the bulb as well using the smart-link-py fork.

danps1, 2019-02-09:
Sorry for the slow reply. I just got round to trying the fork you suggested, but unfortunately, still no luck with this device!

danps1, 2019-02-09:
So 1 other weird behavior to add about this power socket. In the Tuya app on my phone, if I do "Search Device" it won't find it, but if I choose "electrical Outlet" from the default "add manually screen" it will find it. Almost like there's a seperate discovery packet being sent for a power socket?

bigdavediode, 2019-02-27:
For Merkury bulbs, and yours might be similar, I've had luck with linking the bulb to the geeni app and then resetting the device to defaults in the app. Then it flashed easily.

kueblc, 2019-03-05:
Hi @danps1, are you still having issues? What is the name and model of the device you were having problems with?

danps1, 2019-03-20:
Hi @kueblc,
Sorry for the slow reply. Yep still having issues. Tried adding to app and resetting per @bigdavediode advice, and also a forked version. They are wall power sockets - fairly generic, from "SRL tech" - https://srltech.com.au/portfolio_page/power-points/

kueblc, 2019-03-20:
Are you able to capture your network activity with `tcpdump` or Wireshark?

danps1, 2019-03-21:
I can certainly try. I assume what I'm looking for is the packets coming from the device when it is in pairing mode?

kueblc, 2019-03-23:
Yes, create an access point and try recording your network activity as you try to pair to that network.

v0d0r, 2019-03-29:
The fork also worked for me on a Tuya Bulb.  Thanks for this.

-------------------------------------------------------------------------------------------
The tuya fork and branch I used that worked again was obtained this way:
git clone -b smart-link-py https://github.com/kueblc/tuya-convert.git
-------------------------------------------------------------------------------------------

kueblc, 2019-03-31:
@v0d0r Glad it worked for you, FYI this branch has been merged into master

WilkyF18, 2019-04-01:
@danps1 did you ever get the plug to flash?  I have Teckin SP10 and SP20.  I cannot get them to 'appear.'  Just get never ending dots "............."

Same as you.  Would not 'discover' in the Tuya app either, but I could manually add it in the Tuya app.  

I've tried in "easy mode" rapidly flashing led, and "AP mode" slow flashing led.

WilkyF18, 2019-04-03:
I connected to vtrust-flash using a spare laptop instead of my iPhone and both the SP10s and one SP20 worked. Flashed them to Tasmota.

Unfortunately, the other SP20 I can’t even get it into paring mode, the LED on it never flashes no matter how long I hold the power button. Makes no “clicking” sound. I must have bricked it. When I try “./start_flash.sh” never ending dots. I also tried “curl http://10.42.42.42/flash3” but it says nothing can be found at that IP address.

eflemingtra, 2019-04-03:
Hi, I'm having a similar issue with a Tuya Bulb. I purchased two bulbs - the first one took many attempts but eventually I was able to flash it. I hadn't done anything different when it connected, it was just dumb luck. The second bulb, no such luck. I had no issues connecting with the Tuya app to the bulbs.  I have used both the master fork and the one suggested above.

Would providing copies of my logs be of any assistance?

wizdude, 2019-04-09:
@eflemingtra 
> I had no issues connecting with the Tuya app to the bulbs

be aware that if you've connected the bulbs to tuya, there is more than a high possibility that the firmware has now been updated and tuya-convert may be unable to flash them for you.

as per comments posted on one of Dr Zzz's videos entitled "Don't add devices to the Tuya App - it is now blocking Tuya-Convert"
https://www.youtube.com/watch?v=SxNHKvPi4K8

i always get my tuya products "out of the box" and flash them right away and never connect them to tuya's cloud. at some time in the future, however, i know this will no longer work (unless the tuya-convert guys work around this).

the only tip i can give you from all of the tuya-converts that i've done - take the bulb that's not working. plug it into a socket, and then power cycle it 5 times. i usually do about 1 second between switching, so: on, off, on, off, on, off, on, off, on. hopefully the bulb will flash (ready to pair) or will not come on at all (waiting for flash).

always make sure you have a secondary device connected to the vtrust-flash first otherwise it won't work.

wizdude, 2019-04-09:
@danps1 
> They are wall power sockets - fairly generic, from "SRL tech"

if you get totally stuck, you may have to directly connect to them and flash them (make sure you remove them from AC power first!)

russell from SRL has stated he has several customers flashing them with tasmota with no issues. i would assume most of them would be directly connecting to the hardware itself.

mikegeeezy, 2019-09-13:
Now that this has been merged, can someone please help me out with the correct command to clone it?  I've got a bunch of new Feit bulbs that aren't converted with the traditional method, like to give this a shot.

`git clone -b python https://github.com/ct-Open-Source/tuya-convert`

Is hitting me for authentication - which I don't have.
Thanks!

kueblc, 2019-09-28:
Closing - continue to comment if you have related concerns, or open a new issue.

For those that had HTTPS firmware, check out #279

closed 2019-09-28 by kueblc