ct-Open-Source/tuya-convert #80: GU10 RGB bulb bricked after Flash/Undo?!

Maschine2, 2019-02-06:
HI,

I own a Tuya based RGB LED bulb with GU10 socket. I bought it on Amazon Germany:
https://www.amazon.de/gp/product/B07F7S2PDG/ref=oh_aui_search_asin_title?ie=UTF8&psc=1

This kind of bulb is also described here (AiLight): http://tinkerman.cat/ailight-hackable-rgbw-light-bulb/#lightbox-gallery-gdb3MMfF/7/

I tried to flash the device with a Raspberry Pi 3B+. Connection worked and I could retrieve the following details:

```
READ FLASH: http://10.42.42.42/get?read=HEX-ADDRESS
ChipID: 4c140e
MAC: DC:4F:22:4C:14:0E
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 0
system_upgrade_userbin_check: user1 0x01000
```

Since I was not sure which alternative firmware to use I used the UNDO command (curl http://10.42.42.42/undo) to go back to the original firmware.

After this, the lamp stays off if connected to power. No chance to switch the light on. Also, the pairing mode (three times on/off) is not working anymore. The bulb is off.

What can I try to get the bulb back to a working mode? Any ideas welcome. Thanks.

Jason2866, 2019-02-09:
Open it and flash it via serial...

markcarline, 2019-02-26:
Did you manage to fix it? Are you able to open the GU type bulbs ?

kueblc, 2019-03-05:
Closing due to lack of response, please reopen if this is still an issue.

closed 2019-03-05 by kueblc

kueblc, 2019-03-14:
@Maschine2 Did you backup the firmware before doing `/undo`? I'm tracking down a bug, any firmware which did not successfully undo would be very helpful.

kueblc, 2019-03-14:
@Maschine2 I've located the bug you were affected by. If you have backed up your firmware and can flash your device over serial, it is possible to recover your stock firmware.

Maschine2, 2019-03-14:
@kueblc, in the meantime I was able to flash my device over serial. Though, tanks for your help. Do you still need more information from my tries?

@markcarline, there is this "milky" round plastic lid on top of the GU10 bulb. Use a small knife or a similar tool to get between the silver ring and the lid. Then open it carefully. (To be sure: Disconnect electricty before!)

kueblc, 2019-03-14:
Very happy to hear that you were able to recover the device over serial. Did you make a backup of the firmware beforehand, either with the `backup.py` script or via `esptool.py` over serial? This data would help verify that the bug has been fixed.