ct-Open-Source/tuya-convert #81: Tuya water leak sensor

uspass, 2019-02-07:
I have one of those tuya water leak sensor.
I have drawn the schematics and made some comments on.
It has, beside 8266, a second processor that actually deals with IO.
All info so far is here:
https://github.com/uspass/Tuya-Water-Leak-Sensor

Any info is welcome.

Thanks,
us

kueblc, 2019-02-08:
Hi @uspass 

What's the issue?

uspass, 2019-02-08:
Hi @kueblc ,

I'm sorry if I have posted in the wrong place.
I was trying to follow this advice: 
"Please at least add the device model number, brand name, geographical area where you have bought the device and its flash mode (as seen in the device information). Add the GPIO assignments as well if you have found them to save the developers of alternative firmwares some time."
On the other hand, I'm stuck as i have no clue what the serial TxRx protocol between ESP and EFM might be.
I have disassembled the code and try to figure it out, but it's a long shot.
I was hoping someone has info on that.

Thank you,
us

kueblc, 2019-02-08:
Ah, I see, thank you for sharing!

I may be able to assist with disassembly. Would you post what you have so far?

uspass, 2019-02-09:
@kueblc 
First of all thank you for your offer.
I have worked in the past with several assembly languages and this is not my first dis-assembly ;).
But is my first ESP. Still learning, so bear with me.
Reading the bin structure we get:

File   offset |   |   |  
-- | -- | -- | --
0x00000 | Header | 0xE9 |  
0x00001 | Nr of segments | 0x03 |  
0x00002 | SPI Flash Interface | 0x00 | QIO
0x00003 | Memory+Speed | 0x20 | 1M, 40MHz
0x00004 | Entry point | 0x40100294 |  
0x00008 | Segment 1 Offset | 0x40100000 |  
0x0000C | Segment 1 Size | 0x00000574 |  
0x00010 | Segment 1 Data |   |  
0x00584 | Segment 2 Offset | 0x3FFE8000 | User data RAM.
0x00588 | Segment 2 Size | 0x00000308 |  
0x0058C | Segment 2 Data |   |  
0x00894 | Segment 3 Offset | 0x3FFE8308 |  
0x00898 | Segment 3 Size | 0x0000021C |  
0x0089C | Segment 3 Data |   |  
0x00AB8 | Zero Padding | 0,0,0,0,0,0,0 |  
0x00ABF | Checksum 0xEF | 0xC0 |  
  |   |   |  
0x00AC0 | Blank | 0xFF,0xFF,…,0xFF |  
  |   |   |  
0x01000 | Header | 0xEA |  
0x01001 | Nr of segments | 0x04 |  
0x01002 | SPI Flash Interface | 0x00 |  
0x01003 | Memory+Speed | 0x20 |  
0x01004 | Entry point | 0x40100004 |  
0x01008 | Segment 1 Offset | 0x00000000 |  
0x0100C | Segment 1 Size | 0x00057D30 |  
0x01010 | Segment 1 Data |   |  
  |   |   |  
0x58D40 | Header | 0xE9 |  
0x58D41 | Nr of segments | 0x03 |  
0x58D42 | SPI Flash Interface | 0x00 |  
0x58D43 | Memory+Speed | 0x20 |  
0x58D44 | Entry point | 0x40100004 |  
0x58D48 | Segment 1 Offset | 0x40100000 |  
0x58D4C | Segment 1 Size | 0x00006634 |  
0x58D50 | Segment 1 Data |   |  
0x5F384 | Segment 2 Offset | 0x3FFE8000 |  
0x5F388 | Segment 2 Size | 0x0000048C |  
0x5F38C | Segment 2 Data |   |  
0x5F818 | Segment 3 Offset | 0x3FFE8490 |  
0x5F81C | Segment 3 Size | 0x0000252C |  
0x5F820 | Segment 3 Data |   |  
0x61D4C | Zero Padding | 0,0,0 |  
0x61D4F | Checksum 0xEF | 0x8A |  
  |   |   |  
0x61D50 | Unknown | 0x4E |  
0x61D51 | Unknown | 0x3B |  
0x61D52 | Unknown | 0xE9 |  
0x61D53 | Unknown | 0x5C |  
  |   |   |  
0x61D54 | Blank | 0xFF,0xFF,…,0xFF |  
0x79000 | Unknown Data |   |  
0x79274 | Blank | 0xFF,0xFF,…,0xFF |  
0x7B000 | Unknown Text Data |   |  
0x7B08A | Blank | 0xFF,0xFF,…,0xFF |  
0x7C000 | Unknown Text Data |   | SSID SsidPassword Key Cloud
0x7C36A | Blank | 0xFF,0xFF,…,0xFF |  
0x7D000 | Unknown Text Data |   |  
0x7D0C1 | Blank | 0xFF,0xFF,…,0xFF |  
0x7E000 | Unknown Text Data |   |  
0x7E00A | Blank | 0xFF,0xFF,…,0xFF |  
0x7F000 | Unknown Text Data |   |  
0x7F006 | Blank | 0xFF,0xFF,…,0xFF |  
0xFB000 | Unknown Text Data |   |  
0xFB064 | Blank | 0xFF,0xFF,…,0xFF |  
0xFC000 | Unknown Text Data |   |  
0xFC080 | Blank | 0xFF,0xFF,…,0xFF |  
0xFD000 | Unknown Text Data |   |  
0xFD36E | Blank | 0xFF,0xFF,…,0xFF |  
0xFE000 | Unknown Text Data |   |  
0xFE36E | Blank | 0xFF,0xFF,…,0xFF |  
0xFF000 | Unknown Text Data |   |  
0xFF020 | Blank | 0xFF,0xFF,…,0xFF |  

I got 2 distinct dis-assemblies:
1. Seg123.idb which is the first 0xE9 header, code loaded at 0x40100000 and the RAM at 0x3FFE8000
2. Seg4567.idb which is the 0xEA header, code loaded at 0x40201010 and the second 0xE9 header, code loaded at 0x40100000 and the RAM at 0x3FFE8000.
Notes:
1. I don't really understand how goes the third code segment (second 0xE9) over the first one as they share the same address space. I am pretty sure there is no mistake as the second code segment does references (routine calls and direct memory access) to the third one and vice versa. I need to get some info about the ESP memory map. All I have is rubbish. Need a deeper understanding. 
2. The 0xEA code segment looks like a SDK to me. 
3. The first and third code have references to some routines located at 0x40000000, but I can't see anything in the bin file that goes there.
4. From the routines structure, it looks like they used a c compiler (at least for the SDK). 
The routine call arguments are passed in a2, a3, etc registers, in that order. a1 is used to keep the stack pointer. a0 is used to do referenced calls. 
Usually a routine sets the stack pointer to get some room, 
addi            a1, a1, 0xF0
save in stack some ax registers used as local variables, (a0 in this case)
s32i            a0, a1, 0
does what it is supposed to do, 
restores ax registers 
l32i.n          a0, a1, 0
restores stack pointer
addi            a1, a1, 0x10
and returns
ret.n
Just before the routine there are constants/offset references used inside it.
5. The bin file is retrieved  from a Tuya registered sensor. I have a second one unregistered and I can get a virgin bin if needed. 
6. I reckon I can retrieve the MCU (EFM) bin file. From what I have read the EFM is 8051 based so it would be easier for me to understand the serial protocol from that end. But that doesn't change things much because the protocol is described [here](https://docs.tuya.com/en/mcu/mcu-protocol.html)
My goal is to get some firmware like tasmota to talk to MCU.

I have uploaded the idb's in my git.
If you need anything just let me know.

Thank you,
us

ctandi, 2019-02-11:
Hey everyone,

would this be of any help?
https://docs.tuya.com/en/mcu/mcu-protocol.html

Cheers!

uspass, 2019-02-11:
@ctandi 
Thank you.
I'm aware of that doc. Actually it is referred in Note 6.

Cheers,
us

epicurean70, 2019-03-04:
Is this the product that has the Tuya water leak sensor?
https://www.aliexpress.com/item/2018-NEO-COOLCAM-Smart-WiFi-Water-Sensor-Flood-Leak-Detector-Alarm-APP-Notification-Alert-No-Hub/32882682503.html

uspass, 2019-03-04:
@epicurean70 
Yes. That's it.

epicurean70, 2019-05-20:
Has anyone been able to flash tasmota via tuya-convert for this water sensor?

uspass, 2019-05-22:
I did. 
The problem is the MCU powers the ESP for few seconds and then cuts off the power.
Tasmota should be modified to transmit the info during those seconds.

kueblc, 2019-11-17:
Closing due to age and relevance. Thanks @uspass for documenting your work.

closed 2019-11-17 by kueblc

jsiegmund, 2020-06-09:
@uspass so how exactly did you make those modifications? I have the same sensor looking to flash Tasmota on it. I succesfully did the same with a wifi power switch, but with this sensor the same procedure using tuya-convert is not working.

ZeppDK, 2020-07-28:
@uspass i would also like to know how you did it