ct-Open-Source/tuya-convert #84: RaspberryPiZeroW - successful flash

YARA-CODER, 2019-02-09:
Just to let you know - I have just flashed Neo Coolcam plug with RPi0W (latest Raspbian Strech, fresh install, connected via USB-Ether) 
Only modification is that you need to get tarball binaries with nodejs for armv6l.
Simply unpack in to` /usr/local/` before runnnin prerequisites script.

swbradshaw, 2019-02-09:
npm install for Pi Zero W:

```
curl -o node-v11.9.0-linux-armv6l.tar.gz https://nodejs.org/dist/latest/node-v11.9.0-linux-armv6l.tar.gz
tar -xzf node-v11.9.0-linux-armv6l.tar.gz
sudo cp -r node-v11.9.0-linux-armv6l/* /usr/local/
npm -v
```

swbradshaw, 2019-02-10:
@YARA-CODER I have my Pi Zero W connected to HDMI/keyboard. I was able to install npm but when I run the script, the "vtrust-flash" AP is never created. Not sure what to try next.

mobidarabbit, 2019-02-11:
@swbradshaw On my Pi Zero w (running with stretch lite) i had to "killall wpa_supplicant" to get the AP working. You have to do this after every boot, if you dont configure your pi to not start wpa_supplicant. Maybe you have the same issue.

pelrun, 2019-02-11:
It won't work if your Pi is connected to a wifi network. So either kill wpa_supplicant or remove the credentials from the wpa_supplicant.conf file so it won't connect to anything.

Yes, this means you can't ssh in to a headless Pi over wifi. Either hook up a monitor and keyboard, or configure it for USB-Ethernet gadget mode and plug it into your PC.

swbradshaw, 2019-02-12:
Thanks for the help. I was able to get the AP created and I can connect to it from my phone and see the "Hello World" message from the site. However, when I continue the script with my Tuya module in pairing mode (I can see it's AP from my phone), the process doesn't do anything. The logs show that the script sends the UDP broadcast, but the Martin Jerry switch isn't responding to it :-(

DillonJ, 2019-02-12:
Hey swbradshaw...

If I'm not mistaken, if you can see the device's AP, then it's in AP/compatibility mode and not in smart pairing mode.... 

I'm trying to flash (So far unsuccessfully (https://github.com/ct-Open-Source/tuya-convert/issues/86)) an "eMylo Smart switch"). In general, I believe the LED needs to be flashing quickly when it's in pairing mode (AP mode is slower). I can see and connect to the the vtrust AP but have so far been unable to get further.

swbradshaw, 2019-02-12:
@DillonJ Thanks for the reply. I literally edited my last reply at the same time you wrote that with the same explanation. In my case, the lights don't seem to flash different speeds between AP / smart pair mode, but the LED is blue in smart pairing mode and red in AP mode.  

I'll reply in your other thread.

kueblc, 2019-02-12:
Glad you were all successful, you may close the issue :+1: We'll continue with @DillonJ's troubleshooting in his issue #86

closed 2019-03-05 by kueblc

2016for, 2019-05-21:
> Yes, this means you can't ssh in to a headless Pi over wifi. Either hook up a monitor and keyboard, or configure it for USB-Ethernet gadget mode and plug it into your PC.

Hey @pelrun I would like to use my pi zero w, could you guide me how I can use the "USB-ethernet gadget mode" thingy with my macbook?

Thanks in anticipation

2016for, 2019-06-02:
> npm install for Pi Zero W:
> 
> ```
> curl -o node-v11.9.0-linux-armv6l.tar.gz https://nodejs.org/dist/latest/node-v11.9.0-linux-armv6l.tar.gz
> tar -xzf node-v11.9.0-linux-armv6l.tar.gz
> sudo cp -r node-v11.9.0-linux-armv6l/* /usr/local/
> npm -v
> ```

@swbradshaw So I connect my monitor and keyboard to a fresh pizerow with fresh stretch-lite install and enter these lines. Then follow below instructions of first page right?

```
**INSTALLATION**
# git clone https://github.com/ct-Open-Source/tuya-convert
# cd tuya-convert
# ./install_prereq.sh

**flash loader firmware + backup**
# ./start_flash.sh

```

> killall wpa_supplicant
When do I do this? after ./install_prereq.sh and before ./start_flash.sh?

Thanks in anticipation

swbradshaw, 2019-06-02:
@2016for Correct. You'll run killall wpa_supplicant every time you start the Pi before you run the start_flash.sh

2016for, 2019-06-02:
> @2016for Correct. You'll run killall wpa_supplicant every time you start the Pi before you run the start_flash.sh
@swbradshaw I am getting this error when trying the tar command

```
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```