ct-Open-Source/tuya-convert #86: Clarify instructions and re-requisites

DillonJ, 2019-02-09:
First off, thanks for this awesome script - I have a number of "eMylo" smart switches and am trying to get tasmota installed on them.

A number of questions that could perhaps be addressed in the readme?

 - I presume that this can be run using an RPi via ssh? I only ask because one of the first things the script says is " Starting AP in a screen".
 - Does the Tuya device need to be connected to wifi before the script is run? I.e. does it need to be set up already and registered with the app etc?

For me, the script is stopping at "stopping any apache web server". What might be the cause of this?

Many thanks in advance

kueblc, 2019-02-09:
>     * I presume that this can be run using an RPi via ssh? I only ask because one of the first things the script says is " Starting AP in a screen".

In this case `screen` is referring to the unix utility that creates multiple virtual terminals, not a computer display. You should be able to run this over SSH if you have another internet adapter, such as an ethernet adapter, as the WiFi may not be able to remain connected as a station while in AP mode.

>     * Does the Tuya device need to be connected to wifi before the script is run? I.e. does it need to be set up already and registered with the app etc?

No, the Tuya device does not need to be connected or registered beforehand, tuya-convert will need to reprovision the device with it's own keys in order to perform the upgrade. For this you need to put the device into pairing mode, which varies by device, but usually involves holding a button for 3-5 seconds or turning it off and on 3 times.

> For me, the script is stopping at "stopping any apache web server". What might be the cause of this?

Not sure why it would stop here, but the command running at that point is `sudo service apache2 stop >/dev/null 2>&1`. This is really only relevant if you're running a web server, an apache one at that, so you can comment it out.

Note that in the near future this should be replaced by a more generalized approach, either letting the user decide what services need to be stopped, or using `netstat` or `lsof` to identify what, if any, program is occupying port 80. @nanobyte84 @merlinschumacher

DillonJ, 2019-02-12:
Thanks for this. I've been having a hell of a time getting this working. I'm using an early model RPI (1B I think) and the node installation failed... the only way I could get node and npm working correctly was by removing all installations of both and by following these instructions, adapted from https://raspberrypi.stackexchange.com/questions/4194/getting-npm-installed-on-raspberry-pi-wheezy-image/37976#37976
```
wget https://nodejs.org/dist/v10.15.1/node-v10.15.1-linux-armv6l.tar.xz
sudo mv node-v10.15.1-linux-armv6l.tar.xz /opt
cd /opt
sudo tar -xvf node-v10.15.1-linux-armv6l.tar.xz
sudo mv node-v10.15.1-linux-armv6l nodejs
sudo rm node-v10.15.1-linux-armv6l.tar.xz
sudo ln -s /opt/nodejs/bin/node /usr/bin/node
sudo ln -s /opt/nodejs/bin/npm /usr/bin/npm
```
then I executed the remainder of install_prereq.sh manually.

The only error I am getting now is a keyboard interrupt error in smarthack-web.log. I'm not sure if it's running correctly or not. These are the log contents:

```
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 132, in <module>
    main()
  File "./fake-registration-server.py", line 128, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 132, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1424, in _run_once
    handle._run()
  File "/usr/lib/python3.5/asyncio/events.py", line 126, in _run
    self._callback(*self._args)
  File "/usr/local/lib/python3.5/dist-packages/tornado/ioloop.py", line 758, in _run_callback
    ret = callback()
  File "/usr/local/lib/python3.5/dist-packages/tornado/stack_context.py", line 300, in null_wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/tornado/ioloop.py", line 1229, in _run
    return self.callback()
  File "/usr/local/lib/python3.5/dist-packages/tornado/autoreload.py", line 185, in _reload_on_update
    _check_file(modify_times, path)
  File "/usr/local/lib/python3.5/dist-packages/tornado/autoreload.py", line 192, in _check_file
    modified = os.stat(path).st_mtime
KeyboardInterrupt
Listening on port 80
```
Finally, how does this line in start_flash.sh work without inserting the `node` before `./smartconfig/smartconfig.js`? I get an error unless I insert `node`.

```
$screen_with_log smarthack-smartconfig.log -S smarthack-smartconfig -m -d ./smartconfig/smartconfig.js
```

swbradshaw, 2019-02-12:
@DillonJ My smarthack-web.log looks identical with the KeyboardInterrupt and it works fine.

There is a tuya fork that someone made that removes the NodeJS requirement. He rewrote the smartconfig UDP broadcast functionality in Python.   Maybe that's something you could try? Although it sounds like you got passed the node requirement.

`git clone -b smart-link-py https://github.com/kueblc/tuya-convert.git`

DillonJ, 2019-02-12:
Thanks @swbradshaw. Does the fact that I need to add `node` to the line referenced above not indicate that something is amiss?

swbradshaw, 2019-02-12:
> Thanks @swbradshaw. Does the fact that I need to add `node` to the line referenced above not indicate that something is amiss?

I missed that part. Yeah, you shouldn't have to insert "node" in that line.  The reason it should just work is because the script is marked executable, and the first line in the script references node.

DillonJ, 2019-02-12:
Thanks for the explanation @swbradshaw. The instructions I followed installed node to a different folder... so after correcting that, I get `cannot exec smartconfig.js: permission denied`. I chmodded to +x and it still doesn't work

swbradshaw, 2019-02-12:
You might just want to try that other branch I referenced earlier that doesn't use node.

kueblc, 2019-02-12:
Are you running as root or with `sudo`?

DillonJ, 2019-02-12:
Hi folks..

I tried running the fork and no luck either. I gave up and soldered connections on was able to flash using esptool. 

I haven't seen anyone successfully flash my device anywhere... I'm pretty sure its TYWE2S based. This is [the device](https://www.amazon.fr/t%C3%A9l%C3%A9commande-Interrupteur-minuterie-coupe-circuits-Compatible/dp/B07BQ2HLD2/ref=sr_1_2?ie=UTF8&qid=1549990144&sr=8-2&keywords=emylo+smart+switch): It's an "eMylo Smart Switch".

I have posted pictures of the board here: https://groups.google.com/forum/#!category-topic/sonoffusers/Lkk9rQ33c2g

I wonder if this trouble flashing is the same issue as #76 is alluding to?

swbradshaw, 2019-02-12:
@DillonJ I saw you asking about how Tasmota works. Here is a nice video explaining how to setup an unknown plug: https://youtu.be/m_O24tTzv8g?t=388  You can jump to the 9min mark to see him setup the plug.

closed 2019-03-23 by kueblc