ct-Open-Source/tuya-convert #89: Successfully Flashed Gosund WP3 However Wifi Loses Connection

Photo64, 2019-02-10:
I was able to successfully flash 3 Gosund WP3 smart plugs however I now notice that the wifi loses connection for a split second and then reconnects. I am using all three in Home Assistant and have a desk lamp attached to one. Every few seconds the light flickers and I see in Home Assistant it goes to "unavailable" to "turned on". 
9:59 PM Gosund1 Smart Plug changed to unavailable
9:59 PM Gosund1 Smart Plug turned on

All three smart plugs are doing this. The disconnect must only be a split second but it is constant and is making the plugs unusable. I followed this setup -> https://github.com/ct-Open-Source/tuya-convert  and once in the web page I set the plug up with these parameters:
GPIO4: Switch1, GPIO12: LED2i, GPIO13: LED1i, GPIO14: Relay1 and also did a switchmode1 3 to correct the button on/off when pressed.

Is there something that can be done to correct this behavior or is it do to the Sonoff-Tasmota v6.4.1 being used?

kueblc, 2019-02-10:
This is a Sonoff-Tasmota issue, happy to hear your flash was successful anyway! Hope that you can find a resolution for your firmware issue :smile:

Photo64, 2019-02-10:
I did an OTA upgrade and tried both version 6.3.0 with core 2.3.0 and version 6.4.1.15(e18263d-sonoff) with core 2_5_0/3.0.0-dev(c0f7b44)  but both still have wifi disconnects causing the light to flicker. Does anyone know which version and core that will eliminate the wifi disconnects?

LouiS22, 2019-02-10:
Check the logging menu entry and make sure serial logging is disabled.

Photo64, 2019-02-10:
I set it to "None" but the light is still flickering
[image]

onebyside, 2019-02-10:
Wouldn't this be better addressed in Tasmota chat

Photo64, 2019-02-10:
Have a link to that chat or do you mean create an issue in the Tasmota github?

onebyside, 2019-02-10:
I searched " Sonoff-Tasmota " repositories. Looks to be a few possibilities.

kueblc, 2019-02-10:
As mentioned this is not the place for firmware issues as we do not develop Sonoff-Tasmota here. Since you have successfully flashed your device, you can close this issue and seek help for your firmware via the relevant channels ie [the Sonoff-Tasmota wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting). Thanks and good luck!

Photo64, 2019-02-10:
I'll move this over to the tasmota channels, thanks for your help

closed 2019-02-10 by Photo64