ct-Open-Source/tuya-convert #90: Successfully flashed Merkury Innovations BR30 Smart Light Bulb, 65W Tunable White LED

rb2k, 2019-02-11:
I was able to flash and install tasmota on this bulb.
It's available at Walmart in the US (https://www.walmart.com/ip/Merkury-Innovations-BR30-Smart-Light-Bulb-65W-Tunable-White-LED-1-Pack/404320234)

It's a BR30 white LED with tunable color temperature.
I have not been smart enough to figure out  what I have to set the different GPIOs to. So far I can only control color temperature, not light output. That being said, I have very little understanding of what I will have to set, so I'm mostly guessing where to put the PWMs :)

onebyside, 2019-02-11:
https://youtu.be/m_O24tTzv8g

rb2k, 2019-02-11:
Do you mean "Watch this if you want to learn how to figure out the GPIO settings" ? :)

(It's a 90 min video, so I'm not 100% sure)

rb2k, 2019-02-11:
So it seems that this combination gives me 2 sliders: Cold / Warm + Dark / Bright

GPIO4 PWM1
GPIO5 PWM1
GPIO12 PWM2

They seem to work. The toggle button does work as well.

Thanks for the link!

rb2k, 2019-02-11:
And this is the homeassistant config that I think works. There might be an opportunity to reduce some optional keys, but this allows me to dim the light and regulate its temperature:

```
  - platform: mqtt
    name: "Entryway Light"
    command_topic: "cmnd/MerkuryBR30White1/POWER"
    state_topic: "tele/MerkuryBR30White1/STATE"
    state_value_template: "{{value_json.POWER}}"
    availability_topic: "tele/MerkuryBR30White1/LWT"
    brightness_command_topic: "cmnd/MerkuryBR30White1/Dimmer"
    brightness_state_topic: "tele/MerkuryBR30White1/STATE"
    brightness_scale: 100
    on_command_type: "brightness"
    brightness_value_template: "{{value_json.Dimmer}}"
    color_temp_state_topic: "tele/MerkuryBR30White1/STATE"
    color_temp_command_topic: "cmnd/MerkuryBR30White1/CT"
    color_temp_value_template: "{{value_json.CT}}"
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    qos: 1
    retain: false
```

closed 2019-02-11 by merlinschumacher

jaburges, 2019-09-10:
For historical record this works (both GPIO & MQTT code) also for:
Geeni BR30 Tunable white
Geeni Lux 1050 A27 White

rb2k, 2019-09-29:
And another piece of knowledge for the future archeologists:

I have been able to flash it with esphome and used this configuration to be able to switch it on/off and set the color. I'm not quite sure why I don't specify GPIO4 anywhere, but I honestly don't really understand micro controllers that well
```yaml
output:
  - platform: esp8266_pwm
    pin: GPIO5
    id: cold_white
  - platform: esp8266_pwm
    pin: GPIO12
    id: warm_white

light:
  - platform: cwww
    name: "Merkury BR30 Bulb 01"
    cold_white: cold_white
    warm_white: warm_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
```

jaburges, 2020-09-29:
booo - seems Geeni has flashed the firmware :'( 
not able to flash anymore (new RGB BR30 - Prisma Drop)