ct-Open-Source/tuya-convert #94: Fcmila RGB-LED-Lampe wurde erfolgreich geflasht, Tasmota läuft, aber keine Funktionen wie AN / AUS möglich

Nuggelbeb, 2019-02-11:
Hallo,
ich habe folgendes Problem: ich habe nach dieser Anleitung [https://www.heise.de/ct/artikel/Tuya-Convert-IoT-Geraete-ohne-Loeten-vom-Cloud-Zwang-befreien-4283623.html](url) meine Fcmila LED bulb erfolgreich geflasht, die Lampe leuchtet weiß und ist über die IP-Adresse erreichbar. Mit Tasmota lässt sich alles einstellen, aber es geht kein An, kein Aus, keine Farbe, noch sonstiges. Sonoff B1 Modul oder ähnliche funktionieren auch nicht. Bei GPIO ... ist überall - None (00) - 

Kann ich diese Lampe jetzt weg schmeißen oder gibt es noch Hoffnung mit Tasmota?
PS: Update geht auch alles.

ctandi, 2019-02-11:
Moin Nuggelbeb,

da muss ich, ohne sehr vorwurfsvoll wirken zu wollen, an den Artikelhinweis erinnern:"Sie sollten sich im Klaren sein, was es heißt, eine alternative Firmware zu flashen." Aber auch ich kenne den Drang, schnell loslegen zu wollen und dabei gelegentlich mal etwas zu wenig zum Thema zu lesen ;)

Du musst die Lampe jetzt natürlich nicht wegschmeißen, es gibt noch Hoffnung. Erstmal musst du das Geräteprofil auf "Generic" ändern, um alle GPIOs zuweisen zu können. Danach probierst Du nach und nach die GPIOs durch, bis Du für alle drei Farben den jeweiligen GPIO gefunden hast (so wie hier ganz unten "RGBW colour bulbs" https://github.com/arendst/Sonoff-Tasmota/wiki/Mirabella-Genio-Bulb). Es wäre toll, wenn Du dich anschließend nochmal mit den GPIOs hier melden würdest, damit wir diese ins Wiki aufnehmen können. Der nächste mit dieser Lampe muss dann nicht mehr suchen.

Gruß
Andi

kueblc, 2019-02-11:
Hi @Nuggelbeb 

You can try the latest development build. We *just* implemented support for this type of RGB controller. https://github.com/arendst/Sonoff-Tasmota/pull/4991

Nuggelbeb, 2019-02-11:
mmm.,,, ich bin Anfänger und habe keine Ahnung

Also, habe das eingegeben und die Farbe blau ist schaltbar. (also nicht Schrott)

Modultyp = generisch
D3 GPIO0 => Keine
TX GPIO1 => Keine
D4 GPIO2 => Keine
RX GPIO3 => Keine
D2 GPIO4 => PWM1 (37)
D1 GPIO5 => PWM4 (40)
D6 GPIO12 => PWM2￼ (38)
D7 GPIO13 => PWM5 (41)
D5 GPIO14 => PWM3￼ (39)
D8 GPIO15 => Keine￼
D0 GPIO16 => Keine

Was muss ich da probier?

kueblc, 2019-02-11:
You'll need the latest development build to get this working and set the module type to SYF05

Nuggelbeb, 2019-02-11:
@kueblc  Sorry, ich bin das erste Mal hier auf GitHub, kenne mich noch nicht so richtig aus. 
Wo finde ich das letzte Entwicklungsbuild? Das downloade ich dann und füge es in Tasmota als Update ein?

kueblc, 2019-02-11:
I recommend [this guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Create-your-own-Firmware-Build-without-IDE) on the Sonoff-Tasmota wiki for building the firmware. You can then use the firmware upgrade function in the web portal to upload your custom build.

It's safest to do your own build from source as you'll know exactly what's in the binary, plus you'll be able to predefined settings for your device and network.

If you have trouble with this I can build and upload a binary for you, but best learn to fish than to have a fish!

Nuggelbeb, 2019-02-11:
Ok, builden habe ich herausgefunden über Atom, aber was muss im Verzeichnis "Sonoff Tasmota" geändert werden?

kueblc, 2019-02-11:
You can customize your build in `sonoff/my_user_config.h`. Specify your WiFi credentials, your MQTT server, make sure USE_SM16716 is enabled, and turn off anything you don't need to save space. You'll want to keep the final binary below 512kb. You may also wish to modify `platformio.ini` to specify wish version to build, otherwise it'll build all of them and will take a long time.

Nuggelbeb, 2019-02-11:
Sry, aber welche letzte Binärdatei?

kueblc, 2019-02-11:
It will be in `.pioenvs/sonoff(-DE)/firmware.bin`

Nuggelbeb, 2019-02-11:
Okay firmware.bin ist fertig und auch geflasht. Aber die Fcmila LED bulb taucht nicht auf. Was stelle ich da noch ein?

kueblc, 2019-02-12:
Set the module type to SYF05 and you should be able to toggle power and change brightness via web portal, and also set color via MQTT.

Nuggelbeb, 2019-02-12:
Leider geht nichts. Sie schaltet nicht und Farbwechsel ist mit MQTT auch nicht möglich. Sie flackert ganz leicht.
ioBroker zeigt 
[image]

[image]

kueblc, 2019-02-12:
Does the web page toggle not work at all? Does a slider appear?

Can you try to set GPIO13 to `SM16716 PWR`?

Also, try to send commands from the web console, such as `color ff0000` or `white 50`.

Nuggelbeb, 2019-02-12:
mit GPIO13 - SM16716 PWR`geht An/Aus
Schieberegler ist da - aber geht nicht

Nuggelbeb, 2019-02-12:
Befehle von der Webkonsole
[image]

keine Änderung

Nuggelbeb, 2019-02-12:
Bei -white 50- geht sie aus

07:46:16 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":49,"Color":"0,0,0,124","HSBColor":"0,0,0","Channel":[0,0,0,48]}

kueblc, 2019-02-12:
Okay sounds like your model is slightly different. Can you post any model numbers you can find on the bulb?

Let's go back to try to figure out the GPIOs then we can remap the SM16716 pins correctly. You said you were able to get a blue light before. Was that a light blue like a cold white or was it deep blue?

With model set to generic and GPIOs set to PWM channels like you had before, you can address each one individually with the command `channel<x> <0-100>` as in `channel1 100`. See if you can figure out what GPIO has which effect. Let us know the results!

Nuggelbeb, 2019-02-12:
Hallo, erst einmal möchte ich mich bedanken für die sehr gute Hilfe. Für mich ist das alles noch Neuland. Dabei dachte ich, dass es einfach ist und schnell geht, aber naja. 
Anbei sende ich erstmal ein Bild von der Lampe. Die Eingaben mache ich dann etwas später.
[image]

kueblc, 2019-02-12:
Hmm interesting, this indeed looks very different from https://github.com/gsimon75/Tasmota-SYF05-Notes/blob/master/led_board.jpg.

You can clearly see the ESP microcontroller on the right, same board as the LEDs and drivers, instead of on a dedicated board as Tuya distributes. From your picture front left I can identify a SLM211A constant current driver, which takes a PWM from the microcontroller. I'm not able to make out the large 8 pin chip in the back.

I don't see an SM16716 chip anywhere on here so that would explain why the SYF05 module isn't working for you. Once we identify the rear chip we'll have a better understanding of what this needs to drive the LEDs correctly.

Nuggelbeb, 2019-02-12:
Also die Lampe ist laut Verpackung eine Fcmila Bulb LED. 
[image]

Die Daten des Chips: 
BHB31
25Q8CCSI
PP3338

Vielleicht können Sie mir doch noch helfen oder mir sagen, an wen ich mich wenden kann. 
Danke nochmal.

kueblc, 2019-02-12:
Happy to help! Looks like the mystery chip is just some flash memory, not relevant for us.

Have you tried testing the individual GPIOs with `channel<x> <level>`? Any findings?

Nuggelbeb, 2019-02-13:
Leider nein.
Bei channel1 100 wird sie weiß, gibt man channel1 50 ein geht die Helligkeit runter. Weitere channels bringen keine Veränderung, nur die Anzeige - Command: unknown - 

06:47:42 CMD: channel1 100
06:47:42 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,0,0,0,255","HSBColor":"0,100,100","Channel":[100,0,0,0,100],"CT":500}

06:48:37 CMD: channel1 50
06:48:37 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"127,255,0,0,255","HSBColor":"90,100,100","Channel":[49,100,0,0,100],"CT":500}

johng2001, 2019-02-13:
I have the same bulb    if set Tasmota to generic 18
GPI/O 4  PWM 1= white  and PWM3= blue
GPI/O 12 =green
GPI/O 13 =blue
GPI/O 14 = red

if set Tasmota to  Arilux LC11 Module
You can  change (some) colors with the cold/warm slider, brightness with the dark/bright slider
and toggle on/off.

I have a backup of the original firmware if it might be useful,but cant seem to attach files here ?

johng2001, 2019-02-13:
Can someone add this info to the  Queo compatible devices  Wiki  entry please

ChipID: 61dd05
MAC: B4:E6:XX:XX:XX:XX
SdkVersion: 1.5.3(aec24ac9)
BootVersion: 4
BootMode: 1
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1048576
FlashChipSize: 1048576
FlashChipMode: 3
system_upgrade_userbin_check: user2 0x81000

johng2001, 2019-02-13:
Trying a color picker wheel from Node-Red   can only get white and red and adjust brightness of them.

08:15:10 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":8,"Color":"000000050E","HSBColor":"0,0,0","Channel":[0,0,0,1,5],"CT":402}
08:15:11 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":7,"Color":"000000050C","HSBColor":"0,0,0","Channel":[0,0,0,1,4],"CT":402}
08:15:11 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":6,"Color":"000000040A","HSBColor":"0,0,0","Channel":[0,0,0,1,3],"CT":402}
08:15:11 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":5,"Color":"0000000309","HSBColor":"0,0,0","Channel":[0,0,0,1,3],"CT":402}
08:15:11 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":4,"Color":"0000000207","HSBColor":"0,0,0","Channel":[0,0,0,0,2],"CT":402}
08:15:11 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":4,"Color":"0000000207","HSBColor":"0,0,0","Channel":[0,0,0,0,2],"CT":402}
08:15:14 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":3,"Color":"0000000205","HSBColor":"0,0,0","Channel":[0,0,0,0,1],"CT":402}
08:15:14 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":2,"Color":"0000000103","HSBColor":"0,0,0","Channel":[0,0,0,0,1],"CT":402}
08:15:15 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":3,"Color":"0000000205","HSBColor":"0,0,0","Channel":[0,0,0,0,1],"CT":402}
08:15:15 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":8,"Color":"000000050E","HSBColor":"0,0,0","Channel":[0,0,0,1,5],"CT":402}
08:15:15 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":9,"Color":"0000000610","HSBColor":"0,0,0","Channel":[0,0,0,2,6],"CT":402}
08:15:15 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":10,"Color":"0000000712","HSBColor":"0,0,0","Channel":[0,0,0,2,7],"CT":402}
08:15:15 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":10,"Color":"0000000712","HSBColor":"0,0,0","Channel":[0,0,0,2,7],"CT":402}
08:15:20 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"F952FF0000","HSBColor":"298,68,100","Channel":[97,32,100,0,0],"CT":499}
08:15:22 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF40860000","HSBColor":"338,75,100","Channel":[100,25,52,0,0],"CT":499}
08:15:23 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF16D40000","HSBColor":"311,91,100","Channel":[100,8,83,0,0],"CT":499}
08:15:24 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FB27FF0000","HSBColor":"299,85,100","Channel":[98,15,100,0,0],"CT":499}
08:15:25 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"4662FF0000","HSBColor":"231,73,100","Channel":[27,38,100,0,0],"CT":499}
08:15:26 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"537BFF0000","HSBColor":"226,67,100","Channel":[32,48,100,0,0],"CT":499}
08:15:29 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFFFF0000","HSBColor":"0,0,100","Channel":[100,100,100,0,0],"CT":499}
08:15:30 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"9CFF560000","HSBColor":"95,66,100","Channel":[61,100,33,0,0],"CT":499}
08:15:33 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":94,"Color":"62EF000000","HSBColor":"95,100,94","Channel":[38,93,0,0,0],"CT":499}
08:15:34 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF2B800000","HSBColor":"336,83,100","Channel":[100,16,50,0,0],"CT":499}
08:15:38 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":4,"Color":"0A01050000","HSBColor":"333,90,4","Channel":[3,0,1,0,0],"CT":499}
08:15:41 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON","Dimmer":100,"Color":"61C5FF0000","HSBColor":"202,62,100","Channel":[38,77,100,0,0],"CT":499}
08:15:46 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"OFF"}
08:15:46 MQT: stat/Queo_Fcmila/POWER = OFF
08:15:48 MQT: stat/Queo_Fcmila/RESULT = {"POWER":"ON"}
08:15:48 MQT: stat/Queo_Fcmila/POWER = ON

kueblc, 2019-02-13:
> Leider nein.
> Bei channel1 100 wird sie weiß, gibt man channel1 50 ein geht die Helligkeit runter. Weitere channels bringen keine Veränderung, nur die Anzeige - Command: unknown -
> 
> 06:47:42 CMD: channel1 100
> 06:47:42 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"255,0,0,0,255","HSBColor":"0,100,100","Channel":[100,0,0,0,100],"CT":500}
> 
> 06:48:37 CMD: channel1 50
> 06:48:37 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":100,"Color":"127,255,0,0,255","HSBColor":"90,100,100","Channel":[49,100,0,0,100],"CT":500}
> 
> tele/sonoff/STATE = {"Time":"2019-02-13T06:47:52","Uptime":"0T10:20:14","Vcc":3.403,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Dimmer":100,"Color":"255,0,0,0,255","HSBColor":"0,100,100","Channel":[100,0,0,0,100],"CT":500,"Scheme":0,"Fade":"ON","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"**********","BSSId":"E8:DF:70:C1:1F:8A","Channel":11,"RSSI":100}}

@Nuggelbeb Did you return the module to Generic and set the GPIO pins to PWMs? What were the settings you used?

>  I have a backup of the original firmware if it might be useful,but cant seem to attach files here ?

@johng2001 You can post it if you place it inside a .zip file first.

> Can someone add this info to the Queo compatible devices Wiki entry please

The wiki is open to edits from the public.

Nuggelbeb, 2019-02-13:
@kueblc Ich habe jetzt folgendes eingestellt
[image]

und da bekomme ich 

[image]

Lampe lässt sich ein- und ausschalten. Mit unterstem Schieberegler heller und dunkler machen und mit dem oberen Schieberegler (kalt, warm) die Farbe zwischen rot und blau wechseln. Leider fehlt die Farbe Gelb. Warum? Müssen noch andere GPIO´s belegt werden?

DavinKD, 2019-02-13:
An easy way to figure it out is to set each GPIO to a relay (1-8).  Then you turn each relay on and off and see what color shows on the bulb.  Once you figure that out, you set to PWM.

PWM1 = Red
PWM2 = Green
PWM3 = Blue
PMW4 = Cold White (I think)
PWM5 = Warm White (I think)

The whites could be swapped.

Nuggelbeb, 2019-02-14:
> An easy way to figure it out is to set each GPIO to a relay (1-8). Then you turn each relay on and off and see what color shows on the bulb. Once you figure that out, you set to PWM.
>
> PWM1 = Red [...]

mit welchen GPIO muss ich anfangen ?

kueblc, 2019-02-14:
Ah well this makes it tremendously straightforward :smile: 

Set GPIO04 to PWM4, GPIO12 to PWM2, GPIO13 to PWM3, GPIO14 to PWM1. Then you should be able to set color with `color <hex>` via MQTT or the web portal, turn it on and off, and dim.

Nuggelbeb, 2019-02-14:
Ich bin raus. Lampe ist nicht mehr erreichbar. Programmiermodus - 3 x An/Aus - geht auch nicht und Serial geht auch nicht. Aber sie leuchtet weiß und blau blinkert.

johng2001, 2019-02-14:
I tried more things and managed to get Node-Red color picker working (I think) correctly
though the colors are correct they seem a bit dimer ( as if white is always mixed in )

Generic 18
???      I/O   2  = PWM 4
White   I/O   4  = PWM 5 
Green  I/O  12 = PWM 2
Blue     I/O  13 = PWM 3
Red     I/O  14 = PWM 1

adjust the cold/warm slider  to bring back  pure white
dark/bright  works as expected.

johng2001, 2019-02-14:
I tried more things and managed to get Node-Red color picker working (I think) correctly
though the colors are correct they seem a bit dimer ( as if white is always mixed in )

Generic 18
???      I/O   2  = PWM 5
White   I/O   4  = PWM 4
Green  I/O  12 = PWM 2
Blue     I/O  13 = PWM 3
Red     I/O  14 = PWM 1

adjust the cold/warm slider  to bring back  pure white
dark/bright  works as expected.

kueblc, 2019-02-14:
Don't worry @Nuggelbeb, it's just reset it's configuration. Check to see if it's broadcasting a WiFi AP called sonoff-XXXX. Connect and reconfigure your WiFi settings at 192.168.4.1.

What caused it to reset?

kueblc, 2019-02-14:
@johng2001 Do you have both cold and warm white LEDs in this bulb? When you define 5 PWMs, PWM4 is CW and PWM5 is WW.

johng2001, 2019-02-15:
I tried both PWM 4 and 5  into I/O 4 , the white looks the same to me... also  as long as  the remaining PWM is put into  any one of the unused I/O's  then  cold/warm slider can be used to bring back white
and  dark/bright slider  works as expected  from tasmota web page.
to get the colors I  use ( because it works and is easy)  Node-Red color picker  to send MQTT commands.

still can't attach files here (maybe my version of Firefox is not supported)

johng2001, 2019-02-15:
Best settings so far for me.
GPIO 2 PWM 4   (any unused I/O just to get slider + PWM 4 or 5 can be swapped seems same result)
                                                                                        pwm 4 should = cold white
GPIO 4 PWM 5 = white							pwm 5 should = warm white				
GPIO 12 PWM 2 = green							pwm 2 should = green
GPIO 13 PWM 3= blue							pwm 3 should = blue
GPIO 14 PWM 1  = red							pwm 1 should = red

in node red color picker   format HEX + show  hue and lightness slider.
use cold/warm slider in Tasmota to return to White  obviously there is a MQTT command but
enough for now.

Nuggelbeb, 2019-02-15:
> Don't worry @Nuggelbeb, it's just reset it's configuration. Check to see if it's broadcasting a WiFi AP called sonoff-XXXX. Connect and reconfigure your WiFi settings at 192.168.4.1.
> 
> What caused it to reset?

Ja, ich habe bei den GPIOs rumgespielt und dann ging nichts mehr. Dann habe ich sie nicht mehr im Netz gefunden, auch unter 192.168.4.1 ist sie nicht zu finden. Hardwareseitig geht auch nichts mehr, Lötpins sind abgerissen und ging ja vorher auch nur über W-Lan.

kueblc, 2019-02-15:
If it still blinks, it's not dead! When you compiled your firmware, what was the value of `WIFI_CONFIG_TOOL` and `WIFI_CONFIG_NO_SSID`?

kueblc, 2019-02-15:
(These are found in `my_user_config.h`)

Nuggelbeb, 2019-02-15:
#define WIFI_CONFIG_TOOL       WIFI_RETRY
#define WIFI_CONFIG_NO_SSID    WIFI_WPSCONFIG

kueblc, 2019-02-15:
If your router supports WPS, press the button on your router and then power your bulb. It should hopefully reconnect and you'll be able to redo your configuration. If you know what combination of GPIO settings caused the last reset, let us know so we can avoid that again.

Nuggelbeb, 2019-02-15:
Leider hat das auch nichts gebracht. Die Lampe ist im 7. Himmel. 
Ich glaube, ich habe auf GPIO 0 - Relay 1 oder so eingestellt. Leider weiß ich es nicht mehr, habe zu viel rumgespielt.

johng2001, 2019-03-01:
my node red setup for this bulb  flashed with Tasmota

[{"id":"fa350b71.b8ea2","type":"ui_switch","z":"27b0e016.c0e5a8","name":"","label":"ON / OFF","tooltip":"","group":"517aae8c.06068","order":18,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"cmnd/Queo_Fcmila/power","style":"","onvalue":"on","onvalueType":"str","onicon":"","oncolor":"","offvalue":"off","offvalueType":"str","officon":"","offcolor":"","x":136,"y":2342.75,"wires":[["c0484a50.12baf"]]},{"id":"c0484a50.12baf","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"","qos":"","retain":"","broker":"2902344d.99c694","x":444,"y":2341.25,"wires":[]},{"id":"45da852.2fbdefc","type":"ui_slider","z":"27b0e016.c0e5a8","name":"Set Brightness","label":"SET BRIGHTNESS","tooltip":"","group":"517aae8c.06068","order":2,"width":0,"height":0,"passthru":true,"outs":"all","topic":"","min":"1","max":"100","step":1,"x":132.75,"y":2401.5,"wires":[["2e6aad51.34d77a"]]},{"id":"2e6aad51.34d77a","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/Dimmer","qos":"","retain":"","broker":"2902344d.99c694","x":548,"y":2461.75,"wires":[]},{"id":"f1ccb0ee.c896","type":"mqtt in","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/Dimmer","qos":"2","broker":"2902344d.99c694","x":849,"y":2457.25,"wires":[[]]},{"id":"ae8b953d.e7bf8","type":"ui_colour_picker","z":"27b0e016.c0e5a8","name":"Color Picker","label":"COLOR PICKER","group":"517aae8c.06068","format":"hex","outformat":"string","showSwatch":true,"showPicker":true,"showValue":true,"showHue":true,"showAlpha":true,"showLightness":true,"dynOutput":"false","order":1,"width":0,"height":0,"passthru":true,"topic":"","x":135.75,"y":2472.5,"wires":[["a461755b.a3599"]]},{"id":"73089cc.5176a64","type":"ui_dropdown","z":"27b0e016.c0e5a8","name":"Select Color","label":"SELECT COLOR","tooltip":"","place":"CHOOSE","group":"517aae8c.06068","order":4,"width":0,"height":0,"passthru":true,"options":[{"label":"RED","value":"eb0000","type":"str"},{"label":"GREEN","value":"00b80c","type":"str"},{"label":"BLUE","value":"0000CD","type":"str"},{"label":"WHITE","value":"0xFFFFFF","type":"str"},{"label":"ORANGE","value":"f52100","type":"str"},{"label":"PURPLE","value":"FF00FF","type":"str"},{"label":"YELLOW","value":"FFFF00","type":"str"},{"label":"AQUA","value":"00FA9A","type":"str"},{"label":"MAROON","value":"8B0000","type":"str"}],"payload":"","topic":"","x":140.75,"y":2536.5,"wires":[["a461755b.a3599"]]},{"id":"a461755b.a3599","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/color","qos":"","retain":"","broker":"2902344d.99c694","x":550,"y":2541.5,"wires":[]},{"id":"741c5d43.7f3b9c","type":"mqtt in","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/color","qos":"2","broker":"2902344d.99c694","x":849,"y":2541.75,"wires":[[]]},{"id":"d1a578a5.54674","type":"ui_button","z":"27b0e016.c0e5a8","name":"","group":"517aae8c.06068","order":5,"width":0,"height":0,"passthru":true,"label":"cycle color","tooltip":"","color":"","bgcolor":"","icon":"","payload":"3","payloadType":"str","topic":"cmnd/Queo_Fcmila/scheme","x":158.75,"y":2756.5,"wires":[["e3ae9a81.e6e7e"]]},{"id":"e3ae9a81.e6e7e","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"","qos":"","retain":"","broker":"2902344d.99c694","x":425.75,"y":2757.5,"wires":[]},{"id":"e1ffc018.53d6d","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/scheme","qos":"","retain":"","broker":"2902344d.99c694","x":648.75,"y":2755.25,"wires":[]},{"id":"1308eda3.df1dc2","type":"ui_button","z":"27b0e016.c0e5a8","name":"","group":"517aae8c.06068","order":6,"width":0,"height":0,"passthru":false,"label":"Rapid cycle color","tooltip":"","color":"","bgcolor":"","icon":"","payload":"4 speed 2","payloadType":"str","topic":"cmnd/Queo_Fcmila/scheme","x":181.75,"y":2828.25,"wires":[["6558acac.d4f3ec"]]},{"id":"6558acac.d4f3ec","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"","qos":"","retain":"","broker":"2902344d.99c694","x":425.5,"y":2827.5,"wires":[]},{"id":"3d0731d5.43c99e","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/scheme","qos":"","retain":"","broker":"2902344d.99c694","x":651.5,"y":2828.5,"wires":[]},{"id":"517aae8c.06068","type":"ui_group","z":"","name":"FCmilla Smart Bulb","tab":"43155387.db55dc","order":1,"disp":true,"width":"6","collapse":false},{"id":"2902344d.99c694","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"43155387.db55dc","type":"ui_tab","z":"","name":"FCmilla Lamp Bedroom","icon":"dashboard","order":3,"disabled":false,"hidden":false}]

johng2001, 2019-03-02:
My Node red settings for this bulb ( mistake in first try)

[{"id":"fa350b71.b8ea2","type":"ui_switch","z":"27b0e016.c0e5a8","name":"","label":"ON / OFF","tooltip":"","group":"517aae8c.06068","order":18,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"cmnd/Queo_Fcmila/power","style":"","onvalue":"on","onvalueType":"str","onicon":"","oncolor":"","offvalue":"off","offvalueType":"str","officon":"","offcolor":"","x":136,"y":2342.75,"wires":[["c0484a50.12baf"]]},{"id":"c0484a50.12baf","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"","qos":"","retain":"","broker":"2902344d.99c694","x":444,"y":2341.25,"wires":[]},{"id":"45da852.2fbdefc","type":"ui_slider","z":"27b0e016.c0e5a8","name":"Set Brightness","label":"SET BRIGHTNESS","tooltip":"","group":"517aae8c.06068","order":2,"width":0,"height":0,"passthru":true,"outs":"all","topic":"cmnd/Queo_Fcmila/Dimmer","min":"1","max":"100","step":1,"x":132.75,"y":2401.5,"wires":[["2e6aad51.34d77a"]]},{"id":"ae8b953d.e7bf8","type":"ui_colour_picker","z":"27b0e016.c0e5a8","name":"Color Picker","label":"COLOR PICKER","group":"517aae8c.06068","format":"hex","outformat":"string","showSwatch":true,"showPicker":true,"showValue":true,"showHue":true,"showAlpha":true,"showLightness":true,"dynOutput":"false","order":1,"width":0,"height":0,"passthru":true,"topic":"","x":135.75,"y":2472.5,"wires":[["a461755b.a3599"]]},{"id":"73089cc.5176a64","type":"ui_dropdown","z":"27b0e016.c0e5a8","name":"Select Color","label":"SELECT COLOR","tooltip":"","place":"CHOOSE","group":"517aae8c.06068","order":4,"width":0,"height":0,"passthru":true,"options":[{"label":"RED","value":"eb0000","type":"str"},{"label":"GREEN","value":"00b80c","type":"str"},{"label":"BLUE","value":"0000CD","type":"str"},{"label":"WHITE","value":"0xFFFFFF","type":"str"},{"label":"ORANGE","value":"f52100","type":"str"},{"label":"PURPLE","value":"FF00FF","type":"str"},{"label":"YELLOW","value":"FFFF00","type":"str"},{"label":"AQUA","value":"00FA9A","type":"str"},{"label":"MAROON","value":"8B0000","type":"str"}],"payload":"","topic":"","x":140.75,"y":2536.5,"wires":[["a461755b.a3599"]]},{"id":"a461755b.a3599","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/color","qos":"","retain":"","broker":"2902344d.99c694","x":550,"y":2541.5,"wires":[]},{"id":"2e6aad51.34d77a","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/Dimmer","qos":"","retain":"","broker":"2902344d.99c694","x":548,"y":2461.75,"wires":[]},{"id":"f1ccb0ee.c896","type":"mqtt in","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/Dimmer","qos":"2","broker":"2902344d.99c694","x":849,"y":2457.25,"wires":[[]]},{"id":"741c5d43.7f3b9c","type":"mqtt in","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/color","qos":"2","broker":"2902344d.99c694","x":849,"y":2541.75,"wires":[[]]},{"id":"c1b85ead.2a1ae","type":"comment","z":"27b0e016.c0e5a8","name":"built in tasmote color cycle","info":"","x":202.75,"y":2726.75,"wires":[]},{"id":"d1a578a5.54674","type":"ui_button","z":"27b0e016.c0e5a8","name":"","group":"517aae8c.06068","order":5,"width":0,"height":0,"passthru":true,"label":"cycle color","tooltip":"","color":"","bgcolor":"","icon":"","payload":"3","payloadType":"str","topic":"cmnd/Queo_Fcmila/scheme","x":158.75,"y":2756.5,"wires":[["e3ae9a81.e6e7e"]]},{"id":"1308eda3.df1dc2","type":"ui_button","z":"27b0e016.c0e5a8","name":"","group":"517aae8c.06068","order":6,"width":0,"height":0,"passthru":false,"label":"Rapid cycle color","tooltip":"","color":"","bgcolor":"","icon":"","payload":"4 speed 2","payloadType":"str","topic":"cmnd/Queo_Fcmila/scheme","x":181.75,"y":2828.25,"wires":[["6558acac.d4f3ec"]]},{"id":"e3ae9a81.e6e7e","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"","qos":"","retain":"","broker":"2902344d.99c694","x":425.75,"y":2757.5,"wires":[]},{"id":"6558acac.d4f3ec","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"","qos":"","retain":"","broker":"2902344d.99c694","x":425.5,"y":2827.5,"wires":[]},{"id":"3d0731d5.43c99e","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/scheme","qos":"","retain":"","broker":"2902344d.99c694","x":651.5,"y":2828.5,"wires":[]},{"id":"e1ffc018.53d6d","type":"mqtt out","z":"27b0e016.c0e5a8","name":"","topic":"cmnd/Queo_Fcmila/scheme","qos":"","retain":"","broker":"2902344d.99c694","x":648.75,"y":2755.25,"wires":[]},{"id":"517aae8c.06068","type":"ui_group","z":"","name":"FCmila Smart Bulb","tab":"43155387.db55dc","order":1,"disp":true,"width":"6","collapse":false},{"id":"2902344d.99c694","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"43155387.db55dc","type":"ui_tab","z":"","name":"FCmila Lamp Bedroom","icon":"dashboard","order":3,"disabled":false,"hidden":false}]

kueblc, 2019-03-05:
Unfortunately sounds like @Nuggelbeb's bulb is bricked, so no amount of configuration assistance would resolve this. Closing, if there's any further questions please open a new issue on the relevant project.

closed 2019-03-05 by kueblc

nightfly0815, 2019-06-27:
Hi in witch mode should the bulb are for the the tuya converter?Fast or short blinking?

wowik2018, 2020-05-23:
[image]

Работает отлично