ct-Open-Source/tuya-convert #96: Big list of successes

jimkernsjr, 2019-02-15:
Hi All!

Thanks for the great project.  I had good success with it, but I must say the prereqs are a bit specific so it seems absolutely imperative that it's used on a new, clean install.  I thought "I know linux well, I'll make it work" on my laptop, and I did, for one device and it never did again.  After it kept switching my wifi to unmanaged mode, I decided I wasted too much time and lets just follow the instructions and move on.  To my joy, I came up with this big list of supported devices.  The first 2 will please everyone in the US:  :)

Merkury-Walmart_Singleplug
Merkury-Walmart_WifiBulb-A19SingleColor

AWP01L-7307
c128-3outletOutdoor
Dilisens_SP201
iMahPlug_BuffNo03-SWN03-us_nightlight
KMC_4outletWall
KMC_5outletpowerstrip-4switchable-wUSB
KMC_70011
Premiertek_HS1050
ShenzhenAvatarControls_OD-01
TanTan_Dualplug-WP2
Welcuic_FK170-701
Xenon_PW702U
ym-sg01ux

I have backup binfiles, logs, and the flashinfo if needed.

Thanks again for all this hard work - I was hording these I bought mostly from Amazon and was dreading doing it the "old fashioned way"!

kueblc, 2019-03-05:
Hi @jimkernsjr 

Thanks for your report! Have you been able to edit the wiki to add your findings?

ragenhe, 2019-03-13:
@jimkernsjr - Do you possible remember which flash method was successful for you on the Merkury white wifi bulb?  I can't get flash3 to work, and I'm attempting to finish the flash in recovery mode.

Thanks!

kueblc, 2019-09-28:
Closing due to inactivity, feel free to open a new issue

closed 2019-09-28 by kueblc

jimkernsjr, 2019-09-28:
Hello..
I didn't use a flash method...I used tuya convert.