ct-Open-Source/tuya-convert #97: ANCCY SP1 GPIO Config missing

jnikodemus, 2019-02-15:
Hi,

I've successfully flashed the **Anccy SP1 using firmware 6.4.1**. But in GPIO-Configuration I'm _missing_ some of the options such as

- **GPIO4: HLW8012 CF**
- **GPIO5: HLWBL CF1**
- **GPIO12: HLWBL SELi**

Everything except the powersensor seems to work fine though...
Is there anyone who has the same problem or maybe fixed it already?

**btw: Great work!**

kueblc, 2019-02-15:
Glad your flash was successful. Your issue is a Tasmota one, you can close your issue here and refer to Sonoff-Tasmota for support if needed.

FWIW I think you might need to compile your firmware with the `USE_HLW8012` feature enabled. Perhaps the prebuilt `sensors` binary has that enabled.

Anyway you'll get better firmware specific support in that firmware's support channels.

jnikodemus, 2019-02-15:
Ah! I see...
Thanks for your answer!

closed 2019-02-15 by jnikodemus

kueblc, 2019-02-15:
Just noticed this in the changelog
```
 * 6.4.1.14 20190203
 * Add user configuration of HLW8012 and HJL-01/BL0937 Energy Monitoring as used in Sonoff S31, Pow Ra and many Tuya based devices
```
So make sure you're using *6.4.1.14* or higher