ct-Open-Source/tuya-convert #98: Successfully Flashed

mediacowboy, 2019-02-15:
2 - MaxKare Dual Outlet
5 - TanTan WP3

For the energy monitoring on the MaxKare you have to run Dev Tasmota.

danhounshell, 2019-03-04:
+1 on the TanTan WP3, I flashed 4 of them this weekend and got them working in HA. Looks to be the same thing as the Gosund WP3, just different brand name on it. https://blakadder.github.io/templates/gosund_wp3.html

Munns86, 2019-03-19:
> +1 on the TanTan WP3, I flashed 4 of them this weekend and got them working in HA. Looks to be the same thing as the Gosund WP3, just different brand name on it. https://blakadder.github.io/templates/gosund_wp3.html

If you don't mind me asking did you any issues getting the TanTan's to flash. I'm doing this on a Pi3b and can't get it to work. Once I put my TanTan in pair modem that is where it seems to fail. The script doesn't seem to find the TanTan.

kueblc, 2019-03-23:
Thanks for the report @mediacowboy. Feel free to add your findings to the wiki for easy visibility.

@Munns86 Open a new issue if you're having problems.

closed 2019-03-23 by kueblc