ct-Open-Source/tuya-convert #100: kali-rolling currently not supported by NodeSource

maikthulhu, 2019-02-17:
The install_prereq.sh script attempts to pull down the installer from NodeSource and execute it (as per their automatic install instructions).  The installer executes `lsb_release -s -c` to get the distro codename, but currently this returns `n/a` (expecting jessie or sid or whatever).  NodeSource doesn't support Kali and, as such, attempting to use the recommended distribution and default flashing instructions doesn't work.

The rest of the prereqs finish installing and it ends up installing node 10, but the smartconfig process fails its attempt due to the debug module not being found (relevant stacktrace below).

scripts/smarthack-smartconfig.log:
```
internal/modules/cjs/loader.js:583
    throw err;
    ^

Error: Cannot find module 'debug'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/root/tuya-convert/scripts/smartconfig/smartconfig.js:16:15)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
```

kueblc, 2019-02-17:
Node is being phased out of the project in favor of an all Python solution. You can try the `smart-link-py` branch by doing `git clone -b smart-link-py https://github.com/kueblc/tuya-convert.git` as this removes the dependency on NodeJS.

maikthulhu, 2019-02-17:
Ahh I see.  I wiped out kali in favor of a raspbian install.  I'll close out this issue since there's not much anyone can do about node support for kali here.  Thanks for the info, I'll give that branch a try.

closed 2019-02-17 by maikthulhu