ct-Open-Source/tuya-convert #99: Successfully flashed Feit BPA800 RGBW - Need help figuring out GPIO Tasmota Setup

jtroberts1, 2019-02-17:
I have Successfully flashed the Feit BPA800 RGBW light bulb however I need help figuring out the GPIO's and Tasmota setup.  I have removed the bulb cover and I can see inside.  I have taken a picture and can clearly see GPIO's labeled...  4,14,5,12,13.  the top PCB with the LED's has the SM16726P chip.  I am not 100% sure, but the RGB LED's are a single package, just like a ws2812b.  I can get the RGB led's to turn on, but cannot change their colors.  Here's what I have so far...

GPIO5 = soft White LED's (the light yellow LED's)
GPIO14 = RGB LED's
[image]
[image]

Warm White LED = ??? (orangish LED's)
how to change/set colors

Thanks

TheZoker, 2019-02-17:
I'll guess you should better ask that in the Tasmota GitHub repo:
https://github.com/arendst/Sonoff-Tasmota/issues

cscheib, 2019-02-20:
@jtroberts1 are those the ones from Costco?  I have a few, but I haven't gotten around to setting up a Raspberry Pi to flash them yet.

There's some discussion here that I believe applies to these: https://github.com/arendst/Sonoff-Tasmota/pull/4991

jtroberts1, 2019-02-21:
Yes sir.  the only thing I should note, and I think many people run into this as well, after following the instructions to setup tuya-convert, reboot the PI.  I tried without a reboot, and it never flashed.  Rebooted the PI, and worked perfectly.

I'll check out the link you sent.  TY

cogneato, 2019-03-02:
2 of these for under $30 at Costco. They were gone for a few months at ours but are now back. I couldn't resist. Flashed them successfully as well but can't get color.

kueblc, 2019-03-02:
It's likely that the link @cscheib posted addresses your issue, but please refer to the Sonoff-Tasmota community to assist with post installation configuration.

badabing2005, 2019-03-03:
You can try this approach
- Set the module to Generic 18
- Then one GPIO at a time set the value to PWM1, save
- When it restarts, toggle and see if any color comes up, if it does, you've determined the GPIO for that color, take a note of the color and GPIO
- Rinse and repeat for PWM2, PWM3, PWM4 (If you have RGBW)

Hope it helps

closed 2019-03-05 by kueblc

Sawadee2u, 2019-07-07:
Could you share the connections you used for programming? New firmware no longer allows Tuya-Convert?

Sawadee2u, 2019-07-07:
For reference 
Gnd and 3.3 v are labeled on 2 of the 8 feed through pins to the sister board (4 on each side).
Tx, Rx, & GPIO 0 (ESP8266EX pins 26, 25, & 15) have round contact pads on the reverse side of the 8266 side. Follow connection from pin to feed through hole to reverse side right to the pad. Solder to the pad with light wires temporarily. Again, new firmware on the current versions stopped Tuya-convert from flashing.

simonrash, 2019-07-12:
Hope this helps explain the wiring required. The wires on the right are the the serial data. Tx is on the far right with the Rx to the left of it. The furthest away on the left set of four is GPIO0.
While I have not used them this time as I was not sure so connected directly to the 3v3 and GND on the joining connector. The one closest is of the four is GND and the next one in is 3v3. I have not proved yet that it can be programmed using these power lines. 

> PLEASE DON'T TRY THIS AT HOME.
I was unable to program the board using an external 3v3 supply so plugged it into a light socket for it's power supply this time. 

[image]

rlowens, 2019-09-08:
> Hope this helps explain the wiring required. The wires on the right are the the serial data. Tx is on the far right with the Rx to the left of it. The furthest away on the left set of four is GPIO0. [...]

Here's the test pads labeled to be clear on the TYLC2V module
[image]
(I used these 2 images from Tuya and played with contrast to trace the pins, and they agree with StealthChesnut meter-testing. https://images.tuyacn.com/smart_res/developer_default/TYLC2V-1.png https://images.tuyacn.com/smart_res/developer_default/TYLC2V-2.png )

martinvana, 2019-09-17:
Great pictures and description! (I discovered these *after* I took the thing apart and took my set of photos ;-)
Did you figure out Tasmota template to work with this HW?
Thanks!

jbardi, 2019-12-02:
This is the Tasmota template for this bulb

{“NAME”:" BPA800/RGBW",“GPIO”:[255,255,255,255,37,38,255,255,141,142,140,255,255],“FLAG”:0,“BASE”:48}

jtroberts1, 2019-12-05:
On the blakadder.github.io website you will find a bulb titled "Feit Electric BPA800/RGBW/AG/2 RGBW"  DO NOT use this for the bulbs purchased from Costco.  Instead, use the bulb title "Feit Electric OM60/RGBW/CA/AG RGBW".  In Tasmota use console and set "setoption37 54".  The template is...
{"NAME":"OM60/RGBW","GPIO":[0,0,0,0,140,37,0,0,38,142,141,0,0],"FLAG":0,"BASE":18}

Tchekda, 2021-05-26:
For those interested. This is my setup : 
```
{"NAME":"Feit Bulb","GPIO":[1,1,1,1,2944,2912,1,1,416,2976,417,1,1,0],"FLAG":0,"BASE":18}
```
with `SetOption37 1`
Model : `BPA800/RGBW/AG/2/UK`

Joshyakadamien, 2024-07-10:
> Here's the test pads labeled to be clear on the TYLC2V module [image] (I used these 2 images from Tuya and played with contrast to trace the pins, and they agree with StealthChesnut meter-testing. [...]

"Okay now I know what was wrong. Apparently you have to unplug the cable for IO0 5 seconds after boot. Now I have successfully flashed tasmota "

Src:
https://www.reddit.com/r/tasmota/comments/10tpggx/trying_to_flash_a_tylc2v/